Koenkk/zigbee2mqtt #5528: Docker Compose Deployment with Conbee II from Dresden Elektronik

nicedevil007, 2021-01-03:
Hey guys,

I have a conbee II stick that is updated to the latest firmware from dresden elektronik since today.
Unfurtunately I can't get zigbee2mqtt to run with the provided docker compose file:

```
version: 2

services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    hostname: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - zigbee2mqtt_data:/app/data
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    network_mode: host
```

The stick is definetly on /dev/ttyACM0 because it is working with it when I'm using the phoscon webui.

Logfile of the docker container tells me this:

```
Zigbee2MQTT:info  2021-01-03 14:33:18: Logging to console and directory: '/app/data/log/2021-01-03.14-33-18' filename: log.txt,
Zigbee2MQTT:info  2021-01-03 14:33:18: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d),
Zigbee2MQTT:info  2021-01-03 14:33:18: Starting zigbee-herdsman (0.13.46),
Zigbee2MQTT:error 2021-01-03 14:33:18: Error while starting zigbee-herdsman,
Zigbee2MQTT:error 2021-01-03 14:33:18: Failed to start zigbee,
Zigbee2MQTT:error 2021-01-03 14:33:18: Exiting...,
Zigbee2MQTT:error 2021-01-03 14:33:18: Error: No path provided and failed to auto detect path,
    at Function.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:82:27),
    at Generator.next (<anonymous>),
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:24:58),
    at processTicksAndRejections (internal/process/task_queues.js:97:5),
npm ERR! code ELIFECYCLE,
npm ERR! errno 1,
npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`,
npm ERR! Exit status 1,
```

Koenkk, 2021-01-03:
Check the last point of https://www.zigbee2mqtt.io/information/supported_adapters.html#conbee-ii

nicedevil007, 2021-01-03:
was able to get it working now... sry I should missed this line.

Can you tell me what would be a right publish message to change a bulb from on to off?`

tryed this without any effect :(

```
{
  "state": "ON", // Or "OFF", "TOGGLE"
  "brightness": 255, // Value between 0 and 255
  "color": {"x": 0.123, "y": 0.123} // Color in XY
}
```
or this to turn it off

```
{
  "state": "OFF", // Or "OFF", "TOGGLE"
  "brightness": 255, // Value between 0 and 255
  "color": {"x": 0.123, "y": 0.123} // Color in XY
}

```

Koenkk, 2021-01-03:
`{"state":"OFF"}`, check e.g. https://www.zigbee2mqtt.io/devices/4713407.html#light

Closing since OP issue has been fixed.

closed 2021-01-03 by Koenkk