Koenkk/zigbee2mqtt #5534: Get state automatically ?

Jeoffreybauvin, 2021-01-03:
## Is your feature request related to a problem? Please describe

I'm using Ikea Tradfri  bulbs.  On [this page](https://www.zigbee2mqtt.io/devices/LED1546G12.html), I can see that : 

> To read the state send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"state": ""}.

It's working, of course, but does z2m trigger it automatically eventually ?

## Describe the solution you'd like
Maybe when the bulb is on, report back the state  automatically to z2m ?

## Describe alternatives you've considered

## Additional context

sjorge, 2021-01-03:
If you enable reporting, the bulb will report back.
https://www.zigbee2mqtt.io/information/report.html

On that note, the table is out of date as it looks you can manually setup reporting for all the listed brand aside from gledopto and hue.

Jeoffreybauvin, 2021-01-03:
Great !  I can confirm it's working :) 

Maybe the documentation needs to be updated :)

Thanks @sjorge

closed 2021-01-03 by Jeoffreybauvin

sjorge, 2021-01-03:
> Maybe the documentation needs to be updated :)

It probably should, although this was a very recent discovery (last week)
There are some more changes in the pipeline to automatically do the proper setup based on colorCapabilities attribute, so I suspect once this all gets ironed over a bit, the docs will follow.