Koenkk/zigbee2mqtt #5538: Ikea OTA Test Server option not working

filewalker1, 2021-01-04:
# Bug Report
## What happened
Updated zigbe2mqqt HomeAssistant Plugin to latest v1.17.0 to fix the problem with broken updates from ikea server. As described in the changelog/documentation i added "ikea_ota_use_test_url: true" to the advanced section of the zigbee2mqtt plgin configuration.yaml and restarted the plugin. I still get the truncated Firmware not long enough message. WebUI-> Settings -> Raw -> i am not able to find the ikea test server option.

## What did you expect to happen 
Ikea tradfri updates working without "truncated" error message

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC26X2R1
Adapter firmware version: 20201026

jeroenvdpoll, 2021-01-04:
I experience the same.

Add to this the following I get the following supervisor log message:
21-01-04 09:37:40 WARNING (MainThread) [supervisor.addons.validate] Unknown options ikea_ota_use_test_url

This is part of my config:
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
advanced:
  ikea_ota_use_test_url: true
  log_level: debug

Koenkk, 2021-01-04:
Please open this issue here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

closed 2021-01-04 by Koenkk