Koenkk/zigbee2mqtt #5542: Support for device Hue Outdoor Econic wall lantern Philips

vinta7, 2021-01-04:
Hi.
I have several 'Hue Outdoor Econic wall lantern Philips 1744030P7' lamps outside, can someone please make support for these ?, today there is support for 'Hue Outdoor Econic wall lantern Philips 1743930P7', the only difference between the 2 versions is the psychical mounting of the support arm,  1744030P7 have the arm on the top, 1743930P7 have the arm in the bottom. So it should be an copy/paste 1743930P7 -> 1744030P7 with a different picture.
Regards
[image]
new picture

sjorge, 2021-01-04:
You can update devices.js to see if it works when you change:

```
zigbeeModel: ['1743930P7']
```

to

```
zigbeeModel: ['1743930P7', '1744030P7']
```

Here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2915

If it works, you can make a PR (or I can do it)

vinta7, 2021-01-05:
Thanks for the update, I will try it soon but I having some issues with hassio-zigbee2mqtt.

svrooij, 2021-01-06:
@sjorge you should copy the permalink to the line, that file is changing a lot so a specific line in master might be a different line tomorrow.

sjorge, 2021-01-06:
> @sjorge you should copy the permalink to the line, that file is changing a lot so a specific line in master might be a different line tomorrow.

Probably a good idea yeah, fixed

vinta7, 2021-01-14:
Hi, it works great, I would be grateful if you could update the repository for me :-), I am a newbie on github

Koenkk, 2021-01-15:
Closing this as https://github.com/Koenkk/zigbee-herdsman-converters/pull/2081 is merged. Thanks for helping out @sjorge 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-15 by Koenkk