Koenkk/zigbee2mqtt #5543: Lidl motion sensor HG06335 fails to pair with 1.17.0 / 1.16.2

tehohosma, 2021-01-04:
Hi,

I am trying to pair the lidl motion sensor HG06335 which should be already supported. I've tried the tips from issues 5201 and 5179 but none of them seem to work in my case. Tried to maky the zigbee-herdsman log but I am not sure if I was successful. The issue has been happening with 1.16.2 and now with 1.17.0 (has never worked).

Tried removing and restoring the battery, interview starts and shortly device leaves the network
Tried using the reset button, pretty much the same end result
Tried close / far from the adapter, same result

Device in question is the 0xec1bbdfffec83699 
[lidl_motion.txt.txt](https://github.com/Koenkk/zigbee2mqtt/files/5764154/lidl_motion.txt.txt)
[zigbee-herdsman_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5764155/zigbee-herdsman_log.txt)

in the log files. Closest tip I could get was this:
debug 2021-01-04 11:30:47: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
debug 2021-01-04 11:30:55: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":65}'
debug 2021-01-04 11:30:56: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
debug 2021-01-04 11:30:58: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2021-01-04 11:31:01: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"2019.12.12"}'
debug 2021-01-04 11:31:05: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{}'
debug 2021-01-04 11:31:13: No converter available for 'HG06335' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'
debug 2021-01-04 11:35:57: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"_TZ1800_fcdjzz3s"}'
debug 2021-01-04 11:35:59: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
debug 2021-01-04 11:36:07: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
debug 2021-01-04 11:36:10: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":65}'
debug 2021-01-04 11:36:13: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417

robbinonline, 2021-01-04:
I have to same problem with this motion sensor. Used the last version 1.17.1
Interview started and then failed.

Zigbee2MQTT:info  2021-01-04 16:12:04: Starting interview of '0xec1bbdfffec7406e'
Zigbee2MQTT:info  2021-01-04 16:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffec7406e","ieee_address":"0xec1bbdfffec7406e","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-01-04 16:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffec7406e"},"type":"pairing"}'
Zigbee2MQTT:info  2021-01-04 16:12:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffec7406e","ieee_address":"0xec1bbdfffec7406e"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-01-04 16:12:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffec7406e"},"type":"device_announced"}'

Koenkk, 2021-01-04:
`No converter available for 'HG06335' ` indicates it's paired correctly and recognised as a `HG06335`. `No converter..` is logged because these (not useful messages) are not converted to MQTT.

Kiachma, 2021-01-06:
Hi,

I'm experiencing similar issues with the Silvercrest(Lidl) Motion Sensor.
I twice got the sensor to show up in HA, once with battery and motion and once with battery and tamper. Battery was always null and motion/tamper unresponsive.

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/5777320/z2m.log)

freakinruben, 2021-01-12:
Hi,

I'm also experiencing problems with pairing two `HG06335`. `HG06336` does pair correctly.

```
Zigbee2MQTT:info  2021-01-12 21:29:39: Device '0xbc33acXXXXXXXXXX' joined
Zigbee2MQTT:info  2021-01-12 21:29:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acXXXXXXXXXX","ieee_address":"0xbc33acXXXXXXXXXX"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-01-12 21:29:39: Starting interview of '0xbc33acXXXXXXXXXX'
Zigbee2MQTT:info  2021-01-12 21:29:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acXXXXXXXXXX","ieee_address":"0xbc33acXXXXXXXXXX","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-01-12 21:29:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acXXXXXXXXXX"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-01-12 21:29:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acXXXXXXXXXX"},"type":"pairing"}'
Zigbee2MQTT:error 2021-01-12 21:31:41: Failed to interview '0xbc33acXXXXXXXXXX', device has not successfully been paired
Zigbee2MQTT:info  2021-01-12 21:31:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acXXXXXXXXXX","ieee_address":"0xbc33acXXXXXXXXXX","status":"failed"},"type":"device_interview"}'
```

tetsuo55, 2021-01-28:
Connecting the sensor to the hub once fixed this for me

Koenkk, 2021-01-28:
In case somebody still has issues with this, share the herdsman debug log when trying to pair.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

timoline, 2021-01-29:
Sorry, can't reproduce, it lookslike it just works fine over here

closed 2021-01-29 by Koenkk

tetsuo55, 2021-01-31:
@Koenkk does this help?

https://del.dog/movuvagath.txt

tetsuo55, 2021-01-31:
And here's the log for pairing it after doing the workaround of connecting it to the Lidl hub once.
https://del.dog/ukysugnico.txt

freakinruben, 2021-01-31:
For me it works now!

Koenkk, 2021-01-31:
@tetsuo55 looks like you have the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/5814, please keep a look there as we are trying to fix it.

FlexxFR, 2021-02-20:
Hi I'm also having similar problems.

`Info Device '0xbc33acfffe00f4f4' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"pairing"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"device_announced"}'
Warning Device '0xbc33acfffe00f4f4' left the network
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0xbc33acfffe00f4f4"},"type":"device_leave"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"device_removed"}'`

However, 1 out of 10 repairs provides some more logs, at least a (partial?) interview

`info  2021-02-20 10:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"pairing"}'
info  2021-02-20 10:45:38: Device '0xbc33acfffe00f4f4' joined
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4"},"type":"device_joined"}'
info  2021-02-20 10:45:38: Starting interview of '0xbc33acfffe00f4f4'
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4","status":"started"},"type":"device_interview"}'
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"device_connected"}'
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"pairing"}'
debug 2021-02-20 10:45:38: Device '0xbc33acfffe00f4f4' announced itself
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe00f4f4","ieee_address":"0xbc33acfffe00f4f4"},"type":"device_announce"}'
info  2021-02-20 10:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe00f4f4"},"type":"device_announced"}'
debug 2021-02-20 10:45:39: Received Zigbee message from '0xbc33acfffe00f4f4', type 'readResponse', cluster 'genBasic', data '{"modelId":"TY0202"}' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:39: Skipping message, definition is undefined and still interviewing
debug 2021-02-20 10:45:39: Received Zigbee message from '0x00158d0003532650', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:41: Received Zigbee message from '0xbc33acfffe00f4f4', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ1800_fcdjzz3s"}' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:41: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"_TZ1800_fcdjzz3s"}'
debug 2021-02-20 10:45:45: Received Zigbee message from '0xbc33acfffe00f4f4', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:45: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
debug 2021-02-20 10:45:47: Received Zigbee message from 'smartplug_washer', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,5]}' from endpoint 1 with groupID 0
info  2021-02-20 10:45:47: MQTT publish: topic 'zigbee2mqtt/smartplug_washer', payload '{"current":0,"energy":0.05,"linkquality":63,"power":0,"power_outage_memory":"on","state":"ON","voltage":236}'
debug 2021-02-20 10:45:48: Received Zigbee message from '0x00158d00039267b0', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:48: Received Zigbee message from '0xbc33acfffe00f4f4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:48: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
debug 2021-02-20 10:45:57: Received Zigbee message from '0xbc33acfffe00f4f4', type 'readResponse', cluster 'genBasic', data '{"appVersion":65}' from endpoint 1 with groupID 0
debug 2021-02-20 10:45:57: No converter available for 'HG06335' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":65}'
warn  2021-02-20 10:46:00: Device '0xbc33acfffe00f4f4' left the network`

FlexxFR, 2021-02-20:
Oh, just read https://github.com/Koenkk/zigbee2mqtt/issues/5814
Its now part of the edge-version?

@Koenkk , how can I know if / when this is part of the "stable" normal version? As Zigbee2mqtt novice I'm a bit intimidated to switch to the edge version if it's only a matter of days/weeks before it;s in the main branch.

cdrehner, 2021-02-21:
@FlexxFR I'd see myself a novice too, but thanks to your comment I just switched to edge and it seems to work so far!
It wasn't a big deal:

- take a snapshot (just in case)
- Backup config from: Supervisor → Dashboard → Zigbee2mqtt → Configuration to text file
- Stop addo nand disable auto start & watchdog for zigbee2mqtt
- reload your repositories (... in addon store)
- install zigbee2mqtt edge
- paste config from textfile to zigbee2mqtt edge, enable auto start & watchdog and start addon
- reboot host
- join motion sensor

FlexxFR, 2021-04-28:
Any update on this one?
It still does not work for me using the latest normal release.

@Koenkk , any plans of making this part of the normal release? Or will this be special forever?

Koenkk, 2021-04-29:
@FlexxFR should be in 1.18.1 already

VincentSC, 2021-12-21:
It looks like there are 2 types. Got 8 door sensors and 3 of them won't pair.

As I got 94 devices in the network, the herdsman log is like looking to F1 racing. So no idea how to solve it, as I'm not willing to remove all devices.

adekloet, 2021-12-22:
Hi @VincentSC If you have a tuya hub or something similar, please connect the not working 3 to that other Zigbee hub. After that see if they work with that hub. If they work, reset them and remove them from the test hub and try reconnecting them to Z2MQTT. They might just work after that. (I have no good explanation why, but I got 2 of them here at home working again)

VincentSC, 2021-12-22:
The Lidl hub is Tuya-based, so it makes sense. Unfortunately I don't have any, but my dad has. I see on https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#automatic-checking-for-available-updates that TuYa/Lidl/Silvercrest does not have public OTA

I'd like to use the logging, as I have many more experimental devices I like to make compatible with the code, but seemingly I need to have a separate test-network with a single device to avoid working with log-floods. But while I type this... I realize I actually have an old zigbee stick. :)