Koenkk/zigbee2mqtt #5545: Last seen setting from frontend crashes Z2M

GSzabados, 2021-01-04:
## What happened
Z2M Crashed when Last seen setting was enabled from Frontend's settings, used to be working before with Elapsed enabled as well. I believe it is related to one of the breaking changes from 2020.12.1, might be the Elapsed is the issue, but I am unable to set Last seen at all.
## What did you expect to happen
Enable two new state values.
## How to reproduce it (minimal and precise)
HA 2020.12.7 and Z2M 1.17.0, enable from Frontend's settings the Last seen first and then Elapsed.
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: 20190608

A log from the Z2M add-on:

```
UnhandledPromiseRejectionWarning: Error: Unsupported type '[object Object]'
    at formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:100:15)
    at Object.formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:215:33)
    at Controller.publishEntityState (/zigbee2mqtt-1.17.0/lib/controller.js:323:46)
    at publish (/zigbee2mqtt-1.17.0/lib/extension/receive.js:152:22)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.17.0/lib/extension/receive.js:168:13)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.17.0/lib/controller.js:381:44)
(node:240) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:240) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:240) UnhandledPromiseRejectionWarning: Error: Unsupported type '[object Object]'
    at formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:100:15)
    at Object.formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:215:33)
    at Controller.publishEntityState (/zigbee2mqtt-1.17.0/lib/controller.js:323:46)
    at publish (/zigbee2mqtt-1.17.0/lib/extension/receive.js:152:22)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.17.0/lib/extension/receive.js:168:13)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.17.0/lib/controller.js:381:44)
(node:240) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:240) UnhandledPromiseRejectionWarning: Error: Unsupported type '[object Object]'
    at formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:100:15)
    at Object.formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:215:33)
    at Controller.publishEntityState (/zigbee2mqtt-1.17.0/lib/controller.js:323:46)
    at publish (/zigbee2mqtt-1.17.0/lib/extension/receive.js:152:22)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.17.0/lib/extension/receive.js:168:13)

```

Koenkk, 2021-01-04:
Can you provide the debug log starting before setting the option till the crash?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

HB9VCT, 2021-01-05:
I can confirm the same problem. Here is the log:

Zigbee2MQTT:info  2021-01-05 10:35:47: MQTT publish: topic 'zigbee2mqtt/bridge/response/config/last_seen', payload '{"data":{"value":"epoch"},"status":"ok","transaction":"thto0-1"}'
Zigbee2MQTT:info  2021-01-05 10:35:47: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"1.17.0","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":{"transaction":"thto0-1","value":"epoch"},"legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00158d0004019509":{"friendly_name":"0x00158d0004019509"},"0x00158d00053dfd85":{"friendly_name":"0x00158d00053dfd85"},"0x00158d000572a365":{"friendly_name":"Motion1"},"0x842e14fffedfc04a":{"friendly_name":"Losonho relay"},"0xec1bbdfffe782154":{"friendly_name":"0xec1bbdfffe782154"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://core-mosquitto","user":"mqtt"},"passlist":[],"permit_join":false,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":false,"version":"1.17.0"}'
/zigbee2mqtt-1.17.0/lib/util/utils.js:100
        throw new Error(`Unsupported type '${type}'`);
        ^
Error: Unsupported type '[object Object]'
    at formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:100:15)
    at Object.formatDate (/zigbee2mqtt-1.17.0/lib/util/utils.js:215:33)
    at Frontend.onWebSocketConnection (/zigbee2mqtt-1.17.0/lib/extension/frontend.js:97:43)
    at WebSocketServer.emit (events.js:315:20)
    at /zigbee2mqtt-1.17.0/lib/extension/frontend.js:68:30
    at Frontend.authenticate (/zigbee2mqtt-1.17.0/lib/extension/frontend.js:61:9)
    at /zigbee2mqtt-1.17.0/lib/extension/frontend.js:66:18
    at WebSocketServer.completeUpgrade (/zigbee2mqtt-1.17.0/node_modules/ws/lib/websocket-server.js:330:5)
    at WebSocketServer.handleUpgrade (/zigbee2mqtt-1.17.0/node_modules/ws/lib/websocket-server.js:246:10)
    at Frontend.onUpgrade (/zigbee2mqtt-1.17.0/lib/extension/frontend.js:65:18)
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-05T09_39_08_997Z-debug.log
2021-01-05T10:39:11: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2021-01-05T10:39:11: PM2 log: App [npm:0] starting in -fork mode-
2021-01-05T10:39:11: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- advanced.last_seen should be string
- advanced.last_seen should be equal to one of the allowed values
If you don't know how to solve this, read https://www.zigbee2mqtt.io/information/configuration.html
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-05T09_39_14_156Z-debug.log
2021-01-05T10:39:15: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2021-01-05T10:39:15: PM2 log: App [npm:0] starting in -fork mode-
2021-01-05T10:39:15: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js

Koenkk, 2021-01-05:
Fixed, thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-05 by Koenkk

GSzabados, 2021-01-05:
@Koenkk, thank you very much! Happy to help!