Koenkk/zigbee2mqtt #5549: Failed to pair ROB_200_04-0

georgeavd, 2021-01-04:
## What happened
When pairing a ROB_200_007-0 the device is not (completely) paired and can't be used in zigbee2mqtt. Pairing of other devices is blocked.
The following debug message is generated:
"TypeError: Cannot read property 'replace' of undefined"

In the (by me added) debug-node (in Z2M admin) the following is listed for the paired device:
zigbee2mqtt/bridge/log : msg.payload : Object
object
message: array[10]
0: object
1: object
2: object
3: object
4: object
5: object
6: object
7: object
8: object
9: object
description: "-"
friendly_name: "0x000d6f0013exxxxx"
ieeeAddr: "0x000d6f0013exxxxx"
lastSeen: 1609776560128
networkAddress: zzzzz
type: "Unknown"
vendor: "-"
type: "devices"

The address is correct but the device is not listed in Z2M Admin.

In Zigbee2MqttAssistant the device is shown but without an image and not recognized as a ROBB Smart device.
## What did you expect to happen
Device being paired correctly.

## How to reproduce it (minimal and precise)
Remove device with Zigbee2MqttAssistant from the database and restart both the Zigbee2Mqtt- and Zigbee2mqttAssistant-containers. Yes I'm running on docker (RPi4) now.
Perform a factory reset of the device and try to pair again.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2538 and CC2530
Adapter firmware version: ZB 3.0

Koenkk, 2021-01-04:
> "TypeError: Cannot read property 'replace' of undefined"

Please provide the full debug log

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

georgeavd, 2021-01-04:
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url": "

De logfile.

debug 2021-01-04 18:17:50: Loaded state from file /app/data/state.json
info  2021-01-04 18:17:50: Logging to console and directory: '/app/data/log/2021-01-04.18-17-49' filename: log.txt
debug 2021-01-04 18:17:50: Removing old log directory '/app/data/log/2021-01-04.16-04-23'
info  2021-01-04 18:17:50: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
info  2021-01-04 18:17:50: Starting zigbee-herdsman (0.13.46)
debug 2021-01-04 18:17:50: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2021-01-04 18:17:52: zigbee-herdsman started
info  2021-01-04 18:17:52: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200327,"transportrev":2},"type":"zStack30x"}'
debug 2021-01-04 18:17:52: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b0009d80362","panID":6755}
info  2021-01-04 18:17:52: Currently 8 devices are joined:
info  2021-01-04 18:17:52: kantoor (0x00158d00046065d6): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-01-04 18:17:52: licht-1 (0x04cf8cdf3c7ccba6): GZCGQ01LM - Xiaomi MiJia light intensity sensor (EndDevice)
info  2021-01-04 18:17:52: badkamer (0x00158d0004641aeb): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-01-04 18:17:52: woonkamer_achter (0x00158d0004607445): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2021-01-04 18:17:52: cc2530_router (0x00124b001f320455): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
info  2021-01-04 18:17:52: woonkamer_eettafel (0xec1bbdfffead0ad0): ROB_200-004-0 - ROBB ZigBee AC phase-cut dimmer (Router)
info  2021-01-04 18:17:52: woonkamer_boekenkast (0xec1bbdfffead0e8f): ROB_200-004-0 - ROBB ZigBee AC phase-cut dimmer (Router)
info  2021-01-04 18:17:52: overkapping_spots (0xec1bbdfffead0e91): ROB_200-004-0 - ROBB ZigBee AC phase-cut dimmer (Router)
info  2021-01-04 18:17:52: Zigbee: disabling joining new devices.
info  2021-01-04 18:17:52: Connecting to MQTT server at mqtt://192.168.2.124
debug 2021-01-04 18:17:52: Using MQTT login with username: get-mqtt
info  2021-01-04 18:17:53: Connected to MQTT server
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/kantoor', payload '{"battery":91,"humidity":36.29,"pressure":1009.6,"temperature":20.85,"voltage":2985}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/licht-1', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"voltage":3100}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"battery":91,"humidity":38.76,"pressure":1009.9,"temperature":21.07,"voltage":2985}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/woonkamer_achter', payload '{"battery":100,"humidity":42.58,"pressure":1010.1,"temperature":21.35,"voltage":3005}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/cc2530_router', payload '{"led":true}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/woonkamer_eettafel', payload '{"brightness":120,"state":"OFF"}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/woonkamer_boekenkast', payload '{"brightness":75,"state":"ON"}'
info  2021-01-04 18:17:53: MQTT publish: topic 'zigbee2mqtt/overkapping_spots', payload '{"brightness":115,"state":"ON"}'
debug 2021-01-04 18:17:53: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"07cdc9d","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200327,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b0009d80362","panID":6755},"permit_join":false,"version":"1.17.0"}'
debug 2021-01-04 18:17:53: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00124b001f320455":{"friendly_name":"cc2530_router"},"0x00158d00046065d6":{"friendly_name":"kantoor"},"0x00158d0004607445":{"friendly_name":"woonkamer_achter"},"0x00158d0004641aeb":{"friendly_name":"badkamer"},"0x04cf8cdf3c7ccba6":{"friendly_name":"licht-1"},"0xec1bbdfffead0ad0":{"friendly_name":"woonkamer_eettafel"},"0xec1bbdfffead0e8f":{"friendly_ [... cut ...] ordinator"},{"description":"Aqara temperature, humidity and pressure sensor","friendly_name":"kantoor","ieeeAddr":"0x00158d00046065d6","lastSeen":1609778917949,"manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","modelID":"lumi.weather","networkAddress":64329,"powerSource":"Battery","type":"EndDevice","vendor":"Xiaomi"},{"dateCode":"20191118","description":"MiJia light intensity sensor","friendly_name":"licht-1","hardwareVersion":1,"ieeeAddr":"0x04cf8cdf3c7ccba6","lastSeen":1609

Koenkk, 2021-01-04:
I don't see the `TypeError: Cannot read property 'replace' of undefined` error in this log.

georgeavd, 2021-01-04:
Correct, that info is (as described) coming from a debug-node I've placed
in the Z2M Admin flow.
It's a response on the content of the message array it received from the
zigbee2mqtt-database.

As far as I can conclude from it is that the device type/id is not
recognized and therefore not entered correctly/completely initialized.
Thus ending up with an entry in the zigbee2mqtt database, but not being
entered in the configuration.yml file.
The result is that the device is probably not receiving the network key and
therefore not able to communicate.

georgeavd, 2021-01-24:
Hi Koen,

Just tested on my previous 'non docker' environment with zigbee2mqtt 1.15.0 and everything works as expected.
After an update to 1.17.0 I stil can connect devices to this version of 1.17.0.

Conclusion: there must be something wrong related to the docker configuration, but I still don't know what/why because other zigbee-devices (like BW-IS2 and lumi.sen_ill.mgl01) can join my zigbee docker setup.

What happens is that the newly joined devices are added to the configuration.yaml file but are not listed in the device list of the Z2M-admin web interface. And besides that when pressing a ON-button (1) on a ROB_200-007-0 LED goes on but no message is send to zigbee2mqtt at all and when pressing an OFF-button no LED on the switch is going at all.
As far as I can read from the debug-log the Vendor and type are not recognised.

For ROB_200-007-00:
{"description":"-","friendly_name":"0x000d6f00106a7923","ieeeAddr":"0x000d6f00106a7923","lastSeen":1611505190717,"networkAddress":25287,"type":"Unknown","vendor":"-"}

For Aqara cube:
{"description":"-","friendly_name":"0x000d6f00106a7923","ieeeAddr":"0x000d6f00106a7923","lastSeen":1611505190717,"networkAddress":25287,"type":"Unknown","vendor":"-"}

Which debug info would you like to receive from me, do you want to have a look via E.g. Teamviewer?

Regards George

Koenkk, 2021-01-24:
Can you provide the herdsman debug log when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

georgeavd, 2021-01-24:
Paring: 0x000d6f00106a7923 (ROB_200-007-0

[20210124 herdmann debug.zip](https://github.com/Koenkk/zigbee2mqtt/files/5863207/20210124.herdmann.debug.zip)

Koenkk, 2021-01-25:
You already stop z2m before interview fails, however it also seems the device does not respond. Any chance you can pair the device close to the coordinator?

georgeavd, 2021-01-25:
Pairing twice: 0x000d6f00106a7923 and 0x000d6f0013e8a2c8

[20210125 herdsman debug.txt.zip](https://github.com/Koenkk/zigbee2mqtt/files/5869439/20210125.herdsman.debug.txt.zip)

Koenkk, 2021-01-26:
Still the same error, next step would be to sniff the traffic when trying to pair: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

georgeavd, 2021-01-27:
Find attached the requested Wireshark capture (trying to join the 2 ROBB_200-007-0 device).

[20210127 zigbee sniffer.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5883250/20210127.zigbee.sniffer.pcapng.zip)

Koenkk, 2021-01-28:
Do you have another Zigbee network running on the same channel?  (looks like Hue?)

[image]

georgeavd, 2021-01-28:
We don't have Hue devices in our house and during joining the device was within 20cm of the controller.

But that does not explain why other devices can join my zigbee network.

Koenkk, 2021-01-29:
Maybe the joining fails because there is another network on the same channel. Since you have another coordinator (CC2530), can you try setting that up on a different channel (e.g. 15). See https://www.zigbee2mqtt.io/information/configuration.html on how to change the channel.

georgeavd, 2021-02-08:
Do you mean on the same Pi and setup and only exchange the cc2538 with cc2530 and change the channel?

When testing with a CC2530 on my previous domotica RPi I was able to join both devices to zigbee2mqtt.
Can this problem be caused by the cc2538? Or can that be out ruled?

georgeavd, 2021-02-08:
I came across this info [https://github.com/jethome-ru/zigbee-firmware/issues/2](url).
So I'm now looking for a method to perform a firmware upgrade of my CC2538+CC2592 USB-stick without the need for any soldering.

georgeavd, 2021-02-13:
I've V1 of the CC2538+CC2592 USB-stick so I first have to order a JTAG-programmer (at AliExpress) because this HW-version can't be updated via USB.

georgeavd, 2021-02-26:
Yesterday I received the JTAG-programmer and today I've flashed the CC2538+CC2592 USB-stick to firmware version 20201010.
After that I was able to both the ROBB_200-007-0 devices and my Xiaomi MFKZQ01LM (cube) as well.

So the issue was related to [https://github.com/jethome-ru/zigbee-firmware/issues/2](url) and solved bij the firmware upgrade of the cc2538.

Thanks for your patience and keep up the good work.

closed 2021-02-26 by Koenkk