Koenkk/zigbee2mqtt #5551: ERROR: Publish 'set' 'system_mode' to 'Thermostat living room' failed: 'Error: TRV system mode heat is not recognized.' on  Siterwell  GS361A-H04

lorek123, 2021-01-04:
## What happened
When I am trying to set up the system mode to heat it fails with error:
```DEBUG: Received MQTT message on 'zigbee2mqtt/Thermostat living room/set' with data '{"system_mode":"heat"}'
DEBUG: Publishing 'set' 'system_mode' to 'Thermostat living room'
ERROR: Publish 'set' 'system_mode' to 'Thermostat living room' failed: 'Error: TRV system mode heat is not recognized.'
DEBUG: Error: TRV system mode heat is not recognized. at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3038:23) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:224:52) at Controller.callExtensionMethod (/app/lib/controller.js:381:44) at processTicksAndRejections (internal/process/task_queues.js:97:5)
```
## What did you expect to happen
successful system mode change
## How to reproduce it (minimal and precise)
change system mode to heat for Siterwell  GS361A-H04
## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: CC2652 https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/
Adapter firmware version: 20201026

michu89, 2021-01-04:
I have exactly the same problem, only difference is that I use CC2531 with zStack12 coordinator firmware into it.

```
Zigbee2MQTT:debug 2021-01-05 00:26:43: Received MQTT message on 'zigbee2mqtt/glowica_sypialnia/set/system_mode' with data 'heat'
Zigbee2MQTT:debug 2021-01-05 00:26:43: Publishing 'set' 'system_mode' to 'glowica_sypialnia'
Zigbee2MQTT:error 2021-01-05 00:26:43: Publish 'set' 'system_mode' to 'glowica_sypialnia' failed: 'Error: TRV system mode heat is not recognized.'
Zigbee2MQTT:debug 2021-01-05 00:26:43: Error: TRV system mode heat is not recognized.
    at Object.convertSet (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:3038:23)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.17.0/lib/extension/publish.js:224:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.17.0/lib/controller.js:381:44)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
Zigbee2MQTT:info  2021-01-05 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'system_mode' to 'glowica_sypialnia' failed: 'Error: TRV system mode heat is not recognized.'","meta":{"friendly_name":"glowica_sypialnia"},"type":"zigbee_publish_error"}'
```

Koenkk, 2021-01-05:
I don't know this device, but heat is not a supported mode, supported are: off, auto, manual, comfort, eco, boost, complex. 

Can you check if setting `manual` turns off the thermostat?

lorek123, 2021-01-05:
Log from Web UI below, what is interesting changing mode to comfort started second part of the initial configuration. eco, boost, complex didn't show any visible changes.
```
DEBUG: Received MQTT message on 'zigbee2mqtt/Thermostat small room/set' with data '{"system_mode": "manual"}'
DEBUG: Publishing 'set' 'system_mode' to 'Thermostat small room'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871730517,"linkquality":72,"local_temperature":25.3,"preset":"manual","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":1,"transid":114}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871730625,"linkquality":72,"local_temperature":25.3,"preset":"manual","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":1,"transid":115}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871731330,"linkquality":72,"local_temperature":25.3,"preset":"manual","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received MQTT message on 'zigbee2mqtt/Thermostat small room/set' with data '{"system_mode": "comfort"}'
DEBUG: Publishing 'set' 'system_mode' to 'Thermostat small room'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":5}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871775710,"linkquality":78,"local_temperature":25.3,"preset":"comfort","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":1,"transid":116}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871775825,"linkquality":78,"local_temperature":25.3,"preset":"comfort","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,250],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":1,"transid":117}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871776752,"linkquality":78,"local_temperature":25.3,"preset":"comfort","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":1,"transid":118}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871781709,"linkquality":78,"local_temperature":25.3,"preset":"comfort","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":1,"transid":119}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":48,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871792262,"linkquality":81,"local_temperature":25.3,"preset":"manual","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received MQTT message on 'zigbee2mqtt/Thermostat small room/set' with data '{"system_mode": "eco"}'
DEBUG: Publishing 'set' 'system_mode' to 'Thermostat small room'
DEBUG: Received Zigbee message from 'Thermostat small room', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[4],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":6}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":45,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871907785,"linkquality":78,"local_temperature":25.3,"preset":"eco","valve_detection":" [... cut ...] m endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/Thermostat small room', payload '{"away_mode":"OFF","battery":45,"child_lock":"UNLOCKED","current_heating_setpoint":25,"last_seen":1609871982477,"linkquality":78,"local_temperature":25.3,"preset":"complex","valve_detection":"ON","window_detection":"ON"}'
DEBUG: Received MQTT message on 'zigbee2mqtt/Thermostat small room/set' with data '{"system_mode": "manual"}'
DEBUG: Publishing 'set' 'system_mode' to 'Thermostat small room'
```

Systm21, 2021-01-07:
There are some more Problems with the Siterwell Devices. 
1. You cannot switch the window detection from the GUI, because there is no converter (but publishing over MQTT works fine)
2. You cannot switch the Valve Detection From the GUI, because there is no converter (but publishing over MQTT works fine)
3. You cannot get a State/Running State in the GUI (because there is no converter)
4. You cannot get the valve position from the device, although you can switch it.

[image]

Would be really great, if Z2M wouls Support it 😍

netweaver1970, 2021-01-07:
I second that, as I have 5 of them. Currently too many other projects at the same time but one day I'll start to get to the ultimate goal of multi-zone PID controlled valves with central boiler control. Sort of cheap Evohome. 
Would be nice if the GUI and back-end on Z2M are also working.

Thanks.

> There are some more Problems with the Siterwell Devices.
> 
> 1. You cannot switch the window detection from the GUI, because there is no converter (but publishing over MQTT works fine)
> 2. You cannot switch the Valve Detection From the GUI, because there is no converter (but publishing over MQTT works fine)
> 3. You cannot get a State/Running State in the GUI (because there is no converter)
> 4. You cannot get the valve position from the device, although you can switch it.
> 
> [image]
> 
> Would be really great, if Z2M supports it

michu89, 2021-01-08:
My logs are below. I tried all the options. But even though "preset" changes, "system_mode" still stays off. Before update I was able to change "system_mode". When I was upgrading Z2M this TRV was in "off" mode, rest of them were in "auto" and they stood in it. I haven't touched them, only changed SP of heating, nothing more. One which was turned off now I can't turn into "on", even though on TRV I see that it receives correct mode preset for "off" (sets lower temp. and no icon on TRV) and "manual"  (hand icon on TRV). I can't move it into "auto" (icon with clock) using any of commands shared higher. If I change temperature SP, or any other thing it is correctly read by HA, but if I turn it into any from three modes I receive nothing. Example log from other TRV which was on when updating Z2M (reporting system_mode as auto and preset as "schedule" which corresponds to the timer icon on it I think):
`Zigbee2MQTT:debug 2021-01-08 18:42:06: Received MQTT message on 'zigbee2mqtt/glowica_gabinet/set/current_heating_setpoint' with data '21.5'
Zigbee2MQTT:debug 2021-01-08 18:42:06: Publishing 'set' 'current_heating_setpoint' to 'glowica_gabinet'
Zigbee2MQTT:debug 2021-01-08 18:42:09: Received Zigbee message from 'glowica_gabinet', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,215],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":222}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-08 18:42:09: MQTT publish: topic 'zigbee2mqtt/glowica_gabinet', payload '{"away_mode":"OFF","battery":63,"child_lock":"LOCKED","current_heating_setpoint":21.5,"linkquality":105,"local_temperature":22.6,"preset":"schedule",**"system_mode":"auto"**,"valve_detection":"ON","window_detection":"ON"}'
Zigbee2MQTT:debug 2021-01-08 18:42:09: Received Zigbee message from 'glowica_gabinet', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,215],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":6,"transid":229}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-08 18:42:09: MQTT publish: topic 'zigbee2mqtt/glowica_gabinet', payload '{"away_mode":"OFF","battery":63,"child_lock":"LOCKED","current_heating_setpoint":21.5,"linkquality":105,"local_temperature":22.6,"preset":"schedule","system_mode":"auto","valve_detection":"ON","window_detection":"ON"}'`

TRV which is wrong (was off during update):
Off:
`Zigbee2MQTT:debug 2021-01-06 12:32:41: Received MQTT message on 'zigbee2mqtt/glowica_sypialnia/set' with data '{"system_mode":"off"}'
Zigbee2MQTT:debug 2021-01-06 12:32:41: Publishing 'set' 'system_mode' to 'glowica_sypialnia'`

`Zigbee2MQTT:debug 2021-01-06 12:34:13: Received MQTT message on 'zigbee2mqtt/glowica_sypialnia/set' with data '{"system_mode":"auto"}'
Zigbee2MQTT:debug 2021-01-06 12:34:13: Publishing 'set' 'system_mode' to 'glowica_sypialnia'
Zigbee2MQTT:debug 2021-01-06 12:34:13: Received Zigbee message from 'glowica_sypialnia', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":174}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-06 12:34:13: MQTT publish: topic 'zigbee2mqtt/glowica_sypialnia', payload '{"away_mode":"OFF","battery":72,"child_lock":"LOCKED","current_heating_setpoint":22,"linkquality":28,"local_temperature":21.8,"preset":"schedule","system_mode":"off","valve_detection":"ON","window_detection":"ON"}'
Zigbee2MQTT:debug 2021-01-06 12:34:14: Received Zigbee message from 'glowica_sypialnia', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":5,"transid":178}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-06 12:34:14: MQTT publish: topic 'zigbee2mqtt/glowica_sypialnia', payload '{"away_mode":"OFF","battery":72,"child_lock":"LOCKED","current_heating_setpoint":22,"linkquality":28,"local_temperature":21.8,"preset":"schedule","system_mode":"off","valve_detection":"ON","window_detection":"ON"}'`

Auto:
`Zigbee2MQTT:debug 2021-01-08 18:36:31: Received MQTT message on 'zigbee2mqtt/glowica_sypialnia/set' with data '{"system_mode":"auto"}'
Zigbee2MQTT:debug 2021-01-08 18:36:31: Publishing 'set' 'system_mode' to 'glowica_sypialnia'
Zigbee2MQTT:debug 2021-01-08 18:36:34: Received Zigbee message from 'glowica_sypialnia', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":0,"transid":218}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-08 18:36:34: MQTT publish: topic 'zigbee2mqtt/glowica_sypialnia', payload '{"away_mode":"OFF","battery":72,"child_lock":"LOCKED","current_heating_setpoint":20,"linkquality":60,"local_temperature":20.8,"preset":"schedule","system_mode":"off","valve_detection":"ON","window_detection":"ON"}'
Zigbee2MQTT:debug 2021-01-08 18:36:34: Received Zigbee message from 'glowica_sypialnia', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":4,"fn":0,"status":6,"transid":4}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-08 18:36:34: MQTT publish: topic 'zigbee2mqtt/glowica_sypialnia', payload '{"away_mode":"OFF","battery":72,"child_lock":"LOCKED","current_heating_setpoint":20,"linkquality":60,"local_temperature":20.8,"preset":"schedule","system_mode":"off","valve_detection":"ON","window_detection":"ON"}'
[... cut ...]
Zigbee2MQTT:debug 2021-01-08 18:45:28: Received Zigbee message from 'glowica_sypialnia', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,220],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":6,"transid":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-01-08 18:45:28: MQTT publish: topic 'zigbee2mqtt/glowica_sypialnia', payload '{"away_mode":"OFF","battery":72,"child_lock":"LOCKED","current_heating_setpoint":22,"linkquality":63,"local_temperature":20.8,"preset":"complex","system_mode":"off","valve_detection":"ON","window_detection":"ON"}'`

closed 2021-02-15 by github-actions[bot]

dawsonweb, 2021-03-03:
When attempting to set system_mode in z2m in HA I get error:
"Publish 'set' 'system_mode' to 'name or trv" failed: 'Error: TRV system mode heat is not recognized.'"
[image]
z2m[ devices page](https://www.zigbee2mqtt.io/devices/GS361A-H04.html) definitely shows that as a supported mode.