Koenkk/zigbee2mqtt #5552: Error: Adapter doesn't support LED

LM1LC3N7, 2021-01-04:
## What happened
I have an error at startup:

```bash
> node index.js

Zigbee2MQTT:info  2021-01-04 17:01:49: Logging to console and directory: '/app/data/log/2021-01-04.17-01-48' filename: log.txt
Zigbee2MQTT:info  2021-01-04 17:01:49: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Zigbee2MQTT:info  2021-01-04 17:01:49: Starting zigbee-herdsman (0.13.46)

Zigbee2MQTT:info  2021-01-04 17:02:00: zigbee-herdsman started
Zigbee2MQTT:info  2021-01-04 17:02:00: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:error 2021-01-04 17:02:00: Failed to start zigbee
Zigbee2MQTT:error 2021-01-04 17:02:00: Exiting...
Zigbee2MQTT:error 2021-01-04 17:02:00: Error: Adapter doesn't support LED
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:315:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:24:58)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-01-04T17_02_00_501Z-debug.log
```

## What did you expect to happen
I bought this adapter: https://slae.sh/projects/cc2652/
Because it is supported officially and does not need additional tools to flash it.

When I set up the container, it starts and stop after few seconds.

## How to reproduce it (minimal and precise)
1. Use a Raspberry Pi 4
2. Use the same stick cc2652
3. Create the container using this USB device, connected on `ttyUSB0`.

## Debug info
Zigbee2MQTT version: 1.6.1
Adapter hardware: CC2652RB 
Adapter firmware version: Last official (flashed today)

Koenkk, 2021-01-04:
Remove `disabled_led` from your configuration.yaml (for this adapter we don't support controlling the LED)

LM1LC3N7, 2021-01-07:
Hello @Koenkk,

Thank you very much, this was the source cause.
As I did not deployed the container myself, but through a domotic solution, I was not aware of the default configuration.

closed 2021-01-07 by LM1LC3N7