Koenkk/zigbee2mqtt #5553: Hue Strip LCL001 cannot finish interview

noohi, 2021-01-04:
Hi,

same as  #3973, but that one was closed with not 100% clear confirmation that it is now supported.

Zigbee2MQTT version: latest dev
Adapter hardware: CC1352P-2
Works on Hue app, got updated to latest firmware as well.

Here it is with herdsman log to cut to the chase :).
Let me know if I can help in any way.
Thanks.

`Zigbee2MQTT:info  2021-01-04 18:02:45: Starting interview of '0x0017880108d6f624',
Zigbee2MQTT:info  2021-01-04 18:02:45: MQTT publish: topic 'mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880108d6f624","ieee_address":"0x0017880108d6f624","status":"started"},"type":"device_interview"}',
2021-01-04T17:02:45.522Z zigbee-herdsman:controller:device:log Interview - start device '0x0017880108d6f624',
Zigbee2MQTT:info  2021-01-04 18:02:45: MQTT publish: topic 'mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880108d6f624"},"type":"pairing"}',
2021-01-04T17:02:55.711Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9290,"extaddr":"0x0017880108d6f624","parentaddr":0},
2021-01-04T17:02:55.711Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:02:55.711Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:02:55.711Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:02:55.713Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:02:57.935Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":34626,"extaddr":"0x0017880108d6f624","parentaddr":65164},
2021-01-04T17:02:57.935Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:02:57.935Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:02:57.935Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:02:57.937Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:03:05.688Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9290,"extaddr":"0x0017880108d6f624","parentaddr":0},
2021-01-04T17:03:05.688Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:03:05.688Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:03:05.688Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:03:05.690Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:03:08.725Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 1,
2021-01-04T17:03:32.170Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 2,
2021-01-04T17:03:37.783Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9290,"extaddr":"0x0017880108d6f624","parentaddr":0},
2021-01-04T17:03:37.783Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:03:37.783Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:03:37.783Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:03:55.223Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 3,
2021-01-04T17:04:13.448Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9290,"extaddr":"0x0017880108d6f624","parentaddr":0},
2021-01-04T17:04:13.448Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:04:13.448Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:04:13.448Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:04:18.098Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":1175,"extaddr":"0x0017880108d6f624","parentaddr":32957},
2021-01-04T17:04:18.098Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:04:18.098Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:04:18.098Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:04:18.100Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:04:18.567Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 4,
2021-01-04T17:04:39.165Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":58123,"extaddr":"0x0017880108d6f624","parentaddr":42757},
2021-01-04T17:04:39.165Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:04:39.165Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:04:39.165Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:04:39.167Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:04:41.633Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 5,
2021-01-04T17:04:51.590Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":445,"extaddr":"0x0017880108d6f624","parentaddr":7695},
2021-01-04T17:04:51.590Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:04:51.590Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:04:51.590Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:04:51.591Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:04:51.591Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":445,"extaddr":"0x0017880108d6f624","parentaddr":7695},
2021-01-04T17:04:51.591Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:04:51.591Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:04:51.591Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:05:04.087Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":58123,"extaddr":"0x0017880108d6f624","parentaddr":42757},
2021-01-04T17:05:04.087Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:05:04.087Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:05:04.087Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' is already in database with different networkAddress, updating networkAddress,
2021-01-04T17:05:04.088Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true',
2021-01-04T17:05:04.748Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x0017880108d6f624', attempt 6,
2021-01-04T17:05:04.748Z zigbee-herdsman:controller:device:log Interview - failed for device '0x0017880108d6f624' with error 'Error: Interview failed because can not get node descriptor ('0x0017880108d6f624'),
2021-01-04T17:05:04.750Z zigbee-herdsman:controller:error Interview failed for '0x0017880108d6f624 with error 'Error: Interview failed because can not get node descriptor ('0x0017880108d6f624')',
Zigbee2MQTT:error 2021-01-04 18:05:04: Failed to interview '0x0017880108d6f624', device has not successfully been paired,
[... cut ...]
Zigbee2MQTT:info  2021-01-04 18:05:06: MQTT publish: topic 'mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880108d6f624"},"type":"pairing"}',
2021-01-04T17:05:09.344Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9290,"extaddr":"0x0017880108d6f624","parentaddr":0},
2021-01-04T17:05:09.344Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' joined,
2021-01-04T17:05:09.344Z zigbee-herdsman:controller:log Device '0x0017880108d6f624' accepted by handler,
2021-01-04T17:05:09.344Z zigbee-herdsman:controller:log Not interviewing '0x0017880108d6f624', completed 'false', in progress 'true'`

Koenkk, 2021-01-04:
- Check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
- Does it never complete? (even not after a few minutes?)

noohi, 2021-01-04:
Regarding the basics, I have the older Hue Strips (LST002), they work fine, also 116 other Zigbee devices.
But just got this new Hue Strip the other day and no luck, it just loops interview start.
Then found the similar report, I wonder if that user ever confirmed it working or he moved onto another gateway/device.

Tried also moving it closed to the gateway, resetting with Hue Switch (1+0 pressed), unplug, reboot Z2M...nothing seems to help.
If Hue gateway wasn't pairing to it, I would thought it's broken.

Koenkk, 2021-01-04:
In #3973 the interview is successful so looks good. Would you be able to sniff the traffic of the pairing attempt? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

noohi, 2021-01-05:
OK, got some of the new Lidl devices, they can't join either.
So it's some kind of a hard limit at 116 devices.

What would be my options, can another adapter be added to work in parallel, is this a bug or a feature?

Only thing that comes to mind is to spin up another docker container with another adapter and continue there from 100 onwards, would that work or would they clash?

Using latest firmware and build, I have all kinds of adapters, currently on CC1352P2.

Koenkk, 2021-01-05:
I don't see a reason for a limit at 116 devices, the routers should still be able to join other devices. Since you have multiple adapters I suggest to sniff the traffic of this: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html (you can send the network key in private to me via mail: koenkanters94@.gmail.com, remove . before gmail.com)

noohi, 2021-01-07:
Thanks, I doubt that zigbee protocol is the issue, as it worked before this amount of devices.
I have "gave up" and split the workload, also removed Osram devices from the network as I read that they can cause issues, so will debug and see how it goes.
Until I know more, will rename and close this one.

closed 2021-01-07 by noohi