Koenkk/zigbee2mqtt #5559: Install Zigbeemqtt in a virtual enviroment. Error 134 after "npm ci" comma

fjcerdan27, 2021-01-04:
## What happened
Hello,

 I'm installing zigbee2mqtt2 in a virtual enviroment on a raspberry pi. I have an error after run "npm ci" command. 
The number error is 134.
 
The message in log file is:
1726 error errno 134
1727 error @serialport/bindings@9.0.3 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`

Thanks in advance.

## Debug info
Zigbee2MQTT version: Last

Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

Maurizion, 2021-01-12:
Same error here : 

1714 silly lifecycle   'prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild' ]
1715 silly lifecycle @serialport/bindings@9.0.3~install: Returned: code: 134  signal: null
1716 info lifecycle @serialport/bindings@9.0.3~install: Failed to exec install script
1717 verbose teardown shutting down workers.
1718 info teardown Done in 0.008s
1719 verbose stack Error: @serialport/bindings@9.0.3 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
1719 verbose stack Exit status 134
1719 verbose stack     at EventEmitter.<anonymous> (/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
1719 verbose stack     at EventEmitter.emit (events.js:189:13)
1719 verbose stack     at ChildProcess.<anonymous> (/opt/zigbee2mqtt/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
1719 verbose stack     at ChildProcess.emit (events.js:189:13)
1719 verbose stack     at maybeClose (internal/child_process.js:970:16)
1719 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
1720 verbose pkgid @serialport/bindings@9.0.3
1721 verbose cwd /opt/zigbee2mqtt
1722 verbose Linux 4.19.97-v7+
1723 verbose argv "/opt/zigbee2mqtt/bin/node" "/opt/zigbee2mqtt/bin/npm" "ci"
1724 verbose node v10.15.1
1725 verbose npm  v6.4.1
1726 error code ELIFECYCLE
1727 error errno 134
1728 error @serialport/bindings@9.0.3 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
1728 error Exit status 134
1729 error Failed at the @serialport/bindings@9.0.3 install script.
1729 error This is probably not a problem with npm. There is likely additional logging output above.
1730 verbose exit [ 134, true ]

Gcarlo3053, 2021-02-18:
13 feb 2021
Same error !!
python 3.8.6 
Raspbian:
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian

I need  urgent support
Thanks in advance.

Maurizion, 2021-02-19:
I found a workaround : install without vitual env 👍 

# Setup Node.js repository
sudo curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -

# NOTE 1: If you see the message below please follow: https://gist.github.com/Koenkk/11fe6d4845f5275a2a8791d04ea223cb.
# ## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.
# IMPORTANT: In this case instead of the apt-get install mentioned below; do: sudo apt-get install -y git make g++ gcc

# NOTE 2: On x86, Node.js 10 may not work. It's recommended to install an unofficial Node.js 12 build which can be found here: https://unofficial-builds.nodejs.org/download/release/ (e.g. v12.16.3)

# Install Node.js;
sudo apt-get install -y nodejs git make g++ gcc

# Verify that the correct nodejs and npm (automatically installed with nodejs)
# version has been installed
node --version  # Should output v12.X or v10.X
npm --version  # Should output 6.X

# Clone Zigbee2MQTT repository
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt

# Install dependencies (as user "pi")
cd /opt/zigbee2mqtt
npm ci --production

Gcarlo3053, 2021-02-24:
> i found a workaround : install without virtual env .
> follow this link (installing ) : https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

:-)
👍
thank you ! all solved with your suggestion. I am left with doubt of why it does not work in the virtual environment and why this problem is not reported in the guide. greetings and thanks.

closed 2021-04-03 by github-actions[bot]