Koenkk/zigbee2mqtt #5560: Supported devices not recognized after update.

rodrigorsdev, 2021-01-04:
## What happened
Hello,

I recently updated my home assistant and zigbee2mqtt.

Multiple supported devices are no longer recognized like HEIMAN HS1CG-M, TuYa SNTZ007 and others.

Logs: https://hastebin.com/axewixatig.apache

How can I  add this devices, listed on supported devices, without open a request?

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Thanks.

Koenkk, 2021-01-05:
Can you share your data/database.db?

rodrigorsdev, 2021-01-05:
Sure,

https://hastebin.com/uxukoxuqum.json

Koenkk, 2021-01-05:
I'm not sure what/how it happend but all modelIDs are gone from your database (for the unsupported device). I'm afraid you have to repair those (though it would be nice if there is some way to reproduce this so I can fix it)

rodrigorsdev, 2021-01-05:
How can I fix this? 

I created a new Home Assistant server and installed zigbee2mqtt, I still have the old server and zigbee2mqtt database.

Koenkk, 2021-01-05:
Can you copy over the database of your old server to the new one (make a backup of the new one first, and stop z2m before replacing)

rodrigorsdev, 2021-01-05:
The devices appeared, but with unknown status and does not work.

[image]

Koenkk, 2021-01-06:
Can you provide the herdsman debug log now when triggering it?

rodrigorsdev, 2021-01-06:
it's here

https://hastebin.com/edakogutis.rust

Koenkk, 2021-01-07:
Did you click it during this log (this is only the first 4 seconds of zigbee2mqtt)

rodrigorsdev, 2021-01-07:
It's the first 4 seconds: https://pastebin.com/441Veuz

sneakythr0ws, 2021-01-08:
I have the same problem with Tuya TS0601 'Interview failed'

Koenkk, 2021-01-08:
@rodrigorsdev  getting : 
[image]

rodrigorsdev, 2021-01-08:
I´m sorry

https://hastebin.com/hupojidije.yaml

rodrigorsdev, 2021-01-08:
there is a little more

https://hastebin.com/ejumuluwuk.yaml

Koenkk, 2021-01-08:
Logs are still short. If you are running the HA addon I suggest just grab the logs by executing `docker logs -f [CONTAINER_ID]` on the z2m container (use ssh to login)

rodrigorsdev, 2021-01-08:
how can I do that in hassio?

Koenkk, 2021-01-08:
@rodrigorsdev there are some tutorials available I see; https://www.google.com/search?q=ssh+into+hassio&oq=ssh+into+hassio&aqs=chrome..69i57j0l2j0i22i30l2.2080j0j7&sourceid=chrome&ie=UTF-8

rodrigorsdev, 2021-01-08:
I know how to connect, I don't know how to list the docker containers

[image]

Koenkk, 2021-01-08:
Does it work with: `sudo docker ps` ?

rodrigorsdev, 2021-01-08:
No, 

[image]

rodrigorsdev, 2021-01-11:
I reinstalled zigbee2mqtt and when adding some sensors it worked normally, however the xiaomi sensors are not found and does not generate any record in the log.

Koenkk, 2021-01-11:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

rodrigorsdev, 2021-01-12:
I flashed the CC2531 coordinator and worked.

Thanks for help.

closed 2021-01-12 by rodrigorsdev