Koenkk/zigbee2mqtt #5561: Existing device MEG5113-0300/MEG5165-0000 has new endpoint id

Moony83, 2021-01-04:
While adding a couple of Merten wiser Shutter controls, herdsman ran into a Type Error: "**TypeError: Cannot read property 'bind' of undefined**".
When examining the paring process with herdsman on debug, it seems that the endpoint id is 21 instead of 1. 
(See https://pastebin.com/pdUMskcp)

devices.js has this model currently configured with "**const endpoint1 = device.getEndpoint(1);**"
I suggest mending devices.js and including a potential endpoint id of "21"

Koenkk, 2021-01-05:
Does `const endpoint1 = device.getEndpoint(1) || device.getEndpoint(21);` fix it?

Moony83, 2021-01-05:
Actually i was a bit wrong there, it seems like Endpoint 5 is the one to go. At least partly. 
I managed to integrate with `const endpoint1 = device.getEndpoint(1) || device.getEndpoint(5);`

The device(s) now show up correctly on MQTT and report their linkstate and position.
However, trying to set their position does not make them move and the logfile still notes a couple of "no converter" lines:
https://pastebin.com/JEB0eiaJ

Koenkk, 2021-01-05:
Thanks, fixed the code.

These no converter messages are normal (it just indicates these won't be converter to MQTT, because these messages are not interesting).

closed 2021-01-05 by Koenkk