Koenkk/zigbee2mqtt #5562: Failure to configure Sonoff SNZB-02

nanobra1n, 2021-01-05:
## What happened
Every Z2M startup I get 3 errors for this device such as:
```
Zigbee2MQTT:error 2021-01-04 20:00:36: Failed to configure 'Sonoff LR', attempt 3 (Error: Bind 0x00124b001b783e4a/1 msTemperatureMeasurement from '0x00124b001ca5cb1c/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```
I tried the manual reconfigure through the frontend several times, pressing the sensor button frequently to keep it awake, but to no avail. I removed the device and repaired - still errors. 
`database.db` entry:
```
{"id":24,"type":"EndDevice","ieeeAddr":"0x00124b001b783e4a","nwkAddr":34511,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200526"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2049}},"msRelativeHumidity":{"attributes":{"measuredValue":8166}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":152}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5cb1c","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5cb1c","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca5cb1c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200526","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1609805235608}
```
Otherwise the device works without problems.

## What did you expect to happen
No errors
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.01
Adapter hardware:  LAUNCHXL-CC1352P-2 
Adapter firmware version: 20201113

Koenkk, 2021-01-05:
Does it correctly report temperature, battery and humidity?

nanobra1n, 2021-01-05:
Temperature and humidity agrees with another sensor, battery is reasonable as well (80%)

Koenkk, 2021-01-05:
Good, fixed the issue.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sjorge, 2021-01-06:
Copying this here... 

sjorge:
> @Koenkk I believe this is actually due to an interviewing failure...
>
> These devices only seem to accept the binds briefly after interviewing. They also don't leave the network.
>
> Fresh device: pairs OK
> Remove device: failure
> Reset old device: you hit the error this fixes
> Force remove device: fails to leave, so you get the unknown device is ending stuff warning
> Reset old device: pairs OK configure runs.
>
> Not sure yet what the difference is for z2m with or without a database... interview vs rejoin maybe? I did some wireshark > poking but got nowhere.

Koenkk:
> @sjorge I see, but it seems that in someway devices don't always need this as the devices worked correctly without this: Koenkk/zigbee2mqtt#5562 (lets continue the discussion there)

sjorge, 2021-01-06:
Correct, since the device did not properly 'leave' it still retains the configuration.

With permit join disabled: If you force remove it, take out the battery and put it back, it will keep on sending reports as normal but z2m will ignore it.
With permit join enabled: If you force remove it, take out the battery and put it back, z2m will interview the device and setup the reporting, configure works fine at this state.

I think configure only works after poking the genBasic cluster like during interview.

ianmxyz, 2021-01-25:
I'm not sure this is the same issue but I'm having problems when the battery level is reported below 80%/2900mV. Replacing the battery doesn't update the device. Force remove and pressing the pairing button finds it again, changing the friendly name back gets everything working again. Bit of a pain though, I have a device which keeps my lodger's room warm and I can't get in without making an arrangment to press the pairing button.

nvodenicharski99, 2021-01-28:
Hi,
my SNZB-02 is buggy as well. All values are "N/A".
Here are the errors I see in the logs:

```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00124b00226c75dd"},"error":"Failed to configure (Bind 0x00124b00226c75dd/1 msTemperatureMeasurement from '0x00124b0018e31a7f/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"6ob63-1"}'
INFO: Configuring '0x00124b00226c75dd'
ERROR: Failed to configure '0x00124b00226c75dd', attempt 2 (Error: Bind 0x00124b00226c75dd/1 msTemperatureMeasurement from '0x00124b0018e31a7f/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7))
```

Any idea?
(v1.17.0.1)

sjorge, 2021-01-28:
- force remove the device
- enable pairing
- remove/wait 3 sec/insert the batter

It should, hopefully rejoin and pair successfully this time.

nvodenicharski99, 2021-01-28:
I did it like 10 times. 
Let me try one more :)

sjorge, 2021-01-28:
Note the force remove, doing a normal remove won't work... as the device ignores the leave request.

nvodenicharski99, 2021-01-28:
yes, I always use the force ;)

nvodenicharski99, 2021-01-28:
A bit better, but only the link quiality.
{
    "linkquality": 44
}

nvodenicharski99, 2021-01-28:
Here are the joning logs:
`INFO: Device '0x00124b00226c75dd' joined
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00226c75dd","ieee_address":"0x00124b00226c75dd"},"type":"device_joined"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00226c75dd","ieee_address":"0x00124b00226c75dd","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b00226c75dd"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00226c75dd","ieee_address":"0x00124b00226c75dd"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b00226c75dd"},"type":"device_announced"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0x00124b00226c75dd/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00124b00226c75dd/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00226c75dd"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor (SNZB-02)","name":"0x00124b00226c75dd","sw_version":"Zigbee2MQTT 1.17.0"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x00124b00226c75dd","name":"0x00124b00226c75dd battery","state_topic":"zigbee2mqtt/0x00124b00226c75dd","unique_id":"0x00124b00226c75dd_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0x00124b00226c75dd/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00124b00226c75dd/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00226c75dd"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor (SNZB-02)","name":"0x00124b00226c75dd","sw_version":"Zigbee2MQTT 1.17.0"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/0x00124b00226c75dd","name":"0x00124b00226c75dd temperature","state_topic":"zigbee2mqtt/0x00124b00226c75dd","unique_id":"0x00124b00226c75dd_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0x00124b00226c75dd/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00124b00226c75dd/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00226c75dd"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor (SNZB-02)","name":"0x00124b00226c75dd","sw_version":"Zigbee2MQTT 1.17.0"},"device_class":"humidity","json_attributes_topic":"zigbee2mqtt/0x00124b00226c75dd","name":"0x00124b00226c75dd humidity","state_topic":"zigbee2mqtt/0x00124b00226c75dd","unique_id":"0x00124b00226c75dd_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0x00124b00226c75dd/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00124b00226c75dd/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00226c75dd"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor (SNZB-02)","name":"0x00124b00226c75dd","sw_version":"Zigbee2MQTT 1.17.0"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b00226c75dd","name":"0x00124b00226c75dd linkquality","state_topic":"zigbee2mqtt/0x00124b00226c75dd","unique_id":"0x00124b00226c75dd_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
INFO: Device '0x00124b00226c75dd' is supported, identified as: SONOFF Temperature and humidity sensor (SNZB-02)
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature and humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-02","vendor":"SONOFF"},"friendly_name":"0x00124b00226c75dd","ieee_address":"0x00124b00226c75dd","status":"successful","supported":true},"type":"device_interview"}'`

sjorge, 2021-01-28:
I don't see the usual configuration successful message in there though... what does the database.db entry for the devices show? (It should store if it was configured sucesfully or not)

nvodenicharski99, 2021-01-28:
Another strage thing is that on the first pair it reported only the humanity and temperature but always the same values.

nvodenicharski99, 2021-01-28:
How do you open the /share/zigbee2mqtt/database.db, SQLite?

sjorge, 2021-01-28:
> How do you open the /share/zigbee2mqtt/database.db, SQLite?

Nope, it's a text file with one line per device, that line is json.

`gerp 0x00124b00226c75dd /share/zigbee2mqtt/database.db` or better yet `gerp 0x00124b00226c75dd /share/zigbee2mqtt/database.db | jq .` should be sufficient.

nvodenicharski99, 2021-01-28:
I see it.
Trying to find a way to copy it. The Terminal add-on is just a beaty ...

nvodenicharski99, 2021-01-28:
here ar the related records:
`
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e31a7f","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

{"id":17,"type":"EndDevice","ieeeAddr":"0x00124b00226c75dd","nwkAddr":51229,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":5,"hwVersion":1,"dateCode":"20201026"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20201026","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1611823324191}
`

sjorge, 2021-01-28:
Looks like the device never got configured after the interviewing... that would explain the lack of temp/humidity updates.

Here is one of mine:
```{"id":69,"type":"EndDevice","ieeeAddr":"0x00124b00000e9009","nwkAddr":27793,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1967}},"msRelativeHumidity":{"attributes":{"measuredValue":5455}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":154}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":"10","maxRepIntval":"1800","repChange":"50"},{"cluster":1029,"attrId":0,"minRepIntval":"10","maxRepIntval":3600,"repChange":"100"},{"cluster":1,"attrId":32,"minRepIntval":"3600","maxRepIntval":"62000","repChange":0}],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1611824482519}```

Notice for me `meta.configured` is set to `2` as that is the value (matching devices.js' configureKey for the device).
`endpoints.1.configureReportings` doesn't show the expected 3 entries.

The question is, why didn't you get a success or a timeout error after joining 🤔

nvodenicharski99, 2021-01-28:
Can I just use your correct records?

sjorge, 2021-01-28:
> Can I just use your correct records?

No, as z2m needs to configure the device. You can try removing the battery and hitting the configure button in the frontend.
Time is very tricker there though, you need to basically click the configure button at the moment the device_accouncement arrives in the logs :(

nvodenicharski99, 2021-01-28:
I neve had so much fun with Xiaomi devices :)

sjorge, 2021-01-28:
I moved to these from Xiaomi, the only hassle free one is my Heiman, but that one was expensive

ianmxyz, 2021-01-28:
My troublesome one is back to normal now, it was reporting low battery even after a battery change. Hope it keeps on going, I can't get to it to press the pairing button easily.

nvodenicharski99, 2021-02-03:
It is all good now with 1.17.1 :))
Thank you guys!!!

nvodenicharski99, 2021-02-03:
Well, not quite :) The sensor reports the same values all the time :)

nanobra1n, 2021-03-06:
As far as my issue goes this can be closed, unless @sjorge 's [comment](https://github.com/Koenkk/zigbee2mqtt/issues/5562#issuecomment-755459075) still needs to be addressed.

closed 2021-04-15 by github-actions[bot]

aq2kx, 2021-10-16:
Procedure described by darksid3r doesn't work for me. It just exposes the LQI value.

Log says:

Zigbee2MQTT:error 2021-10-16 09:26:20: Failed to ping '0x7cb03eaa0a0466c6' (attempt 1/1, Read 0x7cb03eaa0a0466c6/3 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))

hamgammon, 2021-11-02:
My issues seem to be 2 fold. I had a very low battery on mine when it arrived, so after a few pair attempts it no longer worked. It would not pair right next to the Connector at all or only a few meters away. If able, check the battery with a volt meter/use a new one if possable for testing.

The second part is I added a mains relay swich to my Zigbee network (Salus Controls SR600 Smart Relay) and this is acting as a repeater/router now on my network. When using a new battery and the relay on my network it paired correctly first time.

kobuki, 2021-11-03:
I have similar issues with this sensor. Tried all procedures shown in this thread, sometimes one works, sometimes it doesn't. After the latest mass re-pairing because of a network key change T and H didn't come back, whatever I tred. I finally decided to ditch this device, it causes so many headaches that it's not worth the time. The Aqara THP sensor has none of these problems, for instance, for a similar price.

ddacunha, 2023-07-18:
For what is worth, I managed to pair successfully by doing this many times. On my successful attempt I was holding the pair button after inserting the battery. My issue is that the censor shows the temperature in F when HA is configured in C :(

hksthff, 2024-04-06:
Bought two devices: one took several attempts to be interviewed correctly; the other one was interviewed correctly on the first attempt.