Koenkk/zigbee2mqtt #5572: Connect to Amazon AWS MQTT server on Raspberry OS

BFFonseca, 2021-01-05:
## What happened
Can't connect Zigbee2MQTT to an Amazon AWS server with certificates on Raspberry OS.

## What did you expect to happen
Connect to the AWS server with the correct certificates and send my Zigbee sensors info to that server with Zigbee2MQTT

## How to reproduce it (minimal and precise)
Installed Raspberry OS on an SD card, booted and installed Zigbee2MQTT.

Create an Amazon AWS things, certificate and policy (these are working. Tested on MQTT Explorer).

Adjusted configuration.yaml according to https://www.zigbee2mqtt.io/information/configuration.html

#### configuration.yaml
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: z2m
  server: 'mqtts://server.iot.eu-west-2.amazonaws.com:8883' #AWS server hidden for confidentiality reasons
  user: ''
  password: ''
  client_id: 'client_RspB'     
  ca: '/home/pi/Desktop/Certificates/AWS/AmazonRootCA1.crt'
  client: '/home/pi/Desktop/Certificates/AWS/client-certificate.crt'
  key: '/home/pi/Desktop/Certificates/AWS/client-private.key'
  force_disable_retain: true
  version: 3
serial:
  adapter: deconz
  port: null
advanced:
  log_level: debug
  network_key: GENERATE
```

## Debug info
Hardware: Raspberry Pi 3 B
OS: Rasberry OS Linux 10 (buster)
Zigbee2MQTT version: 1.17.0 (commit #07cdc9d) - Latest, just updated before posting
Adapter hardware: Conbee II
Adapter firmware version: 2.07.01 / 9/14/2020 - 26660700

## Console Output Log
https://pastebin.com/nPGs8zTg (AWS server hidden for confidentiality reasons)

## Questions

Would like to know if anyone as ever tried this, can't find anything online.

If you have, what were your results?

Any insight on what might be causing my failed connection atempt?

I'm considering trying the Home Assistant OS, would that be a better fit?

closed 2021-02-12 by github-actions[bot]

andrea-spotsoftware, 2021-06-30:
@BFFonseca did you solved the problem? I'm on the same issue

hoangmnsd, 2022-04-28:
Is there any update on this? I'm having same question.

maxemann96, 2023-07-28:
For any one finding this issue:
I had the same problem. It was caused by an incorrect iam permission. With `version: 3` or `version: 4` the issue was that zigbee2mqtt hangs. With `version: 5` the error was an authentication failure.

I needed to adjust my iam policy, since zigbee2mqtt will send to a topic derived from the base topic, so with the above example, and a thing named `test`, zigbee2mqtt will publish to `z2m/test`. The following iam policy solves the issue (Please always think before copying some random permission policies from the internet!!):
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": [
        "iot:Publish",
        "iot:Receive",
        "iot:PublishRetain"
      ],
      "Resource": [
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topic/z2m/*",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topic/z2m/*",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topic/z2m/*"
      ]
    },
    {
      "Effect": "Allow",
      "Action": [
        "iot:Subscribe"
      ],
      "Resource": [
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topicfilter/z2m/*",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topicfilter/z2m/*",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:topicfilter/z2m/*"
      ]
    },
    {
      "Effect": "Allow",
      "Action": "iot:Connect",
      "Resource": [
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:client/AWS_CLIENT_ID",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:client/AWS_CLIENT_ID",
        "arn:aws:iot:AWS_REGION:AWS_ACCOUNT_ID:client/AWS_CLIENT_ID"
      ]
    }
  ]
}
```