Koenkk/zigbee2mqtt #5573: Can a light switch double press action control the relay instead of a single press?

smowk, 2021-01-05:
I don't know if this is an appropriate feature request or if there is a technical limitation but I'm curious if the relay on a light switch (let's say Xiaomi QBKG12LM) can be controlled by a (physical) **double press** a button. 

Let me try to make this clear with an example:

**_Scenario:_**

A room has a smart bulb (permanently powered), a smart switch and maybe a sensor. When a person enters the room the smart bulb gets lit up...nothing special.

When a guest or unknowing family member toggles the switch out of muscle memory, the bulb loses power...again, nothing special, old problem, multiple solutions.

**_Partial solution:_**

Run the switch button connected to the bulb in decoupled mode, wait for the mqtt message of the button press action in the automaton system (homeassitant) and send the command to toggle the bulb directly to it from the automation platform; this way the bulb never loses power and it can be controlled by the physical light switch, automation platform, etc. - and this works great...unless you consider redundancies

**_Problem:_**

If the automation platform is down, the RPI/server, zigbee antenna, router or any part of the automation system is broken you completely lose control over the relay of the switch (after is set to run in decoupled mode) and therefore you lose control over the light in that room.

**_Solution:_**

Being able to control the relay of the switch from a double click (or hold, or both or whatever).

No guest will have the reflex to double click a light switch (and cut the power to the bulb) and even if the entire automation system is down you would still be able to control the relay by using the physical switch (as long as you still have power..i guess I can't prepare for everything)

I'm thinking a configuration looking like this would clarify the scenario:

```
{
  "operation_mode": {
    "button": "double_left", 
    "state": "control_left_relay"
  }
}
```

I don't know if this is something that can be solved by z2m or if there is a limitation in the switches firmware.
Anyway, Thank you for the amazing platform you developed!

itavero, 2021-01-10:
This is something that would need to be handled by the firmware within the switch.
If it is possible to configure this behavior with the "official" app and gateway for this product, then it might be possible to sniff the Zigbee communication and add support for it in zigbee2mqtt as well.

On the other hand, I'm really wondering why you have a smart switch/relay controlling the power to a smart bulb. Wouldn't a simple "remote control" make more sense?

itavero, 2021-01-10:
Sorry. I didn't check the actual device. I see it can decouple it.

I think the easy solution here is to add an automation in your home automation system that toggles the output when the "double" action is received.

smowk, 2021-01-10:
> I think the easy solution here is to add an automation in your home automation system that toggles the output when the "double" action is received.

The whole point of my suggestion is to have it working when the automation system is down, so it should be handled by the switch

itavero, 2021-01-10:
> The whole point of my suggestion is to have it working when the automation system is down, so it should be handled by the switch

In that case, my first message becomes valid again. If the original app/gateway can configure the device to do that via ZIgbee, then (in theory) it should be possible to add this to zigbee2mqtt as well.
I think the documentation contains some pointers on sniffing Zigbee traffic.

smowk, 2021-01-10:
ok, thank you for your reply...the thing is, I have no idea if the gateway supports this, I never used one. Everything I have in my home has local control; I only bought Zigbee stuff when I discovered zigbee2mqtt, but I will google it

closed 2021-02-18 by github-actions[bot]