Koenkk/zigbee2mqtt #5574: TS0601_thermostat functions not fully in frontend

salopette, 2021-01-05:
Hello, I made a request here, but seems to be the wrong one.
Must be fixed in the z2m.

https://github.com/nurikk/z2m-frontend/issues/243

Koenkk, 2021-01-05:
Added all except: holidays, window_detection_params, workdays. Those are complex types which we don't support yet.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-05 by Koenkk

salopette, 2021-01-05:
Window detection was already available !?

[image]

Koenkk, 2021-01-05:
window_detection_params not window_detection

salopette, 2021-01-05:
A lot has happened! Many Thanks!
Can you still add "local_temperature_calibration": to it?

nurikk, 2021-01-06:
> A lot has happened! Many Thanks!
> Can you still add "local_temperature_calibration": to it?

Check how Koen did this before and try to do like him. First time it would be difficult, but eventually you'll learn how to do this. 
https://github.com/Koenkk/zigbee-herdsman-converters/commit/1976109d7462eb3c072848d1140807d34f5212e0

salopette, 2021-01-06:
I am unfortunately only a user, not a programmer.
my knowledge of English is also 0.

Koenkk, 2021-01-06:
> A lot has happened! Many Thanks!
> Can you still add "local_temperature_calibration": to it?

added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)