Koenkk/zigbee2mqtt #5576: Saswell SEA801-Zigbee/SEA802-Zigbee - Fully ON or OFF valve

semihselcuk, 2021-01-05:
## Is your feature request related to a problem? Please describe
I have a few of these TRVs at home and I'm always frustrated when I turn the heater on and TRVs built-in temperature sensors heat up very quickly (since they are right near the radiators) before the rest of room and valves automatically closes to restrict the water flow. That`s why I use external temperature sensors. 
So basically what I am doing, monitoring the external temp. sensor and when room needs to be heated, I am setting the TRV`s temperature set point to the max which is 30. 

## Describe the solution you'd like
During my tests I`ve realized that even though it is set to maximum sometimes it stills closes the valves little bit to restrict water flow but when I go near the TRV and just click the (+) plus symbol one more time LCD changes to "ON" and opens the valve fully for full flow. 
So I am hoping is there any way to trigger this "ON" functionality via mqtt.

I have already checked what changes on state when click (+) plus symbol to activate "ON" functionality. Pretty much nothing changes "current_heating_setpoint" still 30.0.

Example state:
{
    "anti_scaling": "ON",
    "away_mode": "OFF",
    "battery_low": "false",
    "child_lock": "LOCKED",
    "current_heating_setpoint": "30.0",
    "frost_detection": "ON",
    "linkquality": 39,
    "local_temperature": "25.6",
    "local_temperature_calibration": 0,
    "preset_mode": "none",
    "system_mode": "heat",
    "window_detection": "ON"
}

## Describe alternatives you've considered
I have also tried to set temperature more than 30C. Such as 31C but unfortunately nothing happened. 

## A few photos
Max (30C) temp. you can set via mqtt.
[image]

After clicking the plus button on the TRV. It activates the "ON"
[image]

chrisghz, 2021-01-10:
I'm glad I came across this post. I too have been left scratching my head over this. It doesn't seem that there is a command exposed for the on feature which I also would like control over. I can confirm that when the TRV is at 30.0°c the valve is almost half closed, if you indeed press the + once or the middle button with the thermometer, moon and dot symbols on this also activated the "on" function. The off is available if you send set heat to off mqtt command it's just the ON we need. Position would be lovely too! 
Has anyone else worked out a way around it? They're great little TRV's for the money.

semihselcuk, 2021-01-10:
Thanks @chrisghz  for your comment. I agree with you this great little TRVs definitely worth the money. Thanks for the tip for turning the OFF functionality. I hope somebody figure out how to turn on the ON functionality as well.

insipiens, 2021-01-24:
It doesn't support those options unfortunately, at least when I sniffed it for setting up in Z2M it never showed. I guess its possible the firmware might have changed and it offers this functionality - the manual that came with it would possibly indicate that.

I have a new one on order so will see if its got newer firmware when it arrives etc.

semihselcuk, 2021-01-25:
@insipiens thanks for your comment. I do not have ability to sniff MQTT messages so would be great if you could do that.

denbue, 2021-01-29:
I'd also highly appreciate this feature! For our heater the temperature difference between 30°C and ON is immense.

chrisghz, 2021-01-29:
> I'd also highly appreciate this feature! For our heater the temperature difference between 30°C and ON is immense.

Absolutely. I watched the difference in the valve position between 30° and ON and you're right it's about 4-5mm of displacement, which is a fair amount.

semihselcuk, 2021-03-01:
Has anyone knows how to sniff the MQTT messages in easy way? I have a free CC2351 to use it but don't know where to start.

sorryusernameisalreadytaken, 2021-03-22:
> 
> 
> Has anyone knows how to sniff the MQTT messages in easy way? I have a free CC2351 to use it but don't know where to start.

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
Is that to difficult?

semihselcuk, 2021-03-22:
@sorryusernameisalreadytaken thanks for sharing the link. I do not have free CC2531 stick but I ordered one.

denbue, 2021-03-22:
Thanks @semihselcuk for looking into it!

sorryusernameisalreadytaken, 2021-03-22:
> It doesn't support those options unfortunately, at least when I sniffed it for setting up in Z2M it never showed. I guess its possible the firmware might have changed and it offers this [...]

How can I see my firmwares?

closed 2021-04-30 by github-actions[bot]

maz3max, 2021-11-01:
The TuYa App doesn't support Boost Mode for the SEA802-Zigbee. I tested this device thoroughly.

insipiens, 2021-11-01:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3185

joschi36, 2021-11-01:
> The TuYa App doesn't support Boost Mode for the SEA802-Zigbee. I tested this device thoroughly.

@maz3max  Sad to hear, but thank you very much for your work!

tzbuu, 2022-02-05:
Hi,

I don't know if there is anyone interested about this topic but I found a way to hack all my 5 SEA801 to full open when I set them to 30degree.
I don't know if this work for all SEA 801 but here is what I observed:
After I set TRV to 30 degree I noticed that after a while motor was running even it display 30degree. So I was wondering if there is a feature inside to full open when there is a big difference between set and read temperature only . And I was right. So I tested my TRV outside at 10C and it was always open when temperature was set to 30C.
So what I did, I soldered a 39K resistor seried with the temperature sensor in order to trick the TRV that there is a high difference between set and read. You can compensate this 39k resistor by changing local_temperature_calibration .
You can use other resistor but you need to find out the limit. In my devices, I found an termistor-ntc-100k. You can search for the chart.

Hope this will help you also
Here are two photo's to know how to disassembly it.

[image]
https://photos.google.com/photo/AF1QipOnEHgYDosUDMGWNdLFftNAYehFshTvl0Mky5O1

sorryusernameisalreadytaken, 2022-02-05:
> Hi,
>
> I don't know if there is anyone interested about this topic but I found a way to hack all my 5 SEA801 to full open when I set them to 30degree. I don't know if this work for all SEA 801 but [...]

Awesome finding! I will try to check it this week on my Adapters.

Jarikk, 2022-09-24:
Hello,
Are you able to control the property "local_temperature_calibration"?
I have this TRV, Z2M web interface shows that firmware version is 20180727.
Zigbee2MQTT version [1.27.2]
Coordinator type zStack3x0 (ZigStar Lan stick, CC2652)
Coordinator revision 20220219

But I can't control over this property. I tried to change it using the z2m web interface - it won’t change.
I tried to follow recommendations from the official documentation page (https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html) and sent the following message using the MQTT Explorer:
**topic: zigbee2mqtt_livingroom/Livingroom, TRV 1/set
payload: {"local_temperature_calibration": 5}**
but without success.

What I found is that if I set current heating temperature from the lovelace UI climate card, the following message appears in the mqtt broker:
**topic - zigbee2mqtt_livingroom/Livingroom, TRV 1/set/current_heating_setpoint
payload - 27**
This doesn't follow official instructions, but this definitely works!

I tried to set local_temperature_calibration accordingly
**topic: zigbee2mqtt_livingroom/Livingroom, TRV 1/set/local_temperature_calibration
payload: 3**
but it still doesn't work.

In all these cases I see in the z2m log only messages like these:
Debug 2022-09-24 12:29:21Received MQTT message on 'zigbee2mqtt_livingroom/Livingroom, TRV 1/set/current_heating_setpoint' with data '{ "local_temperature_calibration": 3 }'
Debug 2022-09-24 12:29:21Publishing 'set' 'current_heating_setpoint' to 'Livingroom, TRV 1'

No any responses from the valve. It doesn't asleep for sure, because it immediately reacts to changes of the property 'current_heating_setpoint'.

I will appreciate for any help!

codeclinic, 2024-09-12:
> Hi,
> 
> I soldered a 39K resistor seried with the temperature sensor in order to trick the TRV that there is a high difference between set and read. 

Doesn't this then put the TRV in to a state where the valve is always fully open or fully closed? Thus preventing it from making smoother increases/decreases in temperature?