Koenkk/zigbee2mqtt #5578: TS0601_thermostat not complying/illegal with HA 'climate' documentation standards

Bart-1992, 2021-01-05:
## What happened
Since 1.17.0 the TS0601_thermostat zigbee herdsman converters are not complying Home Assistant 'climate' standards according to : 

[ https://developers.home-assistant.io/docs/core/entity/climate/#hvac-modes](https://developers.home-assistant.io/docs/core/entity/climate/#hvac-modes)
## What did you expect to happen
The HVAC_MODE 'heat' now forces the valve to open fully, without temperature regulation. 

According to HA documentation the 'heat' action is used to set a temperature where the climate entity uses this to regulate temperature.

Now the valve is essentially useless/conflicting with other HA integrations such as Google Assistant, Scheduler component and all the other integrations that do follow the HA standards. 

The valve should work according this post:
#https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-754209191

Name | Description | What TRV should actually do
-- | -- | --
HVAC_MODE_OFF | The device is turned off. | Valve set to 0% / valve operation = close
HVAC_MODE_HEAT | The device is set to heat to a target temperature. | Set TRV to manual preset
HVAC_MODE_AUTO | The device is set to a schedule, learned behavior, AI. | Set TRV to preset picked by user, eg ECO, Schedule etc.

## How to reproduce it (minimal and precise)
not applicable
## Debug info
Zigbee2MQTT version: V1,17,0
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

artkrz, 2021-01-06:
For folks that just want to use these in ON/OFF scenario, this could be easily implemented by exposing valve switch that would toggle it ON/OFF. If someone wants to control their heating this way they can use that switch with generic thermostat.

Bart-1992, 2021-01-06:
@mgrom, are you able to fix this issue? 
Thanks alot!

Rossez, 2021-01-06:
> For folks that just want to use these in ON/OFF scenario, this could be easily implemented by exposing valve switch that would toggle it ON/OFF. If someone wants to control their heating this way they can use that switch with generic thermostat.

I guess most people (as I do) need three way switch:

- ON (valve is fully opened = force open)
- OFF (valve is fully closed = force close)
- NORMAL (valve is in standard operation mode = force normal)

The problem is that this can't be achieved by setting HVAC_MODE only if it changed to comply to HA documentation.

artkrz, 2021-01-06:
> I guess most people (as I do) need three way switch:

What are you basing your assumption on? I mean, this can be true but I don't think We can just say "most people want x or y" without at least doing a poll. Anyway, I would rather have `force` exposed as switch and comply with HA docs than have it that way. If You need this on card You could always add that switch,

Rossez, 2021-01-06:
> What are you basing your assumption on? I mean, this can be true but I don't think We can just say "most people want x or y" without at least doing a poll. Anyway, I would rather have `force` [...]

I said "I guess" not "most people want x or y". :) Sorry if it offended anyone in any way. I agree with you. I will be more than happy if the implementation comply to HA standards and we have `force` exposed (three possible states) so we can use that as we need.

kremik, 2021-01-08:
I think, you can get this value from `system_mode` in `zigbee2mqtt/FRIENDLY_NAME` and change it to requested states via `zigbee2mqtt/FRIENDLY_NAME/set`
with values:
```
{"system_mode":"on"}
{"system_mode":"off"}
{"system_mode":"auto"}
```

closed 2021-02-15 by github-actions[bot]

a8ree, 2022-03-10:
I'm having the same issue as described here.

Other users using ZHA have an additional mode that is not present in Zigbee2MQTT. That mode is 'heat - none'. Whilst the mode is available in Home Assistant, it isn't in Zigbee2MQTT. When I attempt to use this mode, the following error is generated

```
Error Publish 'set' 'preset' to 'Lounge Door Radiator TRV' failed: 'Error: TRV preset off is not recognized.'
Info MQTT publish: topic 'zigbeehouse/bridge/log', payload '{"message":"Publish 'set' 'preset' to 'Lounge Door Radiator TRV' failed: 'Error: TRV preset off is not recognized.'","meta":{"friendly_name":"Lounge Door Radiator TRV"},"type":"zigbee_publish_error"}'
```

My understanding is that when this mode is available, the TRV will respond to the set temperature whilst in heat mode