Koenkk/zigbee2mqtt #5580: Conbee USB umount suddenly when appairing to outlet Ikea

Drealine, 2021-01-05:
# Bug Report

> Zigbee2MQTT:error 2021-01-05 19:08:58: Publish 'set' 'state' to '0x5c0272fffe970290' failed: 'Error: Command 0x5c0272fffe970290/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:debug 2021-01-05 19:08:58: Error: Command 0x5c0272fffe970290/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
    at DeconzAdapter.<anonymous> (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:556:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)

## What happened
On my Home Assistant, when running zigbee2mqtt with IKEA E1603/E1702, if I switch it rapidly or wait more than 5min, the switch not work and my USB Conbee II umount the system. It's work very well with deconz add on. 

## How to reproduce it (minimal and precise)
Pairing Ikea
Switch on/off, rapidly or not
logs report that is not work
usb umount like it on my HA system (after this, I can't see in Hardware info on Supervisor

## Debug Info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: Conbee II
Adapter firmware version: 26580700
Runing zigbee2mqtt on HASS.io on Synology DS220+ with DSM 7.0.

Drealine, 2021-01-05:
Seems that is not a problem with Ikea outlet, but same problems with all sensors. After a while, the USB Conbee doesn't appear.

Drealine, 2021-01-06:
Ok seems that I've a secondary error : 

> usb 3-2: USB disconnect, device number 3
[ 2188.550846] cdc_acm 3-2:1.0: failed to set dtr/rts

I confirm that the USB work well via deconz addon

Koenkk, 2021-01-06:
@chrishae can you help with this?

Drealine, 2021-01-06:
Seems that uninstall this package on the host (no on container) by this command solve the problem : 

> apt-get purge modemmanager

I waiting 1 or 2 days to confirm that.

Drealine, 2021-01-06:
Hm, the problem still appear with uninstall below package.
Regards,

ChrisHae, 2021-01-11:
Don't really know what could cause this but first I would try update the FW to the newest version 26680700.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

Can you please post a zigbee-herdsman debug log from the startup of the adapter. Perhaps the issue happens already there.
https://www.zigbee2mqtt.io/information/debug.html

closed 2021-01-23 by Drealine