Koenkk/zigbee2mqtt #5586: Home Assistant Discovery of Light from External Converter

byrleyar, 2021-01-05:
## What happened
Home Assistant is not discovering the light entity for my unsupported device in my [external converter](https://pastebin.com/h4SNrGAK).  I am trying to expose an RGB light, and z2m is running in hass.io.  The device is added to z2m and listed as 'supported.'  I don't see any errors in home assistant or z2m, but the entity isn't there.  In the dummy converter example in your github repository, it says to go to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js and search for 'const cfg'.  The file must have been recently updated, because 'const cfg' does not pop in a search.  On that page, I found the snipit for exposing a light, but I 100% admit I don't know what I'm doing there.  How do I properly add support for home assistant discovery of a light in an external converter?

## What did you expect to happen
I expected to get a light entity in home assistant.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.0.1 in home assistant
Adapter hardware: Conbee II
Adapter firmware version:

Koenkk, 2021-01-06:
Does this work:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('./converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const device = {
    zigbeeModel: ['XBEE3'],
    model: 'XBee',
    vendor: 'Digi',
    description: 'to control lamp',
    fromZigbee: [fz.on_off, fz.brightness, fz.ignore_basic_report, fz.color_colortemp],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.effect, tz.light_brightness_move, tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp_move, tz.light_colortemp_step, tz.light_hue_saturation_move, tz.light_hue_saturation_step],
    meta: {disableDefaultResponse: true},
    exposes: [e.light_brightness_colortemp_colorxy()],
};

module.exports = device;
```

byrleyar, 2021-01-06:
That does not work.  z2m throws this error:

```
2021-01-06T13:22:46: PM2 log: App [npm:0] online
> zigbee2mqtt@1.17.0 start /zigbee2mqtt-1.17.0
> node index.js
internal/modules/cjs/loader.js:968
  throw err;
  ^
Error: Cannot find module './converters/toZigbee'
Require stack:
- /zigbee2mqtt-1.17.0/lib/util/utils.js
- /zigbee2mqtt-1.17.0/lib/util/settings.js
- /zigbee2mqtt-1.17.0/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:965:15)
    at Function.Module._load (internal/modules/cjs/loader.js:841:27)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at evalmachine.<anonymous>:2:12
    at Script.runInContext (vm.js:130:18)
    at Script.runInNewContext (vm.js:135:17)
    at Object.runInNewContext (vm.js:302:38)
    at Object.getExternalConvertersDefinitions (/zigbee2mqtt-1.17.0/lib/util/utils.js:159:16)
    at getExternalConvertersDefinitions.next (<anonymous>) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/zigbee2mqtt-1.17.0/lib/util/utils.js',
    '/zigbee2mqtt-1.17.0/lib/util/settings.js',
    '/zigbee2mqtt-1.17.0/index.js'
  ]
}
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
Perhaps I need to submit a ticket with `danielwelch/hassio-zigbee2mqtt` to figure out how to reference the other packages?

Hypothetically, if I can figure out the
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('./converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
```
...will this code you proposed add home assistant discovery for this device as a light?

Koenkk, 2021-01-06:
Made a mistake, `const tz = require('./converters/toZigbee');` should be `const tz = require('zigbee-herdsman-converters//converters/toZigbee');`

byrleyar, 2021-01-06:
Amazing!  That external converter is working and I can now communicate with the device via the "exposes" tab of the device page in z2m.  Right now, I'm only seeing a link quality entity in home assistant.  Any ideas how I can add this device in as a color light?  Do I need to modify [this snipit from homeassistant.js](https://github.com/Koenkk/zigbee2mqtt/blob/07cdc9d549a9db4aa94a259388241b71aec4e12d/lib/extension/homeassistant.js#L81)?  I am not sure what `expose` refers to in that snipit of code.

jum, 2021-01-07:
@byrleyar Sorry to be off topic here, I am looking for some sample code to implement a simple zigbee HA device with an XBEE module, do you have some sources to learn how to implement this?

byrleyar, 2021-01-07:
@jum I am just a home hobbyist trying to make a lamp for my kid, but I can certainly share what I've learned along the way.

1. I used XCTU to program the xbee as an end device.  [This guide](https://web.archive.org/web/20180218010916/https://www.falco.co.nz/electronic-projects/xbee-to-smartthings/) that I stumbled upon has been immensely helpful. 
2. That guide discusses the frames you need to send to z2m (via XCTU) to add it as a device to z2m.  This is where I spent a lot of time.  If you follow that guide exactly, you will add your device as a simple on/off switch (`0x0002`).  I wanted to add my xbee as a color light, so I used a device ID of `0x0102`.  [This PDF](https://www.nxp.com/docs/en/user-guide/JN-UG-3076.pdf), although not about the xbee, was very helpful for me to figure out the device IDs.  You will also need to modify your clusters.
3. Once added to z2m, the xbee shows up as an unsupported device.  To fix this, I had to do two things:  add an external converter and modify the database.db file.
4. To add an external converter, I followed [this guidance](https://www.zigbee2mqtt.io/information/configuration.html#external-converters-configuration).  You can see my external converter code above.  It still isn't adding an entity to home assistant, but it isn't forcing any z2m errors, so that's nice.
5. Finally, even with an external converter, z2m doesn't know to use that external converter because z2m does not add a `modelID` to the database file.  I simply edited my database.db file to add `"modelId":"XBEE3"` to the entry for my xbee device.  You need to make sure that whatever you add for modelId is the same as what you used for `zigbeeModel: ['XBEE3']` in your external converter.

As it stands now, I would still love some insight/assistance for my external converter to add this device as a light entity in home assistant.  Thanks all!

jum, 2021-01-07:
@byrleyar I am at the very beginning of this, so I have to do a lot of reading. But thanks a lot for the info!

Koenkk, 2021-01-07:
@byrleyar can you upgrade to latest dev and check again? Pushed a fix for this (looks the same as https://github.com/Koenkk/zigbee2mqtt/discussions/5605)

byrleyar, 2021-01-07:
@Koenkk When moving from z2m to z2m edge, I see that I need to backup my configuration, but what about my database.db and external converter?  Will the edge version pull in all of my devices, or will I need to re-add them all?

Koenkk, 2021-01-07:
@byrleyar you can switch freely

byrleyar, 2021-01-07:
I switched to edge, and my device popped up in home assistant as a light!  Thanks so much, @Koenkk!  For completeness, here is my external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const homeAssistantLight = {
    type: 'light',
    object_id: 'light',
    discovery_payload: {
        brightness: 'brightness',
        color_temp: 'color_temp',
        xy: 'color_xy',
        hs: 'color_hs',
        schema: 'json',
        command_topic: true,
        brightness_scale: 254
    },
};

const device = {
    zigbeeModel: ['XBEE3'],
    model: 'XBee',
    vendor: 'Digi',
    description: 'to control lamp',
    fromZigbee: [fz.on_off, fz.brightness, fz.ignore_basic_report, fz.color_colortemp],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.ignore_rate, tz.effect, tz.light_brightness_move, tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp_move, tz.light_colortemp_step, tz.light_hue_saturation_move, tz.light_hue_saturation_step, tz.light_color],
    meta: {disableDefaultResponse: true},
    exposes: [e.light_brightness_colortemp_colorxy()],
};

module.exports = device; 

```

Koenkk, 2021-01-08:
cool.

Will be available in the new z2m release (1 February).

closed 2021-01-08 by Koenkk

svisso, 2022-10-14:
Thanks all, this helped me to integrate a ZemiSmart wall/ceiling lamp.
It is the first success with an external converter and a massive advancement for me!!