Koenkk/zigbee2mqtt #5590: Wall Touch Switch 4 Gang - 4th gang not work

ARafaelSF, 2021-01-06:
## What happened

I have e 4 gang swith. 

https://www.aliexpress.com/item/4001181765345.html?spm=a2g0s.9042311.0.0.3207b90a5LWe9O

Through the HA interface I can control all the l4 button.

But, when I click on the device it does not send an event to the zigbee2mqtt for the 4th button.
1th to 3th work correctly, but 4th button didn´t work.

I put the log into debug mode and I can see the device sending the message for button 4, but that message is not passed to mqtt.

I also have the 3 button switch, and this one works correctly

## Debug info

In debug log, I pressed buttons 1 to 4 in sequence. Eu 

https://hastebin.com/vemasavono.apache

ARafaelSF, 2021-01-14:
[](url)
[image]

 @Koenkk can you help. 
 
 Repairing din`t work
 
I have two. Both have the same problem.

Koenkk, 2021-01-15:
- https://hastebin.com/vemasavono.apache is empty for me (can you post it on pastebin instead?)
- Can you update to the latest dev branch first? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- This device is currently listed as a 3 gang device, can you provide the data/database.db entry of it?
- Can you provide a screenshot of what you see in HA? I don't expect you can control the `l4` from HA.

ARafaelSF, 2021-01-15:
> * https://hastebin.com/vemasavono.apache is empty for me (can you post it on pastebin instead?)
     Controlling by device - on/of 1, 2, 3, 4, gang - https://pastebin.com/ynvBVwtb
     Controlling by HA- on/of 1, 2, 3, 4, gang - https://pastebin.com/pZDK84MD

> * Can you update to the latest dev branch first? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
    Ok, Updated

> * This device is currently listed as a 3 gang device, can you provide the data/database.db entry of it?
    https://pastebin.com/JhmEAh5j

> * Can you provide a screenshot of what you see in HA? I don't expect you can control the `l4` from HA.
[image]

Koenkk, 2021-01-16:
Seems to be the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/4597 , lets continue there

closed 2021-01-16 by Koenkk