Koenkk/zigbee2mqtt #5591: Ikea E1743 not working as a switch

gerthjan, 2021-01-06:
I hope someone can help me to get the switch to work!

## What happened
The device showed up in my list

## What did you expect to happen
That when i push the switch i get an on or off status.
This is what i see in zigbee2mqtt:
{
    "battery": 60,
    "linkquality": 36,
    "update": {
        "state": "idle"
    },
    "update_available": false
}

## How to reproduce it (minimal and precise)
I've tried removing the E1743 and adding it again, but that didn't solve the problem.

## Debug info
Zigbee2MQTT version: 1.17.0.1
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2021-01-06:
> Adapter firmware version:

what firmware? (https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version)

sjorge, 2021-01-06:
You might also want to read up on https://www.zigbee2mqtt.io/information/binding.html if you want to have it control devices directly.

gerthjan, 2021-01-07:
> > Adapter firmware version:
> 
> what firmware? (https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version)

Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20180507,"transportrev":2},"type":"zStack12"}'

Koenkk, 2021-01-07:
@gerthjan your firmware is too old, updating to the latest will fix it: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

gerthjan, 2021-02-07:
I've had the opertunity to update my and it solved my problem.

closed 2021-02-07 by gerthjan