Koenkk/zigbee2mqtt #5596: Support for Tuya Blind Motor - Recognised as TS0601

milobt, 2021-01-06:
## Information about the device + link
[https://www.aliexpress.com/item/1005001775235402.html?spm=a2g0s.9042311.0.0.24644c4dOe24TO](https://www.aliexpress.com/item/1005001775235402.html?spm=a2g0s.9042311.0.0.24644c4dOe24TO)

## data/database.db entry of the device
`{"id":22,"type":"EndDevice","ieeeAddr":"0x5c0272fffeca9e6c","nwkAddr":5763,"manufId":4098,"manufName":"_TZE200_rddyvrci","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_rddyvrci","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1609436887877}`

Log:
`Zigbee2MQTT:debug 2020-12-31 17:48:07: Received Zigbee message from '0x5c0272fffeca9e6c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-31 17:48:07: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-12-31 17:48:07: Received Zigbee message from '0x5c0272fffeca9e6c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-31 17:48:07: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-12-31 17:48:07: Received Zigbee message from '0x5c0272fffeca9e6c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-31 17:48:07: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2020-12-31 17:48:07: Received Zigbee message from '0x5c0272fffeca9e6c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-31 17:48:07: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2020-12-31 17:48:07: Successfully interviewed '0x5c0272fffeca9e6c', device has successfully been paired
Zigbee2MQTT:warn  2020-12-31 17:48:07: Device '0x5c0272fffeca9e6c' with Zigbee model 'TS0601' and manufacturer name '_TZE200_rddyvrci' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-12-31 17:48:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x5c0272fffeca9e6c","ieee_address":"0x5c0272fffeca9e6c","status":"successful","supported":false},"type":"device_interview"}'`

d-edge82, 2021-01-10:
I have the same curtain motor (returned the unsupported livolo one)
@Koenkk please advise

Koenkk, 2021-01-11:
@d-edge82 see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

d-edge82, 2021-01-11:
> @d-edge82 see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Thanks!
Managed to connect it, now I am struggling with it's behaviour..
Clicking "Down" on HA opens the curtain, "Up" does nothing and "Stop" closes it..

Any idea?

milobt, 2021-01-11:
I got the same result after connecting, I figured it might be because I haven’t yet configured the up and down stopping points,  but still might be an error in the Z2M profile.

Koenkk, 2021-01-11:
Try adding it to `model: 'TS0601_roller_blind',` instead of ` model: 'TS0601_curtain',`

d-edge82, 2021-01-11:
> Try adding it to `model: 'TS0601_roller_blind',` instead of ` model: 'TS0601_curtain',`

Can you please explain where I need to change it?
I am on Hassio with the latest Z2M (edge does not recognize the device)

Koenkk, 2021-01-11:
@d-edge82 can you provide the data/database.db entry of it?

d-edge82, 2021-01-11:
Is this what you are after?

{"id":2,"type":"EndDevice","ieeeAddr":"0x5c0272fffecc0047","nwkAddr":44405,"manufId":4098,"manufName":"_TZE200_rddyvrci","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_rddyvrci","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610387106467}

Koenkk, 2021-01-11:
Can you try if it is fixed now?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

d-edge82, 2021-01-11:
Cool, thanks!

Dev-branch means edge edition of the add-on, right?
I tried that a few days ago and it did not recognize the device,
Does this mean it will be recognized if I update it tomorrow morning?

Koenkk, 2021-01-11:
@d-edge82 should be, first uninstall and install again

d-edge82, 2021-01-11:
OK, waited, uninstalled and installed back again
I now see it shows up as roller_blind, but the issue is still there:

Stop: Closes the curtain
Up: Opens
Down: Stop

[image]

Should I wait longer?

Koenkk, 2021-01-12:
@milobt is it working for you?

milobt, 2021-01-13:
@Koenkk on edge I get the same @d-edge82. Although, I believe the operations to be as follows:

Lovelace Button <-> Blind Operation
Up <-> STOP
Stop <-> DOWN
Down <-> UP

rbrueckner82, 2021-01-13:
same issue

Button

OPEN / doesn't work
CLOSE / open the blind
POSITION /  doesn't work

the warning are at log when i start any event.

WARNING: owvfni3: Unhandled DP #105: {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}

[image]

milobt, 2021-01-14:
I also find that position does not work at all.

Furthermore, considering that these motors run on batteries it would be useful if battery level was available on one of the endpoints.

DomiStyle, 2021-01-14:
@milobt Does setting the position not work or receiving the position?

Judging by @rbrueckner82 warning receiving position can't work yet.

d-edge82, 2021-01-15:
This is the current status for me (on Edge):
Position - works fine.
Close=Stop
Open - Works.
0 = Open
100 = Closed (wish it was the other way around)

On HA the behavior is as following:

Stop = Close
Up = Open
Down = Stop

Position works.

Thanks for the support @Koenkk !

eric-zimmer, 2021-01-15:
Hello, I have got the same zigbee blind motor. I am not able to pair it with zigbee2mqtt (HA) or deconz. How did you do this? Also I have a tuya gateway, also with this I cannot connect the motor. How can I put the device into pairing mode? I haven't found anything for this in the instruction

d-edge82, 2021-01-15:
> Hello, I have got the same zigbee blind motor. I am not able to pair it with zigbee2mqtt (HA) or deconz. How did you do this? Also I have a tuya gateway, also with this I cannot connect the motor. How can I put the device into pairing mode? I haven't found anything for this in the instruction

Hi Eric,

Hold the down button along with the set, this should reset the device and enter pairing mode

eric-zimmer, 2021-01-15:
Hi, thank you for your fast answer.  I thought it need to flash, it doesn't but it connected to deconz, so it should work with z2m, will try this next. thank you.

Koenkk, 2021-01-16:
Can you guys check out the latest dev branch and see if it's fixed there? (@DomiStyle did some fixes here)

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

d-edge82, 2021-01-16:
> Can you guys check out the latest dev branch and see if it's fixed there? (@DomiStyle did some fixes here)
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Just tried;
Close = Close.
Open = stop.
And I have no other options.

Position:
100 = Open
0 = Closed.

Should I try again in a few hours?

DomiStyle, 2021-01-16:
@d-edge82 Position looks good, I reversed open and stop for your model, you can try once @Koenkk merges it.

Stop is only available from Home Assistant.

Koenkk, 2021-01-16:
Merged the fix from @DomiStyle .

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rbrueckner82, 2021-01-16:
i have tested the docker image latest-dev

same @d-edge82 

Close = Close.
Open = Stop.
And I have no other options.

Position:
100 = Open
0 = Closed

 works fine

the battery level would be interesting if possible

thanks for the great support

Koenkk, 2021-01-16:
@rbrueckner82 plesae check again in 30 mins from now

DomiStyle, 2021-01-16:
> Warning TuYa_cover_control: Unhandled DP #105 for _TZE200_rddyvrci: {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}

@rbrueckner82 
Do you happen to know what the battery level currently is? 105 isn't defined for covers and if 40% isn't the current position then it might be the battery level.

rbrueckner82, 2021-01-16:
now the position at 50%
[image]

this is the latest warning

Warning TuYa_cover_control: Unhandled DP #105 for _TZE200_rddyvrci: {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}

Koenkk, 2021-01-16:
> @rbrueckner82 plesae check again in 30 mins from now

Delayed a bit, check 30 mins from now

rbrueckner82, 2021-01-16:
a question like is the command topic in mqtt the state topic "zigbee2mqt /rlsh_childroom" works fine, but the command topic "zigbee2mqtt/rlsh_childroom" does not work, at the broker the  % status arrives

rbrueckner82, 2021-01-16:
> > @rbrueckner82 plesae check again in 30 mins from now
> 
> Delayed a bit, check 30 mins from now

ok i pulled the latest-dev, no changes

d-edge82, 2021-01-16:
Works perfect for me here! thank you so much!

milobt, 2021-01-18:
All works perfectly for me too. Many thanks @Koenkk.

Only additional request is whether battery level is available?

I also get the warning in the log, as @DomiStyle pointed out:
```
Zigbee2MQTT:warn  2021-01-18 11:02:37: TuYa_cover_control: Unhandled DP #105 for _TZE200_rddyvrci:
                {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}
```

DomiStyle, 2021-01-18:
@milobt Can you fully charge the battery and check if the parameter 105 changes? Since it shows 40 for both of you I'm not sure it's related to the battery.

Alternatively you can try checking the logs after a longer period like a week, maybe more warnings pop up.

milobt, 2021-01-18:
Have had on charge all day - green light on the charging plug when charged - and still reports the same value of 40, so sadly must not be associate with the battery level.

```
Zigbee2MQTT:warn  2021-01-18 15:36:02: TuYa_cover_control: Unhandled DP #105 for _TZE200_rddyvrci:
                {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}
```

andreasg73, 2021-01-20:
I have a new device signature AM43-blind zigbee, please add it to the database. Mine reports as _TZE200_zah67ekd.
Thank you

joseant, 2021-01-22:
Same here I have a AM43 and report _TZE200_zah67ekd I have added it to device,js and it is working, the only thing is when I open or close it I get a warning:

WARNING: TuYa_cover_control: Unhandled DP #105 for _TZE200_zah67ekd: {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,40]}}

DomiStyle, 2021-01-22:
@joseant That seems plausible. Can you try replacing your toZigbee.js with [this file](https://github.com/DomiStyle/zigbee-herdsman-converters/blob/tuya-cover-speed/converters/toZigbee.js), restart zigbee2mqtt and set the speed like this:

Publish to topic:

    zigbee2mqtt/FRIENDLY_NAME/set

With value:

    {
        "options":{
            "motor_speed": 80
        }
    }

After that you can check if the speed and/or the report for DP 105 changed.

joseant, 2021-01-22:
@DomiStyle  You are right, now it changes:

warning TuYa_cover_control: Unhandled DP #105 for _TZE200_zah67ekd: {"status":0,"transid":0,"dp":105,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,80]}}

And the motor goes faster

DomiStyle, 2021-01-22:
@joseant Good to know, will send a pull request tomorrow so we can get this change in the dev branch, will add the new `_TZE200_zah67ekd` signature too.

Do the up/down/stop and position commands work properly as is? 100% should be open.

Does the app limit the speed selection somehow or is 0-100 all valid?

@Koenkk Can I add the "speed" attribute to the device state when it gets reported or should I ignore it?

joseant, 2021-01-22:
@DomiStyle up/down/stop and position commands work properly the only thing is the warning I sent you.

Koenkk, 2021-01-22:
@DomiStyle fine with that, but please use `motor_speed` as the name

DomiStyle, 2021-01-22:
@Koenkk Will do.

@joseant I will restrict values from 0-255 then.

joseant, 2021-01-22:
@DomiStyle 
The only thing I have seen is the state is unknown

[image]

DomiStyle, 2021-01-22:
@joseant I noticed that my state is stuck in closed state since forever too, will fix that aswell.

@Koenkk I could have sworn that zigbee2mqtt added the state: closed/open automatically when I wrote the original tuya_curtain functions, did something change?

I assume I can just set state: open/closed? Or is it state: OPEN/CLOSE?

Koenkk, 2021-01-23:
@DomiStyle I'm not aware of any changes here. To update the state after a successful set; add `return {state: {state: value.toUpperCase()}}` here https://github.com/Koenkk/zigbee-herdsman-converters/blob/ac5d5270848c76bbdf516f65ea60d241a91cfaee/converters/toZigbee.js#L3412

DomiStyle, 2021-01-24:
@Koenkk Strange, I still have `state: closed` stuck on my curtain from ages ago.

Wouldn't it make more sense to set the state in the fromZigbee part when the position gets set? Otherwise pulling on the curtain wouldn't update the state, only the position.

Koenkk, 2021-01-24:
@DomiStyle agree, position = 0 would be `CLOSE`, else `OPEN`

joseant, 2021-01-24:
@DomiStyle 
Hello, thanks for the commit, I have review it, the motor speed works fine. But the state is still unknown

[image]

DomiStyle, 2021-01-25:
@joseant Should work once https://github.com/Koenkk/zigbee-herdsman-converters/pull/2150 is merged.

JAJBrouwers, 2021-01-25:
Hello there,

I also have a cover blind on zigbee2mqtt, the only thing that's not working is the "state" it's unknown all the time... When I follow the documentation and publish a message to get the state nothing happens.

And that option for the motor speed, is that only in the dev branche? Because I don't have that option.

Thx in advanced, Jeroen

DomiStyle, 2021-01-25:
@JAJBrouwers The fix for the state is not merged yet.

The option for motor speed is only available in dev and only works for supported devices.

JAJBrouwers, 2021-01-25:
Oke then, thx for you're feedback, I have the normal version, when I take the Dev add-on... Do I need to install everything again?

DomiStyle, 2021-01-26:
@joseant Can you try again with the latest dev version? Everything should work out of the box now.

joseant, 2021-01-26:
@DomiStyle 
Hello.
I have switch to the dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html
But I can't see the changes yet

joseant, 2021-01-27:
@DomiStyle 
I have now tried the dev branch and everything has worked fine 

[image]

JAJBrouwers, 2021-01-27:
Nice, how do I get this update without using the Dev branche? Will there be a update or something?

DomiStyle, 2021-01-28:
@JAJBrouwers Yes, once the next update releases it will be available without dev branch.

Since everything works I guess this can be closed now @Koenkk @milobt.

closed 2021-01-28 by Koenkk

JAJBrouwers, 2021-02-01:
Hi there I have just installed the dev branch, I get the option for the motor speed... when I set it and hit apply, nothing changes. for the state value, it still does not report back open or close.

[14 more replies]