Koenkk/zigbee2mqtt #5597: Removing/changing homeassistant autodiscovery topics

rogiertermeer, 2021-01-06:
## Is your feature request related to a problem? Please describe
The linkquality is not a usable value. It will only increase and all the devices report the same values. So I would like to drop this in stead of sending it to homeassistant.

I used the option "filtered_attributes: ["linkquality"]. Wich is working, but the autodiscovery message for linkquality is still send.

## Describe the solution you'd like
I would like to choose wich autodiscovery messages are send, because i don't need battery low or battery as entities in homeassistant since I'm using node red for sending alerts when a battery is almost empty.

So the option filtered topics would be nice for that.

## Describe alternatives you've considered
Now I'm using node red to read the messages on "homeassistant", filter and change them and then passing them true to the autodiscovery topic homeassistant is using. (I changed the topic).

In Node red i'm also adding some attributes as battery type. So it would also be nice if you can add some attributes with static values your self.

closed 2021-01-06 by Koenkk

rogiertermeer, 2021-01-06:
Thank you for the information.

Now I understand what the documentation means.

rogiertermeer, 2021-01-06:
I've got a question. I added all the sensors I don't want to use but they are still published. Can you tell me what I'm doing wrong:

  '0x00158d00049bfec7':
    friendly_name: Wandschakelaar logeerbed_links
    retain: false
    device_options:
      homeassistant:
        linkquality: null
        battery: null
        action_both: null
        action_left: null
        action_right: null
        action_both_double: null
        action_left_double: null
        action_right_double: null
        action_both_long: null
        action_left_long: null
        action_right_long: null
    filtered_attributes: ["linkquality"]

Koenkk, 2021-01-06:
Hope this explains it:

```yaml
devices:
  '0x00158d00049bfec7':
    friendly_name: Wandschakelaar logeerbed_links
    retain: false
    # if you put it like this only the linkquality sensor for this device will not be discovered
    homeassistant:
      linkquality: null

# device_options are applied to all devices, if you put it like this linkquality sensor for all devices will be disabled 
device_options:
  homeassistant:
    linkquality: null
```

rogiertermeer, 2021-01-06:
Thank you!

I did as below for every device:
```
  '0xbc33acfffe1326f1':
    friendly_name: Bewegingsmelder overloop
    retain: false
    homeassistant:
      linkquality: null
      battery_low: null
      battery: null
      tamper: null
    filtered_attributes: ["linkquality"]
    no_occupancy_since: [300]
```

But after that only the first device is send to mqtt for the next device I see the following error in the logging:

```
Zigbee2MQTT:error 2021-01-06 22:16:15: Failed to call 'HomeAssistant' 'onZigbeeEvent' (TypeError: Cannot convert undefined or null to object

    at Function.keys (<anonymous>)

    at add (/app/lib/extension/homeassistant.js:757:28)

    at /app/lib/extension/homeassistant.js:775:21

    at Array.forEach (<anonymous>)

    at HomeAssistant.discover (/app/lib/extension/homeassistant.js:608:41)

    at HomeAssistant.onZigbeeEvent (/app/lib/extension/homeassistant.js:853:18)

    at Controller.callExtensionMethod (/app/lib/controller.js:381:44))
```
The same thing will happen when I set it as below:
```
device_options:
  homeassistant:
    linkquality: null
```

I'm using the latest version in docker.

Koenkk, 2021-01-07:
This was a bug indeed, fix now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rogiertermeer, 2021-01-07:
Thank you very mutch!!

I'm now running the latest-dev docker image. It's better, but not everything is working as it schould I think.
When I'm using:
```
device_options:
  homeassistant:
    linkquality: null
```
The sensor are still send. When I add them to de device specific settings not anymore. So that works great!

I also added the "device_automation" sensors wich are not send directly but only when the button for example is pressed. But these are still send. Is that how it's supposed to be?

Koenkk, 2021-01-08:
> The sensor are still send. When I add them to the device specific settings it's not send anymore. So that works great!

I cannot reproduce this, can you provide your full `configuration.yaml`?

> I also added the "device_automation" sensors wich are not send directly but only when the button for example is pressed. But these are still send. Is that how it's supposed to be?

Fixed this now

> There is an other thing I found: When I disable discovery of a specific sensor. 

This shouldn't happen:
- `homeassistant: ` overrides control if ha sensor are discovered or not (but does not prevent properties from being published to the state)
- filtered_attributes: allows to filter properties from the published state (but does not prevent a sensor from being discovered)

rogiertermeer, 2021-01-11:
> > The sensor are still send. When I add them to the device specific settings it's not send anymore. So that works great!
> 
> I cannot reproduce this, can you provide your full `configuration.yaml`?
> 
This is my complete configuration.yaml

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.20.20.1'
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - 90
    - 190
    - 154
    - 114
    - 85
    - 28
    - 182
    - 237
    - 43
    - 129
    - 96
    - 24
    - 32
    - 147
    - 126
    - 84
devices:
  '0xbc33acfffe5019d8':
    friendly_name: Bewegingsmelder gang
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe1326f1':
    friendly_name: Bewegingsmelder overloop
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe50143a':
    friendly_name: Bewegingsmelder inloopkast
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe363162':
    friendly_name: Bewegingsmelder garage
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe0eb8c9':
    friendly_name: Bewegingsmelder toilet_beneden
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe109184':
    friendly_name: Bewegingsmelder toilet_boven
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0xbc33acfffe1324a2':
    friendly_name: Bewegingsmelder zolder
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
    no_occupancy_since:
      - 300
  '0x00158d0001e4b333':
    friendly_name: Contactsensor huis_deur_achter
    retain: false
    homeassistant:
    filtered_attributes:
      - linkquality
  '0x00158d00019d06e4':
    friendly_name: Contactsensor huis_deur_voor
    retain: false
    homeassistant:
    filtered_attributes:
      - linkquality
  '0x00158d00049bfec7':
    friendly_name: Wandschakelaar logeerbed_links
    retain: false
    homeassistant:
      action_both: null
      action_left: null
      action_right: null
      action_both_double: null
      action_left_double: null
      action_right_double: null
      action_both_long: null
      action_left_long: null
      action_right_long: null
    filtered_attributes:
      - linkquality
  '0x00158d0005340ce2':
    friendly_name: Portableschakelaar slaapkamer
    retain: false
    homeassistant:
      click: null
      click_single: null
      action_single: null
      click_double: null
      action_double: null
      action_shake: null
    filtered_attributes:
      - linkquality
      - click
  '0x04cf8cdf3c78b29c':
    friendly_name: Lichtsensor woonkamer
    retain: false
    homeassistant:
      illuminance_lux: null
    filtered_attributes:
      - linkquality
  '0x04cf8cdf3c7f7857':
    friendly_name: Lichtsensor overloop
    retain: false
    homeassistant:
      illuminance_lux: null
    filtered_attributes:
      - linkquality
  '0x04cf8cdf3c7b552c':
    friendly_name: Lichtsensor kantoor
    retain: false
    homeassistant:
      illuminance_lux: null
    filtered_attributes:
      - linkquality
  '0x00158d00058669be':
    friendly_name: Tempvochtsensornodisplay badkamer
    retain: false
    homeassistant:
      pressure: null
    filtered_attributes:
      - linkquality
  '0x5c0272fffee17198':
    friendly_name: Tempvochtsensordisplay garage
    retain: false
    homeassistant:
    filtered_attributes:
      - linkquality
  '0x0017880108e72f70':
    friendly_name: Huestekker 1
    retain: false
    homeassistant:
      update_available: null
      update_state: null
    filtered_attributes:
      - linkquality
  '0xbc33acfffe63fdfc':
    friendly_name: Bewegingsmelder kantoor
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
  '0xbc33acfffe698f7b':
    friendly_name: Bewegingsmelder badkamer
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality
  '0x00158d0004874d4d':
    friendly_name: Kubus woonkamer
    retain: false
    homeassistant:
      action_wakeup: null
      action_slide: null
    filtered_attributes:
      - linkquality
  '0x60a423fffe138d52':
    friendly_name: Watersensorintern zolder
    retain: false
    homeassistant:
      battery_low: null
      tamper: null
    filtered_attributes:
      - linkquality

device_options:
  homeassistant:
    linkquality: null
    battery: null
```
I'm on the latest image now and this is still happening.

> > I also added the "device_automation" sensors wich are not send directly but only when the button for example is pressed. But these are still send. Is that how it's supposed to be?
> 
> Fixed this now
> 

I can confirm this. Geat!

> This shouldn't happen:
>
> * `homeassistant: ` overrides control if ha sensor are discovered or not (but does not prevent properties from being published to the state)
> * filtered_attributes: allows to [...]

Now I made a new container with the latest version this problem is gone. Maybe I did something wrong. But it's ok now.

rogiertermeer, 2021-01-12:
I ran into something else as well.
Because I'm editing the mqtt messages in node red before I send them to the home assistant auto discovery topic I'm using the name to identify witch type of sensor it is and In witch location it is. 
I was splitting the name in parts based on the spaces witch are in there. There for in some friendly names I used "_" in stead of spaces. But when I do that the name in the auto discovery message is also using "_".

```
"name": "Bewegingsmelder toilet_boven_occupancy",
```
```
 "name": "Bewegingsmelder garage occupancy",
```
This is the name attribute (not the device.name).

Is this how it supposed to work? If so, can you tell me witch character I can use for this?

Koenkk, 2021-01-12:
Yes this is expected indeed. Maybe you can use `-`.

rogiertermeer, 2021-01-13:
That works fine! Thank you.