Koenkk/zigbee2mqtt #5599: Müller Licht Int. Remotecontrol (Model: 404002)

MahrWe, 2021-01-06:
I successfully paired my 4-button Müller Licht Int. Remotecontrol (Model: 404002). It would show in Devices and identifys with 

0x00158d0002d5019d (0xBA92) | Müller Licht | 404002 | 147 
-- | -- | -- | -- 

So, hoped it would "just work" but I don´t get any battery information or action messages in the log. The only information that shows is LQl (and I am not sure if its actually updated or still from time of pairing).

I updated my CC2531 to the latest zStack12 firmware and repaired the Remote to no avail. Because of an unclosed issue #5344 regarding this remote, I am not sure whether the actual functionality is not yet implemented.
Due to the marketing advertising it as zigbee3.0 device I wonder if zStack3 could help improve things.

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 1.17.0 (Home Assistant)
Adapter hardware: CC2531
Adapter firmware version: 20190608 (zStack12)

MahrWe, 2021-01-06:
I enabled debug-log and copied the pairing process. There is a lot of "No converter available for '404002'" in it. 
Maybe it can be of help.
[Müller remote debug log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5776990/Muller.remote.debug.log.txt)

Koenkk, 2021-01-06:
Is anything logged in the debug log when pressing the buttons?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

MahrWe, 2021-01-06:
I enabled logging in the Addon configuration (replaced info with debug) But I still get no messages when buttons are pressed. 
From time to time my Aqara Cube will send a wake message so I can see other devices work as intended.

Also the last message in my log from device "Wohnzimmer" is from the Lamp which came with the remote.

Koenkk, 2021-01-06:
Seems it's your lucky day, @andreas-seiderer  just created a fix for this in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2042

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-06 by Koenkk