Koenkk/zigbee2mqtt #5601: TuYa valve TS0601: Problem setting force:open/close value

kremik, 2021-01-06:
## What happened
When set force:open value, or any of allowed values via `zigbee2mqtt/FRIENDLY_NAME/set` it is ignored and not set. 
```
{
      "force": "open"
}
```

There is no way how to change force value.

In log there is a following message
`Zigbee2MQTT:error 2021-01-06 20:14:42: No converter available for 'force' (open)`

## What did you expect to happen

I expect to send force:opne or force:close to change the state of the valve - total open or total close.

## How to reproduce it (minimal and precise)
Send the set command with values above and see the devices response.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531
Adapter firmware version: 3.2.6
Accessory:
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

salopette, 2021-01-06:
https://github.com/Koenkk/zigbee2mqtt/issues/5578#issuecomment-755583940

marcin77, 2021-01-07:
I have the same problem, the heating has gone down

kremik, 2021-01-08:
It seems to be a problem also for all /get parameters - getting this error also via web frontend when refresh button is pressed in Exposes tab.

Koenkk, 2021-01-08:
This was removed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/1983 by @mgrom. @mgrom can this be re-added?

mgrom, 2021-01-09:
@Koenkk I've just created PR but some errors occured not related to my change.

Koenkk, 2021-01-09:
@mgrom big thanks again! 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-09 by Koenkk

Lipown, 2023-10-31:
Still ave the problem... I am on latest Z2M 
[image]