Koenkk/zigbee2mqtt #5602: Aqara Temperature Sensor cannot be added anymore

Fridolin34, 2021-01-06:
Hello everybody,

I had an Aqara Sensor added to my system and together with two other sensors of the same kind, it was working flawlessly for round about 2 month now. Yesterday I've realized that the sensor is no longer present and cannot be added anymore. I started everything from scratch, I've updated to the newest version (1.17.0), updated all other components (like Supervisor, HA, ...). But I'm not successful.

When I try to pair the sensor, it starts blinking, but in the z2m log (debug) nothing is to be seen. I've switched the batteries from one Aqara to the other, but the suspicious device seems to be dead.

Is there any idea, what else I could do?

Additionally I tried to switch from the normal z2m to the edge version, using the exact same configuration. In the past this was an easy step, just to try out something. But now, most devices are not longer listed, but the missing Aqara is at least shown there - but without any value updates. Are there different databases used to store the registered entities? Is there any way to force the device to talk to the coordinator and can I see in additional logs, whether the device is still alive or not?

Thanks in advance,
Fridolin

tpftpf, 2021-01-07:
I experienced similar problems. I always succeeded when I first "force remove" the device.
Adding afterwards was possible.

masamu2006, 2021-01-07:
Yo he tenido el mismo problema, pero hice lo siguiente y logré emparejar de nuevo los dispositivos aqara:
1.- Instalé versión 1.17.0.1 de Zigbee2Mqtt
2.- Elimine los dispositivos de /share/zigbee2mqtt/devices.yaml
3.- Elimine los dispositivos de /share/zigbee2mqtt/database.db
4.- Emparejar de nuevo los dispositiovos.

Para mi ha funcionado.

Saludos.

Fridolin34, 2021-01-07:
Thanks for your ideas. 

@tpftpf: I cannot see the device anymore, therefore 'force remove' is unfortunately not possible.

@masamu2006: I tried exactly that. What I haven't done before, was cleaning up the database.db. But after doing so, I still cannot pair the device again.

Fridolin34, 2021-01-08:
I've just run a search to find a file still including remains of this sensor, but it cannot be found. I've searched the whole container as well as the share/... files, but nothing.

Basically, this cannot be, because my lovelace still reports that this sensor is missing. So at least I would have expected to find this entry in a YAML somewhere. 

Does anyone know, where I did not search or where I still could search for entries concerning this sensor?

Thanks.

tpftpf, 2021-01-08:
Ppermit join?
Initiate the pairing close to the controller.
This should work.

Fridolin34, 2021-01-08:
I have found another file, where this sensor was still listed in a category called 'deleted devices'. I just removed it from there, but no luck.

@tpftpf : permit join is set to true ... :(

tpftpf, 2021-01-08:
Your sensor is probably broken.
Battery ok?

tpftpf, 2021-01-08:
To verify: Force delete one of the other sensors and repair.

Fridolin34, 2021-01-08:
Battery has been exchanged and new devices have been added recently. So coordinator is fine. 

Sensor is relatively new - 4-5 weeks. But I don't know, whether it sends a signal or not. It would be nice to know, where I could see, whether the coordinator is receiving something or not.

tpftpf, 2021-01-08:
Did you activate debug for loging?

Fridolin34, 2021-01-08:
yes

tpftpf, 2021-01-08:
I have no more ideas, sorry

Fridolin34, 2021-01-08:
No problem. Thanks for helping anyways.

I just think there has to be a log, where I should be able to see, which signals are arriving the coordinator. I just thought about the mosquitto broker and just enabled logging there. But this seems to be the wrong approach, because this log (on info) does not really log something. I keep on trying.

Fridolin34, 2021-01-08:
Changed it to debug and now I can see a lot more, but I guess mosquitto is already too late. It should be already in the z2m context, but I don't know, where to keep an eye on.

skumka, 2021-01-12:
I hale similar issue. I have 2 sensors which in the past regularly became unavailable. I was able to repair them few weeks back. Now, even after deletion from z2m I cannot add them anymore. Nothing is happening, even vwry close to the Usb cc2538+cc2592 with antenna coordinator.

Fridolin34, 2021-01-12:
If I only knew, whether zigbee2mqtt is the application processing the signals of the coordinator or if there is another layer in between. If the coordinator is not receiving anything, the device is dead. But if the device is just being ignored/declined, then it is a software/configuration issue and can be fixed.

masamu2006, 2021-01-12:
Fridolin34, prueba a detener Zigbee2mqtt y desenchufa el coordinador CC2531 o similar durante unos 10 segundos. Vuelve a conectar y reiniciaZigbee2mqtt.

Saludos.

Fridolin34, 2021-01-12:
@masamu2006  Thanks a lot, this did the trick. I just stopped z2m, then I stopped the whole HA, removed the stick, waited round about one minute, plugged in the stick, restarted everything and ... it worked.

Thanks!

closed 2021-01-12 by Fridolin34