Koenkk/zigbee2mqtt #5608: Sonoff Motion Sensor SNZB-03 Reactivation time

Westeru, 2021-01-07:
## What happened

The sensor SNZB-03 have a pretty long re activation time and i want to make it faster.
I tried to add these lines in device.js

const payload = [{attribute: 'batteryVoltage', minimumReportInterval: 1000, maximumReportInterval: 3000,reportableChange:0}];
await endpoint.configureReporting('genPowerCfg', payload);

 and after repairing the sensor was not recognized any more. 

## What did you expect to happen
Reactivation time to be much faster and the nu state detection "occupacy" in this case to be at 30s

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Westeru, 2021-01-07:
The full device description in devices.js is:
```
fingerprint: [
            {type: 'EndDevice', manufacturerName: 'eWeLink', modelID: '66666', endpoints: [
                {ID: 1, profileID: 260, deviceID: 1026, inputClusters: [0, 3, 1280, 1], outputClusters: [3]},
            ]},
        ],
        zigbeeModel: ['MS01', 'MSO1'],
        model: 'SNZB-03',
        vendor: 'SONOFF',
        whiteLabel: [{vendor: 'eWeLink', model: 'RHK09'}],
        description: 'Motion sensor',
        fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const bindClusters = ['genPowerCfg'];
            await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
            await reporting.batteryVoltage(endpoint);
     
        },

        exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
```

sjorge, 2021-01-07:
> I tried to add these lines in device.js
> 
> ```
> const payload = [{attribute: 'batteryVoltage', minimumReportInterval: 1000, maximumReportInterval: 3000,reportableChange:0}];
> await endpoint.configureReporting('genPowerCfg', payload);
> ```

I'm not seeing where you added these lines in devices.js in your copy mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/5608#issuecomment-756050149.

I'm also not sure what you're trying to-do, configuring a different reporting interval for the batteryVoltage is not going to make any chances to the device's occupancy timeout. 

You'll have better luck adding `fz.occupancy_timeout` to the toZigbee array. But I have a vague memory of this having no effect on this device and it being in the firmware to be 30 seconds, so we can't change it.

sjorge, 2021-01-07:
Apparently it is 60 seconds and occupancy_timeout is indeed not supported https://github.com/Koenkk/zigbee2mqtt/issues/4874#issuecomment-721845936

Westeru, 2021-01-07:
I deleted those line because the sensor was not recognized an that is why there were not there 
```
meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const bindClusters = ['genPowerCfg'];
            const payload = [{attribute: 'batteryVoltage', minimumReportInterval: 1000, maximumReportInterval: 3000,reportableChange:0}];
            await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
            await reporting.batteryVoltage(endpoint);
            await endpoint.configureReporting('genPowerCfg', payload);
        },
```
and the voltage is reported in the same payload as the occupancy so i though if i can make that report faster maybe the occupancy will be reported faster also.

sjorge, 2021-01-07:
> and the voltage is reported in the same payload as the occupancy so i though if i can make that report faster maybe the occupancy will be reported faster also.

That's sadly not how it works. The occupancy trigger comes from an IAS message:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/124b9fface70e399236bd2665c4da3a4b3a2bc1e/converters/fromZigbee.js#L732

The battery report is handled this way: 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/124b9fface70e399236bd2665c4da3a4b3a2bc1e/converters/fromZigbee.js#L250

The later being a attributeReport to the genPowerCfg cluster, the former being a command commandStatusChangeNotification to ssIasZone cluster.

Triggering faster reports to the genPowerCfg cluster has no effect on when the commands are send to ssIasZone cluster.

Westeru, 2021-01-07:
That is sad indeed.
Thank you for the answers, i will mark it as close.

closed 2021-01-07 by Westeru

grampasso-lento, 2025-09-04:
I have implemented a workaround for the 60 seconds cool down time of the sensor. I use the sensor to turn on a light which is typicall on for a short interval, say 20 seconds. Here is the automation that will implement this use case:

alias: Pantry Light
description: Pantry Light
triggers:
  - entity_id:
      - binary_sensor.PantrySensor
    to: "off"
    for:
      hours: 0
      minutes: 0
      seconds: 30
    id: notdetected
    trigger: state
  - entity_id:
      - binary_sensor.PantrySensor
    to: "on"
    id: detected
    trigger: state
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - notdetected
        sequence:
          - data: {}
            target:
              entity_id: switch.PantryLight
            action: switch.turn_off
      - conditions:
          - condition: trigger
            id:
              - detected
        sequence:
          - data: {}
            target:
              entity_id: switch.PantryLight
            action: switch.turn_on
          - action: python_script.set_state
            data:
              entity_id: binary_sensor.PantrySensor
              state: "off"
mode: restart

You need to add the python script integration to make it work