Koenkk/zigbee2mqtt #5610: TS0216

Kalypox, 2021-01-07:
## What happened
My TS0216 isn't recognized.
I bought it here: https://fr.aliexpress.com/item/1005001383368552.html

## What did you expect to happen
Add support for this TS02016.

## How to reproduce it (minimal and precise)
Just pair the device

##Infos
I saw this: https://github.com/Koenkk/zigbee2mqtt/issues/4655
but my device report this:
[image]
I tried to reboot, remove/create new docker, repair the device, but same...
I have this in logs:

```
Zigbee2MQTT:warn  2021-01-07 15:57:54: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2021-01-07 15:57:54: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-01-07 15:57:54: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2021-01-07 15:57:54: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-01-07 15:57:54: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2021-01-07 15:57:54: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-01-07 15:57:54: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2021-01-07 15:57:54: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:info  2021-01-07 15:57:54: Configuring '0xbc33acfffe36d8ee'
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'homeassistant/sensor/0xbc33acfffe36d8ee/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe36d8ee"],"manufacturer":"TuYa","model":"Sound and flash siren (TS0216)","name":"0xbc33acfffe36d8ee","sw_version":"Zigbee2MQTT 1.17.0"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","name":"0xbc33acfffe36d8ee battery","state_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","unique_id":"0xbc33acfffe36d8ee_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'homeassistant/binary_sensor/0xbc33acfffe36d8ee/alarm/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe36d8ee"],"manufacturer":"TuYa","model":"Sound and flash siren (TS0216)","name":"0xbc33acfffe36d8ee","sw_version":"Zigbee2MQTT 1.17.0"},"json_attributes_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","name":"0xbc33acfffe36d8ee alarm","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","unique_id":"0xbc33acfffe36d8ee_alarm_zigbee2mqtt","value_template":"{{ value_json.alarm }}"}'
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'homeassistant/sensor/0xbc33acfffe36d8ee/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe36d8ee"],"manufacturer":"TuYa","model":"Sound and flash siren (TS0216)","name":"0xbc33acfffe36d8ee","sw_version":"Zigbee2MQTT 1.17.0"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","name":"0xbc33acfffe36d8ee linkquality","state_topic":"zigbee2mqtt/0xbc33acfffe36d8ee","unique_id":"0xbc33acfffe36d8ee_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:error 2021-01-07 15:57:54: Failed to configure '0xbc33acfffe36d8ee', attempt 1 (undefined)
Zigbee2MQTT:info  2021-01-07 15:57:54: Configuring '0xbc33acfffe36d8ee'
Zigbee2MQTT:error 2021-01-07 15:57:54: Failed to configure '0xbc33acfffe36d8ee', attempt 2 (undefined)
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:54: Configuring '0xbc33acfffe36d8ee'
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:error 2021-01-07 15:57:54: Failed to configure '0xbc33acfffe36d8ee', attempt 3 (undefined)
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:54: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:55: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":159}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":150}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":150}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":147}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"duration":60,"linkquality":147}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":162,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"linkquality":162,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":171,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":171,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:56: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":171,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:57: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":168,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:57: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":168,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:57: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":168,"voltage":3500,"volume":30}'
Zigbee2MQTT:info  2021-01-07 15:57:57: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe36d8ee', payload '{"alarm":false,"battery":5,"linkquality":168,"voltage":3500,"volume":30}'
[... cut ...]
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 387)

```

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: PiZiGate
Adapter firmware version: 3.1d 

Thanks

closed 2021-02-15 by github-actions[bot]

Kalypox, 2021-02-16:
The problem is still here, how can i open the issue please ?

Koenkk, 2021-02-17:
Have you tried **force** removing it (by clicking the red bin icon) and repair it?