Koenkk/zigbee2mqtt #5614: Since 1.17.0, energy  of Mi power plug ZigBee EU (ZNCZ04LM) does not increase anymore

baderas, 2021-01-07:
## What happened
Prior to 1.17.0, the energy field of my Mi power plug ZigBee EU (ZNCZ04LM) increased as power was used by a device connected to it. After the upgrade 1.17.0, this field does not increase anymore (this was 6 days ago).
The field is stuck at the same value (66.88) all the time and does not increase anymore, although the device connected is drawing power.
There seems to be a new field, called "consumption" that does still count up (added by #1874).
My Zigbee plugs send both fields:
`{"consumer_connected":true,"consumption":81.7,"current":0.5172,"energy":66.88,"last_seen":"2021-01-07T21:06:44+01:00","linkquality":66,"power":56.65,"state":"ON","temperature":28,"voltage":229}`

Here you can see both fields in Grafana (same query, just the field name changed):

## What did you expect to happen
The field counts up as before or is removed from code and documentation.

## How to reproduce it (minimal and precise)
Use a ZNCZ04LM Plug and see how the energy and the consumption field behave.

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: sleash's stick
Adapter firmware version: current

baderas, 2021-01-08:
I also tried a new ZNCZ04LM (never used before), energy is never reported compared to my old ZNCZ04LM that was in use before 1.17.0:
Old one:
`{"consumer_connected":true,"consumption":81.7,"current":0.5172,"energy":66.88,"last_seen":"2021-01-07T21:06:44+01:00","linkquality":66,"power":56.65,"state":"ON","temperature":28,"voltage":229}`

New one:
`{"consumer_connected":false,"consumption":1.84,"current":0.6996,"last_seen":"2021-01-08T18:53:24+01:00","led_disabled_night":true,"linkquality":108,"power":140.62,"state":"ON","temperature":21,"voltage":219}`

It seems that the old one just keeps sending the old pre 1.17.0 energy value forever, the new one never sends any energy value.

Koenkk, 2021-01-08:
Woops, energy was indeed removed. Added it back.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-08 by Koenkk

baderas, 2021-01-08:
Is consumption still available? Whats the difference between consumption and energy? They seem to be the same to me?

Koenkk, 2021-01-08:
They are 100% the same, consumption is still available but deprecated.

JAJBrouwers, 2021-02-01:
@Koenkk 

I also have this problem, it whas working fine. I have update to 1.17.0.1

The energy value stay s the same.
In the mqtt msg it's showing... But always the same value.

NFO: MQTT publish: topic 'zigbee2mqtt/tv wand plug', payload '{"consumption":5.33,"current":0.6682,"energy":1.27,"linkquality":11,"power":147.56,"state":"ON","temperature":21,"voltage":224}'