Koenkk/zigbee2mqtt #5615: Physical Interaction flag

dvbit, 2021-01-07:
I would like to have a way to understand if a device has been actuated physically or through GUI/Automation.
I know that in the first case there's no mqtt set message and this could be "reconstructed" but it would be nice to have it already in a "side property" of the value.

Could be useful in automations where a "human override" feature is desired.

sjorge, 2021-01-07:
Seems like this will be hard to implement.

There is no way, AFAIK to differentiate an atttibuteReport that was the result of a remote telling a bulb to turn on vs one that happen because we told the bulb via the coordinator.

Koenkk, 2021-01-08:
As @sjorge I don't think that such feature is feasible.

dvbit, 2021-01-09:
Thanks for the feedback
I  thought that having access at the direct communication between controller and automation software you had a privileged hook / information.

closed 2021-01-09 by dvbit