Koenkk/zigbee2mqtt #5617: Sonoff SNZB-03 Motion Sensor Lost Battery Level

srtxema, 2021-01-08:
I have zigbee2mqtt in debian with mosquitto.
i have several Sonoff Zigbeet devices work very well but 1 of then don´t show Battery Level.
this is the information I obtain:
{
  "battery_low": false,
  "linkquality": 25,
  "occupancy": true,
  "tamper": false
}

this is the information I receive for rest of Motion Sensor SNZB-03
{
  "battery": 100,
  "battery_low": false,
  "linkquality": 25,
  "occupancy": true,
  "tamper": false,
  "voltage": 3200
}

{"date_code":"20201026","definition":
{"description":"Motion sensor","exposes":[
{"access":1,"description":"Indicates whether the device detected occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},
{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},
{"access":1,"description":"Indicates whether the device is tampered","name":"tamper","property":"tamper","type":"binary","value_off":false,"value_on":true},
{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},
{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SNZB-03","vendor":"SONOFF"},"endpoints":
{"1":
{"bindings":[
{"cluster":"genPowerCfg","target":
{"endpoint":1,"ieee_address":"0x00212effff066630","type":"endpoint"}}],"clusters":
{"input":["genBasic","genIdentify","ssIasZone","genPowerCfg"],"output":["genIdentify"]},"configured_reportings":[
{"attribute":"batteryVoltage","cluster":"genPowerCfg","maximum_report_interval":62000,"minimum_report_interval":3600,"reportable_change":0}]}},"friendly_name":"Sensor_Presencia_Sonoff_3","ieee_address":"0x00124b002266f58d","interview_completed":true,"interviewing":false,"model_id":"MS01","network_address":53596,"power_source":"Battery","supported":true,"type":"EndDevice"},

P3nkiln, 2021-01-09:
I'm also having this issue. I have a 1 motion sensor and 1 door sensor that report correctly and 1 motion sensor and 1 door sensor that don't. Same make and models.

sjorge, 2021-01-09:
You can try to apply the Battery reporting again via the frontend timing can be tricky as the device needs to be awake for it to be (re)configured.

P3nkiln, 2021-01-09:
Where do I reconfigure it?

sjorge, 2021-01-09:
In the frontend, there is now a reporting tab when you view a device. Normally the entry should already be there, just hit apply and make the device up.

srtxema, 2021-01-09:
I don’t find the option of reconfigure neither

sjorge, 2021-01-09:
I’m running dev, not sure it’s in stable yet.

~ sjorge

ARafaelSF, 2021-01-09:
The option is a gear icon, when you are in view of device.
But I have same problem. Reconfigure din't work.

ARafaelSF, 2021-01-09:
For me I have this log when try reconfigure.

ERROR: Failed to configure 'Contato Garagem Maior', attempt 2 (Error: Bind 0x00124b0022435cf3/1 genPowerCfg from '0x00124b0014d956a1/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.17.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7))
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Contato Garagem Maior"},"error":"Failed to configure (Bind 0x00124b0022435cf3/1 genPowerCfg from '0x00124b0014d956a1/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"u83jz-5"}'

sjorge, 2021-01-10:
No the gear icon doesn’t work indeed, you need to do the reporting separately and wake up the device at the right time.

You can also do it over mqtt but the frontend is way easier as you can’t just hit the apply button. 

https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure_reporting

~ sjorge

srtxema, 2021-01-10:
this is the information of zigbee2mqtt log without battery info
[image]

and this from zigbee2mqtt frontend
[image]

closed 2021-02-18 by github-actions[bot]

themad98, 2021-04-13:
> this is the information of zigbee2mqtt log without battery info
> [image]
> 
> and this from zigbee2mqtt frontend
> [image]
> [image]
> Information of reporting tab is the same of rest of devices
> [image]

Having exactly the same issue. Did you solve it?

MadsOBH, 2022-01-16:
I fixed my problem with a reporting reconfig.

A device with working battery reporting, (default i haven't touch anything).
[image]

A device thats not reporting battery information.
[image]

Same device after reconfig reporting, VOLTAGE and BATTERY PERCENT is working.
[image]