Koenkk/zigbee2mqtt #5618: Entity_Not_Found error after assigning friendly names

lazynomad, 2021-01-08:
## What happened
After assigning a friendly name to Ikea Fyrtur blinds, I get "Entity_Not_Found" error. Frontend shows the friendly name, however all the actions throw "entity_not_found" error. Zigbee2Mqtt is setup as a hassio add-on.
Debug logs after issuing "Open" request: [https://pastebin.com/FEMRbaFr](url)

## What did you expect to happen
Before the friendly name is assigned, all the actions work as expected.

## How to reproduce it (minimal and precise)
Pair an Ikea Fyrtur blind, rename from the IEEE address to a friendly name. The device stops working.

## Debug info
Zigbee2MQTT version:
Adapter hardware: Conbee II
Adapter firmware version:

Koenkk, 2021-01-08:
The friendly_name is not allowed to end with `/center` (since this can also be an endpoint name). All below are not allowed:

```js
[
    'left', 'right', 'center', 'bottom_left', 'bottom_right', 'default',
    'top_left', 'top_right', 'white', 'rgb', 'cct', 'system', 'top', 'bottom', 'center_left', 'center_right',
    'ep1', 'ep2', 'row_1', 'row_2', 'row_3', 'row_4', 'relay',
    'l1', 'l2', 'l3', 'l4', 'l5', 'l6', 'l7', 'l8',
    'button_1', 'button_2', 'button_3', 'button_4', 'button_5',
    'button_6', 'button_7', 'button_8', 'button_9', 'button_10',
    'button_11', 'button_12', 'button_13', 'button_14', 'button_15',
    'button_16', 'button_17', 'button_18', 'button_19', 'button_20',
    'button_light', 'button_fan_high', 'button_fan_med', 'button_fan_low',
    'heat', 'cool', 'water',
]
```

This was however not properly checked but has been fixed now. I suggest you rename your device to `living_room/blinds_center`.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-08 by Koenkk

lazynomad, 2021-01-08:
Thank you. That seems to work for the new devices. 
Is there a way to fix the name for the existing corrupted device ? Or is the only way to re-pair it ? I cannot find a clean way to delete the corrupted (in terms of name) device as I get the same "entity_not_found" error.

Koenkk, 2021-01-08:
Stop zigbee2mqtt and change name manually in the configuration.yaml

lazynomad, 2021-01-08:
Sorry if it's a basic question.  I am new to Zigbee2Mqtt.
When you say configuration.yaml, do you mean devices.yaml because that is where I see the devices and their friendly names. I can access the file over the ssh-addon of hassio.

Koenkk, 2021-01-08:
yes, devices.yaml