Koenkk/zigbee2mqtt #5619: Touchlink and Conbee2 (not supported?)

thackel, 2021-01-08:
Trying to factory reset older phillips hue bulbs (9290012573).
zigbee2mqqt: 1.17.0-dev 7b3beaf
coordinator: Conbee2 0x26660700

After sending `zigbee2mqtt/bridge/config/touchlink/factory_reset` I get the following error message:

```
Zigbee2MQTT:error 2021-01-08 08:30:38: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: not supported
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:895:19)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:23:12)
    at DeconzAdapter.restoreChannelInterPAN (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:894:16)
    at Touchlink.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/touchlink.js:147:36)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/touchlink.js:25:65)
    at runMicrotasks (<anonymous>))
```

thackel, 2021-01-08:
I got a reply from Dresden Elektronik on their discord server regarding this issue:
Touchlink is only supported with Conbee 1 and Raspbee 1, not the **2**.

It seems that I have to use the CC2531 to factory reset the Phillips Hue Bulbs... :-/

closed 2021-01-08 by thackel

MattWestb, 2021-01-08:
Im lucky having one raspbee 1 and TL is working great with the old "Wireless Light Control" app but you can trying https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3485 for trying getting it implanted by the company (If they is starting listening on their customers).