Koenkk/zigbee2mqtt #5620: TuYa valve TS0601: Report and enable time setting

kremik, 2021-01-08:
## Is your feature request related to a problem? Please describe
Valves are acting sometimes strange and this is caused because their time is somehow different so it doesn't match with the auto schedule. I don't know why. Time setting via the valve touch interface is not accurate. So having the time exposed via the interface would be great for checking the actual time and for some auto-adjustments via eg. node-red, etc.

## Describe the solution you'd like
Expose hour, minute and day via mqttt attribute and allow to set it via /set topic. 

## Describe alternatives you've considered
I don't see any.

wsobczak, 2021-01-17:
same here, time out of sync on every ts601 device, the only way to properly set time back is to restart valve almost every 2-3 days

wmaliga, 2021-01-19:
I encountered same problem, it seems like time is 2h late.

aegjoyce, 2021-01-26:
I think this issue affects all TS0601-based devices. I have the same issue with x2 Moes BHT-002-GCLZB wall thermostats. Looks like there's a genTime cluster but I don't know where to start with trying to work with it!

See here: https://github.com/Koenkk/zigbee2mqtt/issues/4185#issuecomment-741240615

Koenkk, 2021-01-26:
The latest dev branch contains a fix to send the time every hour (implemented in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2124 by @ayufan). (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ayufan, 2021-01-26:
Yup. It works finally, my schedules are executed properly now.

kremik, 2021-01-26:
> The latest dev branch contains a fix to send the time every hour (implemented in [Koenkk/zigbee-herdsman-converters#2124](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2124) by @ayufan). (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Great to hear that :-) Do you plan to expose the current device time in state message?

ayufan, 2021-01-26:
@kremik I don't think we have a way to get a current device time yet. But, I'm forcing to update it periodically to ensure that it is always in sync.

aegjoyce, 2021-01-30:
Have just switched to Z2M Edge but still problems with time sync on my 2 BHT-002-GCLZB thermostats. Have checked my logs and can't find any records for 'commandSetTimeRequest'. How can I help to debug this?

ayufan, 2021-01-30:
@aegjoyce 

> commandSetTimeRequest

You will not see this. You will see only periodic `manuSpecificTuya.setTime`.

My change covers only `TS0601_thermostat`. Unsure how the one you mentioned is presented. I cannot find it here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js.

aegjoyce, 2021-01-30:
It's there, just too far down to be seen on normal Github view. Somewhere around line 12000:

```
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration, tz.moes_thermostat_min_temperature],
        exposes: [e.child_lock(), exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE)
            .withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
            .withPreset(['hold', 'program'])],
        onEvent: tuya.onEventSetTime,
```

Is the problem that last line? It says `tuya.onEventSetTime` whereas I see your fix references `tuya.onEventSetLocalTime`

ayufan, 2021-01-30:
> It's there, just too far down to be seen on normal Github view 

Oh. This explains why :)

Yes. We might want to add the same approach for `tuya.onEventSetTime` and see if this helps:
 https://github.com/Koenkk/zigbee-herdsman-converters/pull/2124/files

kremik, 2021-01-30:
I can confirm that with the dev branch it works ok and time is synced.

aegjoyce, 2021-01-31:
I appreciate whilst the chipset is the same, the device is different, and I don't mean to hijack this issue, but thought the following might be useful for implementation:

I can confirm that, using the normal (not dev) channel, time syncing works with the BHT-002 if you add `tuya.onEventSetLocalTime` to its entry in devices.js:

```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
        model: 'BHT-002-GCLZB',
        vendor: 'Moes',
        description: 'Moes BHT series Thermostat',
        fromZigbee: [fz.moes_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.moes_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration, tz.moes_thermostat_min_temperature],
        exposes: [e.child_lock(), exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 1).withLocalTemperature()
            .withSystemMode(['off', 'heat']).withRunningState(['idle', 'heat', 'cool']).withPreset(['hold', 'program'])],
        onEvent: tuya.onEventSetLocalTime,
    },
```

However, the time that gets synced is 15 minutes behind the actual time (constant interval) and I think it's resetting the day counter as well (will confirm tomorrow).

I have not been able to get a custom devices.js working the the dev channel so can't confirm whether the latest commits fix this - seems weird that it would be updating with a time that's 15 minutes off?

ayufan, 2021-01-31:
@aegjoyce 

AFAIK You cannot use `.onEventSetLocalTime`, as these devices do parse `timespec` differently. You need to apply the same change to `.onEventSetTime` as it is for `.onEventSetLocalTime`.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L68 vs https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js#L97

aegjoyce, 2021-01-31:
Understood. So what we need is a tweaked version of your commit to work for `.onEventSetTime` - something like this?

```
async function onEventSetTime(type, data, device) {
    const nextTimeUpdate = globalStore.getValue(device, 'nextTimeUpdate');
    const forceTimeUpdate = nextTimeUpdate == null || nextTimeUpdate < new Date().getTime();
    
    if (data.type === 'commandSetTimeRequest' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
        globalStore.putValue(device, 'nextTimeUpdate', new Date().getTime() + 3600 * 1000);{
        
            try {
            const utcTime = Math.round(((new Date()).getTime() - constants.OneJanuary2000) / 1000);
            const localTime = utcTime - (new Date()).getTimezoneOffset() * 60;
            const endpoint = device.getEndpoint(1);
            const payload = {
                payloadSize: 8,
                payload: [
                    ...convertDecimalValueTo4ByteHexArray(utcTime),
                    ...convertDecimalValueTo4ByteHexArray(localTime),
                ],
            };
            await endpoint.command('manuSpecificTuya', 'setTime', payload, {});
        } catch (error) {
            // endpoint.command can throw an error which needs to
            // be caught or the zigbee-herdsman may crash
            // Debug message is handled in the zigbee-herdsman
        }
    }
}
```

I'm really sorry but with my current setup I can only test tweaks to devices.js, I can't test any other modifications to zigbee-herdsman-converters - so I can only propose stuff for others to test :(

ayufan, 2021-01-31:
> I can't test any other modifications to zigbee-herdsman-converters - so I can only propose stuff for others to test :(

Why? `devices.js` is also part of `zigbee-herdsman-converters`.

aegjoyce, 2021-01-31:
I'm running Z2M from the Home Assistant add-on rather than through Docker so don't have access to edit core config files - I can only make a secondary devices.js and refer to it using `zigbee_shepherd_devices: true`

aegjoyce, 2021-01-31:
Ok, getting closer. Managed to get my `devices.js` working on the Dev/Edge channel.

I still can't get my thermostats to sync any sort of time when using `onEventSetTime` but `onEventSetLocalTime` now looks to be actively updating them with the correct time - no time offset or anything. Part of this might be because Dev/Edge just bumped Herdsman to 14.0.40 and the last time I tried it was only using 14.0.3.

The only issue now is it isn't syncing the correct weekday - listing Saturday when it's currently Sunday - however I expect this is because of what you've stated above about the differences in the way `onEventSetTime` and `onEventSetLocalTime` calculate the timestamp to send.

I will submit the above as a new pull request and hope that it fixes the issue when it eventually ends up on the dev channel! Thanks for your help

ayufan, 2021-01-31:
@aegjoyce 

> I will submit the above as a new pull request and hope that it fixes the issue when it eventually ends up on the dev channel! Thanks for your help

I think someone needs to test that first before it can get merged. I'm unsure if this would not break anything else. My change was only affecting thermostat that I had.

RubenKelevra, 2021-02-05:
> same here, time out of sync on every ts601 device, the only way to properly set time back is to restart valve almost every 2-3 days

Strange, the time works great for me - while I don't use it.

But I've added 3V power supplies to them. Are you using rechargeable batteries?

wsobczak, 2021-02-05:
> 
> 
> > same here, time out of sync on every ts601 device, the only way to properly set time back is to restart valve almost every 2-3 days
> 
> Strange, the time works great for me - while I don't use it.
> 
> But I've added 3V power supplies to them. Are you using rechargeable batteries?

no, im using normal  batts
but using latest dev branch everything is ok, time is synced

regards

kremik, 2021-02-05:
> > same here, time out of sync on every ts601 device, the only way to properly set time back is to restart valve almost every 2-3 days
> 
> Strange, the time works great for me - while I don't use it.
> 
> But I've added 3V power supplies to them. Are you using rechargeable batteries?

Upgrade to 1.17.1 it's fixed there already.

closed 2021-02-05 by kremik