Koenkk/zigbee2mqtt #5626: feature request: find __newly__ supported devices via zigbee2mqtt.io

prankousky, 2021-01-08:
Hi everybody,

this is **not** related to a problem, but rather a feature regarding the [official website](https://zigbee2mqtt.io).

## Describe the solution you'd like

Would it be possible to sort the [supported devices](https://www.zigbee2mqtt.io/information/supported_devices.html) by different parameters, specifically by the date they had been added?

I gotta admit, sometimes when I am bored, I just browse through the supported devices trying to see if there is anything I could add to my smart home setup just for the fun of it. As of today, there are **1217** devices available. And (fortunately 🙂) the list keeps growing. But this makes it harder to discover new devices that had not been there when one was looking the last time.

Is there anybody else who does this, or am I the only one?

Koenkk, 2021-01-08:
> specifically by the date they had been added?

We don't store this date so that's currently not possible. However newly added devices are always added to the changelogs: https://github.com/Koenkk/zigbee2mqtt/releases

closed 2021-02-15 by github-actions[bot]

prankousky, 2021-11-05:
I just wanted to ask if it would be possible to integrate storing the date in future additions to the devices list. Haven't been on the website in a while, as I followed your (@Koenkk 's) advice and bookmarked the link from above; however, the [supported devices](https://www.zigbee2mqtt.io/supported-devices/) page has gotten a huge update/redesign since I last visited it...

Perhaps this suggested feature could be implemented there, if anybody else found it useful as well? There is nothing wrong with using the [changelogs](https://github.com/Koenkk/zigbee2mqtt/releases), of course. But now that the devices page looks even better, it'd be great being able to browse that page for newly added devices.