Koenkk/zigbee2mqtt #5627: Toggle not working on Muller Licht RGB Bulb

ambarusa, 2021-01-08:
## What happened

I was trying to control a Muller Licht (https://www.zigbee2mqtt.io/devices/404000_404005_404012.html) RGB bulb via MQTT messages. 

## What did you expect to happen

The toggle command was turning on the bulb with the lowest brightness, and because of this, I couldn't verify, if the transition was also processed.

## How to reproduce it (minimal and precise)
zigbee2mqtt/<LIGHT_NAME>/set
{
  "state": "toggle",
  "brightness": 188,
  "transition": 2
}

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: CC2531
Adapter firmware version: 
20190608

sjorge, 2021-01-08:
I think this is because you have both toggle and brightness specified.

Setting a brightness will turns on the bulb at that brightness, which might conflict with toggle.

Does it work if you do them separately?

ambarusa, 2021-01-08:
Thanks for your feedback. Here's how it works now:

- if I set the bulb to ON - 255, then turn off, then send a toggle: the bulb turns back on with minimum brightness (or very low, it is not refreshing in MQTT traffic)
- if I set the bulb to ON - 255, then turn off by toggle, then send a toggle again, the bulb turns back on with the previous  brightness
Transition still doesn't seem to work, when the bulb is turned off by toggle for example.
I tried to set the brightness, to force this somehow, and set it to the desired level. I thought it's similar to the Home Assistant implementation, where you can set the same attributes for light.toggle and light.turn_on

-

closed 2021-02-15 by github-actions[bot]