Koenkk/zigbee2mqtt #5628: Muller Licht Remote Groups not working properly

ambarusa, 2021-01-08:
## What happened

I was making automations using the inputs from the Muller Licht (https://www.zigbee2mqtt.io/devices/MLI-404011.html) Licht Tint remote control.

## What did you expect to happen

According to the documentation, there are four groups, All of them are working fine, having their own action_group ID, except for the one (group 0), when all three leds are ON on the remote. In this case no messages are received (checked also in debug log level).

> This remote is able to deal with 4 groups: group0 = All three leds lit group1 = first led lit group2 = second led lit group3 = third let lit

## How to reproduce it (minimal and precise)

Set the remote on group0 (All three leds lit), and strart pushing buttons.

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: CC2531
Adapter firmware version:
20190608

merlinschumacher, 2021-01-21:
I can confirm the issue.

merlinschumacher, 2021-01-21:
Using the binding feature in the webinterface makes the remote atleast work with the lamps, but no messages appear in zigbee2mqtt.

henkiejan1, 2021-01-24:
Ooh, good to read. It makes me frustrating i did not got it working. It works well with the the three leds and all the leds. I thought also that group0 original all the groups together should. So i think if you bind the first group it should be bound to group1 the next group to group2 and maybe group3. And when you select group0 you can control all the groups... ?

Koenkk, 2021-01-24:
@merlinschumacher this is probably because reporting is not enabled (https://www.zigbee2mqtt.io/information/report.html). Note that the report will be a deprecated feature from the next release as reporting is automatically setup when doing binding. You can test this in the dev branch: (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) Note 2: if it doesn't work after the bulb probably doesn't support it.

closed 2021-03-03 by github-actions[bot]

Albarion, 2021-04-02:
I can confirm this. If group0 is selected all buttons except the scene buttons publish messages.  With group 1 ,2 or 3 selected ,messages for the scene buttons are published.