Koenkk/zigbee2mqtt #5630: TS0601_thermostat, Entity 'livingroom/heat/right' is unknown

Zagitta, 2021-01-08:
## What happened
I've recently bought 3 additional TS0601_thermostats after having been satisfied with my initial purchase of 1.
However having added them to my network I'm unable to control 2 of them while the third works fine.
Upon sending any commands or removal request to the TRVs (livingroom/heat/right & livingroom/heat/left) I get the following error: `ERROR: Entity 'livingroom/heat/right' is unknown` despite the state of the TRVs having joined the network and updating just fine.

[image]

Here's the debug log when trying to change temperature:

```
DEBUG: Received Zigbee message from 'livingroom/heat/right', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,4],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":26}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/livingroom/heat/right', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":25.5,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":84,"local_temperature":26,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":0,"preset":"manual","system_mode":"auto","update":{"state":"idle"},"update_available":false,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
DEBUG: Received Zigbee message from 'livingroom/heat/right', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,4],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":27}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/livingroom/heat/right', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":25.5,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":84,"local_temperature":26,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":0,"preset":"manual","system_mode":"auto","update":{"state":"idle"},"update_available":false,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
.....
DEBUG: Received MQTT message on 'zigbee2mqtt/livingroom/heat/right/set/current_heating_setpoint' with data '35.0'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"livingroom/heat/right"},"type":"entity_not_found"}'
ERROR: Entity 'livingroom/heat/right' is unknown
DEBUG: Received Zigbee message from 'livingroom/heat/right', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,4],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":54}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/livingroom/heat/right', payload '{"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":1,"away_preset_temperature":15,"battery_low":false,"boost_time":300,"child_lock":"UNLOCKED","comfort_temperature":20,"current_heating_setpoint":25.5,"eco_temperature":15,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}],"linkquality":114,"local_temperature":26,"local_temperature_calibration":-1,"max_temperature":35,"min_temperature":5,"position":40,"preset":"manual","system_mode":"auto","update":{"state":"idle"},"update_available":false,"week":"5+2","window_detection":"OFF","window_detection_params":{"minutes":10,"temperature":5},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":15},{"hour":11,"minute":30,"temperature":15},{"hour":12,"minute":30,"temperature":15},{"hour":17,"minute":30,"temperature":20},{"hour":22,"minute":0,"temperature":15}]}'
DEBUG: Received Zigbee message from 'livingroom/heat/right', type 'commandSetTimeRequest', cluster 'manuSpecificTuya', data '{"payload":[],"payloadSize":0}' from endpoint 1 with groupID 0
```

and for device removal with force:

```
DEBUG: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"livingroom/heat/right","transaction":"1e2cv-1"}'
ERROR: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Device 'livingroom/heat/right' does not exist'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Device 'livingroom/heat/right' does not exist","status":"error","transaction":"1e2cv-1"}'
```

My network map looks like this:
[image]

## What did you expect to happen
I expect to be able to change the state of the TRVs.

## How to reproduce it (minimal and precise)

1. Pair TRV
2. After successful pair, send temperature command to `<friendly_name>/set/current_heating_setpoint`
3. Observe failure

## Things I have tried without success:

1. Reparing both TRVs
2. Remove the offending TRVs manually from `database.db` and `state.json` then reparing.
3. Reparing TRV in a different room where the first purchased TRV resides and works.

## Debug info
Zigbee2MQTT version: 1.17.0 (hassio addon 1.17.0.1)
Adapter hardware: zig-a-zig-ah! (CC2652)
Adapter firmware version: coordinator type = zStack3x0 & coordinator revision = 20200805
Above logs were captured with log level DEBUG and `zigbee_herdsman_debug: true` in the hassio addon configuration.

Zagitta, 2021-01-08:
database.db:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b00216005f3","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x0017880102c28848","nwkAddr":36328,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"RWL021","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0],"outClusterList":[0,3,4,6,8,5],"clusters":{"genBasic":{"attributes":{"modelId":"RWL021","manufacturerName":"Philips","powerSource":3,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190410","swBuildId":"6.1.1.28573"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":12,"inClusterList":[0,1,3,15,64512],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":156}}},"binds":[{"cluster":64512,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190410","swBuildId":"6.1.1.28573","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1610102082654}
{"id":3,"type":"Router","ieeeAddr":"0x14b457fffe2ea339","nwkAddr":29398,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS opal 1000lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 WS opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":2,"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20200312","swBuildId":"2.0.029"}},"genLevelCtrl":{"attributes":{"currentLevel":3}},"genOnOff":{"attributes":{"onOff":0}},"lightingColorCtrl":{"attributes":{"currentX":30138,"currentY":26909,"colorTemperature":454,"colorCapabilities":16}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005f3","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20200312","swBuildId":"2.0.029","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1610102393590}
{"id":4,"type":"Router","ieeeAddr":"0xec1bbdfffea282b4","nwkAddr":12656,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"FLOALT panel WS 30x90","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"FLOALT panel WS 30x90","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":87,"hwVersion":1,"dateCode":"20200107","swBuildId":"1.2.224"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":2}},"lightingColorCtrl":{"attributes":{"colorTemperature":454,"colorCapabilities [... cut ...]
{"id":53,"type":"EndDevice","ieeeAddr":"0x847127fffe961620","nwkAddr":211,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610110109763}
```

LouiS22, 2021-01-08:
After pairing, reboot HA and try it again.

Zagitta, 2021-01-08:
Thank you for the suggestion, however I tried that without success already but just forgot to write it.

Koenkk, 2021-01-08:
Exactly the same as https://github.com/Koenkk/zigbee2mqtt/issues/5618 
 
Read: https://github.com/Koenkk/zigbee2mqtt/issues/5618#issuecomment-756958694

Zagitta, 2021-01-08:
Ah that fixed it, thank you!

closed 2021-01-08 by Zagitta