Koenkk/zigbee2mqtt #5632: The device DIYRuZ_rspm does not respond to commands from Zigbee2Mqtt

khaidakin, 2021-01-08:
## What happened
The DIYRuZ_rspm device passes the interview normally, connects to the network, but does not respond to commands from zigbee2mqtt

## What did you expect to happen
On earlier versions of z2m everything worked without problems, but on versions zigbee2mqtt@1.16.2 and later an error appears

## How to reproduce it (minimal and precise)
Connect the device DIYRuZ_rspm and try to control it via z2m

## Debug info
Zigbee2MQTT version: zigbee2mqtt@1.16.2
Adapter hardware: CC2538, CC2652-E72
Adapter firmware version: 20200921 (Egony)
https://pastebin.com/8r5YvnV4

Koenkk, 2021-01-08:
Does it correctly work with z2m 1.16.1?

khaidakin, 2021-01-08:
I just checked, on version 1.16.1 everything works fine

Koenkk, 2021-01-08:
Can you provide the same log as https://pastebin.com/8r5YvnV4 for z2m 1.16.1?

khaidakin, 2021-01-08:
Yes of course. Here you are https://pastebin.com/nQ4tyAYx

Koenkk, 2021-01-08:
- Commands are send in exactly the same way so this cannot be the issue. Looks like it's something in the adapter/firmware.
- Is this on the same adapter and firmware as the https://pastebin.com/raw/8r5YvnV4 logs?
- If yes, can you sniff the traffic of this: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html and share the sniff + network key here

khaidakin, 2021-01-08:
Yes, it's the same stick. At the moment I am doing this on a test machine with a cc2538 stick and firmware https://github.com/jethome-ru/zigbee-firmware/blob/master/ti/coordinator/cc2538_cc2592/JH_2538_2592_ZNP_USB_20201010.hex
I can't sniff the traffic, since I don't have equipment for this. Sorry.

khaidakin, 2021-01-08:
On another working Raspberry I have a stick CC2652-E72 with firmware version: 20200921 (Egony) installed and there is a similar problem with the DIYRuZ_rspm

Koenkk, 2021-01-08:
Sorry, but I don't see a way to debug this without sniff. From the logs it looks like the device does just not respond to the requests

khaidakin, 2021-01-08:
Here is another log on the 2652 stick on version 1.16.1 https://pastebin.com/6FfU9D8r
I can now update the z2m version to the latest and send a log with an error on the stick 2652

khaidakin, 2021-01-08:
Here's what I have done now. After the device worked fine for me on the stick 2652 and z2m 1.16.1, I updated the system to version 1.17.0. And after that everything worked for me again !!! But then I removed DIYRuZ_rspm from the network and reconnected. Here is the connection log: https://pastebin.com/AxMcBaHj And after reconnecting, I again have the same error: https://pastebin.com/GZAjsTZ1
And here you can clearly see that the problem is in z2m

Koenkk, 2021-01-08:
Can you try updating the entry in `devices.js` to below?

```js
{
    zigbeeModel: ['DIYRuZ_rspm'],
    model: 'DIYRuZ_rspm',
    vendor: 'DIYRuZ',
    description: '[DIYRuZ relay switch power meter](https://modkam.ru/?p=1309)',
    fromZigbee: [fz.diyruz_rspm],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.current(), e.cpu_temperature(), e.action(['hold', 'release'])],
    endpoint: (device) => {
        return {default: 8};
    },
},
```

khaidakin, 2021-01-09:
Everything works great again! Thank you very much!

closed 2021-01-09 by khaidakin