Koenkk/zigbee2mqtt #5634: IKEA Repeater issues

arduino43, 2021-01-08:
Hello,

It is my understanding that to use my 2 IKEA smart blinds I must use the IKEA repeater, which is causing issues on my network.

My problem is that the IKEA repeater becomes the router for several devices on the network when it is powered on, causing issues with most devices that connect to it; my Yale door lock works only a small percentage of the time, Xiaomi door sensors stop reporting and my IKEA blinds fail occasionally. 

1.) If I unplug the repeater, devices previously using it as a router start working, including the IKEA blinds.
2.) If I remove the device  from my configuration (zigbee2mqtt/bridge/config/remove) while it is powered on all of stop working.
3.) If I force remove the device from my configuration (zigbee2mqtt/bridge/config/force_remove) while it is unplugged, all devices stop working.

It seems like joining the device to the network and then unplugging it, but leaving it in the configuration, is working fine for me as it avoids devices using it as a router. 

Has anyone else experienced this issue?

Did you read the FAQ?
Yes

Zigbee2MQTT fails to start?
No

Having issues when using a CC2531?
No

Unsupported device?
No

Device does not pair or interview fails?
No

Bug report?
No

## Debug Info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0

IVI053, 2021-01-18:
You don't have to use the repeater. IKEA just states that you have to because the remotes as well as the blinds are zigbee end devices and at least one router is needed for it to work. You can imagine that they don't want to teach all their customers about zigbee so for simplicity they just state that you need the repeater.
If you already have an mains connected device like a bulb your network will just be fine without them.

arduino43, 2021-01-18:
> You don't have to use the repeater. IKEA just states that you have to because the remotes as well as the blinds are zigbee end devices and at least one router is needed for it to work. You can [...]

I wasn't referring to IKEA saying the repeater is required, I was referring to the documentation on the zigbee2mqtt.io supported device page.

"It’s recommend to first pair the included TRADFRI signal repeater in the room where you want to put the blinds. Now press both buttons on the blind (next to battery lid) for 5 seconds until a white light between both buttons is turned on. The device is now awake and ready to pair for 2 minutes. After that pair the blind by holding the blind battery lid very close to the TRADFRI signal repeater until the white light is turned off (this should mean the pairing is successful)."

It does not specifically say the repeater is required but the setup instructions seem to imply it is.

I am unable to join my blinds without using the repeater. I can unplug the repeater after joining the blinds and everything seems to work. If I remove the repeater from my configuration, the blinds stop working and cannot be joined.

christianzappa, 2021-01-19:
@arduino43 just out of interest, does your blind's remote still work?

arduino43, 2021-01-19:
@christianzappa The remotes had been troublesome at best so I haven't really used them.

One first press the button would announce, then after waiting 10-15 seconds I could use the button as normal for a few hours. It seems the device goes into some sort of power saving mode that other Zigbee buttons do not seem to do.

MattWestb, 2021-01-19:
@arduino43 The Open/Close remote and the blinds is little tricky then they is both sleeping end devices and sending the commands on one little "special" way and normal routers is not understanding it and the signal repeater is not correctly configured without touchlink (group binding of some cluster) that is made i the IKEA GW.
Some user have them working OK others have problems with the blinds.

That your remote have must rejoining its not normal as long the outers is behaving normal and not ageing there sleeping end device out to fast.

The signal repeater is having one good chip and also the PCB antenna is good and is "Xiaomi friendly" that many routing devices not is (no named no blamed).

IVI053, 2021-01-19:
I'm using two of them without a repeater. Works like a charm.

arduino43, 2021-01-19:
@IVI053 interesting - what adapter are you using? What version of zigbee2mqtt?

Thanks for your input

IVI053, 2021-01-20:
LAUNCHXL-CC26X2R1
z2m Version should not matter as I have them since 1.13 I think and bis 1.17. Never any problems

closed 2021-01-28 by arduino43

LeFurotte, 2022-01-30:
> Removed old config and repaired all devices resolved issue.

Hi @arduino43, can you explain what you mean when you "Removed old config and repaired all devices"?  I have weird issues.  I my case, I had 2 blinds that were working perfectly, but since I bought 2 more, I have problems with some blinds disconnecting? Thx in advance.

kapabel, 2022-09-27:
I am facing the same issue with this IKEA repeater. I do not use the buttons for the roller blind but for another application. I use the repeater to extend the network to the other side of the house. With the repeater in place and powered up however, the devices connected to it won't work for some reason. This behavior occurs sometimes not right away but only the next day. I disconnect the repeater and reboot everything, then the network behaves fine. Something fishy is going on with this repeater and I haven't yet reached the bottom of it.