Koenkk/zigbee2mqtt #5636: Zigbee2mqtt crashing during device pairing

yevgenb, 2021-01-08:
## What happened

Zigbee2mqtt crashing during device pairing. Started after upgrade from v 1.14.2 - worked fine before.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Adapter hardware: cc2530 + cc2591
Adapter firmware version: 1.17.0

yevgenb, 2021-01-08:
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/5788822/z2m.log)

yevgenb, 2021-01-08:
Configuration:

{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "mqtt",
    "password": "...."
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "log_level": "info",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [ ....  ],
    "availability_blocklist": [],
    "availability_passlist": []
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {
    "new_api": true
  }
}

Koenkk, 2021-01-08:
`Zigbee2MQTT:error 2021-01-08 10:17:45: Adapter disconnected, stopping` looks that the USB adapter disconnected, can happen for many reason. Suggestions on how to fix can be found under: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

yevgenb, 2021-01-08:
As I mentioned - it was working fine for 6 month and started to crash after upgrade to 1.17.0. The hardware is the same, no updates to OS.

yevgenb, 2021-01-08:
Forgot to mention - it's same device that was paired and worked with zigbee2mqtt before.

yevgenb, 2021-01-08:
Hmm - very strange, after plugging and unplugging stick it started to work again. Closing issue - thank you Koen!

closed 2021-01-08 by yevgenb