Koenkk/zigbee2mqtt #5641: Support for Lidl LivarnoLux Remote (HG06323, Zigbee Model TS1001)

mvaneijken, 2021-01-08:
This switch is bought as part of the Livarno Lux Starterset Zigbee Smart Home at Lidle in the Netherlands.

I tried to follow the steps in the How to support new devices, but got stuck on step 2.

## Information about the device + link
https://www.lidl-shop.nl/LIVARNO-LUX-Starterset-Zigbee-Smart-Home/p100306597

The device is allready reported as supported in the supported devices list, but this one seems to be different internally.
https://www.zigbee2mqtt.io/devices/FB20-002.html

## data/database.db entry of the device
{"id":12,"type":"EndDevice","ieeeAddr":"0x842e14fffe54804a","nwkAddr":53598,"manufId":4098,"manufName":"_TYZB01_bngwdjsr","powerSource":"Battery","modelId":"TS1001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS1001","manufacturerName":"_TYZB01_bngwdjsr","powerSource":3,"zclVersion":3,"appVersion":81,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":81,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1610133127212}

## Debug log:
Jan  8 20:54:23 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:23: Device '0x842e14fffe54804a' announced itself
Jan  8 20:54:23 zigbee2mqtt npm[135470]: #033[32mzigbee2mqtt:info #033[39m 2021-01-08 20:54:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x842e14fffe54804a"}}'
Jan  8 20:54:23 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:23: Saving state to file /opt/zigbee2mqtt/data/state.json
Jan  8 20:54:25 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:25: Received Zigbee message from '0x842e14fffe54804a', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:25 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:25: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:25 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:25: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:26 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:26: Received Zigbee message from '0x842e14fffe54804a', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:26 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:26: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:26 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:26: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:27: Received Zigbee message from '0x842e14fffe54804a', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:27: Received Zigbee message from '0x842e14fffe54804a', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:27: Received Zigbee message from '0x842e14fffe54804a', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:27 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:27: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:28: Received Zigbee message from '0x842e14fffe54804a', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:28: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:28: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:28: Received Zigbee message from '0x842e14fffe54804a', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:28: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:28 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:28: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:29: Received Zigbee message from '0x842e14fffe54804a', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:29: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[34mzigbee2mqtt:debug#033[39m 2021-01-08 20:54:29: Received Zigbee message from '0x842e14fffe54804a', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:29: Received message from unsupported device with Zigbee model 'TS1001'
Jan  8 20:54:29 zigbee2mqtt npm[135470]: #033[33mzigbee2mqtt:warn #033[39m 2021-01-08 20:54:29: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2021-01-08:
This device is already supported in Zigbee2mqtt 1.17.0: https://www.zigbee2mqtt.io/devices/FB20-002.html

closed 2021-01-08 by Koenkk

kmatokov, 2022-01-09:
it may be supported but not working for me, only seeing battery  lever, but no action on keypress.