Koenkk/zigbee2mqtt #5644: mqtt message false to zigbee2mqtt/bridge/request/permit_join not working

sebastianovide, 2021-01-09:
## What happened
I see a json in `zigbee2mqtt/bridge/response/permit_join` with 
`{
    "data": {
        "value": "false"
    },
    "status": "ok"
}`
I see a json in the `zigbee2mqtt/bridge/info` with `"permit_join": true,`

## What did you expect to happen
I expect to see a json in the `zigbee2mqtt/bridge/info` with `"permit_join": false,`

## How to reproduce it (minimal and precise)
sent the value false to `zigbee2mqtt/bridge/request/permit_join` using home assistant mqtt integration

## Debug info
Zigbee2MQTT version: `"version": "1.17.0"`
`    "coordinator": {
        "meta": {
            "maintrel": 1,
            "majorrel": 2,
            "minorrel": 7,
            "product": 1,
            "revision": 20200925,
            "transportrev": 2
        },
        "type": "zStack3x0"
    }`

Koenkk, 2021-01-09:
`zigbee2mqtt/bridge/response/permit_join` should be `zigbee2mqtt/bridge/request/permit_join`. If that doesn't fixes it please provide the debug log of this.

debuglog

sebastianovide, 2021-01-10:
@Koenkk . Correct, the message is sent to `zigbee2mqtt/bridge/request/permit_join`.

It is very strange, and probably Home Assistant problem (????)

After a lot of experiments this is what I've found:
1) all works as expected if the messages is send to mqtt using the mqtt client.
2) clicking on the switch (I've copied the definition from z2m site), it appears as it is sending the right message to mqtt. Even the response is correct. But the json from the json is wrong. See what I get from mqtt 

```
Listen to a topic
 
Topic to subscribe to
zigbee2mqtt/bridge/#
 
Message 231 received on zigbee2mqtt/bridge/response/permit_join at 14:10:
{
    "data": {
        "value": "False"
    },
    "status": "ok"
}
QoS: 0 - Retain: false
Message 230 received on zigbee2mqtt/bridge/logging at 14:10:
{
    "level": "info",
    "message": "MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{\"data\":{\"value\":\"False\"},\"status\":\"ok\"}'"
}
QoS: 0 - Retain: false
Message 229 received on zigbee2mqtt/bridge/info at 14:10:
{
    "commit": "07cdc9d",
    "config": {
        "advanced": {
            "adapter_concurrent": null,
            "adapter_delay": null,
            "availability_blacklist": [],
            "availability_blocklist": [],
            "availability_passlist": [],
            "availability_timeout": 60,
            "availability_whitelist": [
                "0x001e5e090229f2fb"
            ],
            "cache_state": true,
            "cache_state_persistent": true,
            "cache_state_send_on_startup": true,
            "channel": 11,
            "elapsed": true,
            "ext_pan_id": [
                221,
                221,
                221,
                221,
                221,
                221,
                221,
                221
            ],
            "homeassistant_discovery_topic": "homeassistant",
            "homeassistant_legacy_triggers": true,
            "homeassistant_status_topic": "hass/status",
            "last_seen": "ISO_8601_local",
            "legacy_api": true,
            "log_directory": "/app/data/log/%TIMESTAMP%",
            "log_file": "log.txt",
            "log_level": "info",
            "log_output": [
                "console",
                "file"
            ],
            "log_rotation": true,
[... 170 lines cut ...]
    "network": {
        "channel": 11,
        "extended_pan_id": "0xdddddddddddddddd",
        "pan_id": 6754
    },
    "permit_join": true,
    "version": "1.17.0"
}
QoS: 0 - Retain: false
Message 228 received on zigbee2mqtt/bridge/logging at 14:10:
{
    "level": "info",
    "message": "MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{\"commit\":\"07cdc9d\",\"config\":{\"advanced\":{\"adapter_concurrent\":null,\"adapter_delay\":null,\"availability_blacklist\":[],\"availability_blocklist\":[],\"availability_passlist\":[],\"availability_timeout\":60,\"availability_whitelist\":[\"0x001e5e090229f2fb\"],\"cache_state\":true,\"cache_state_persistent\":true,\"cache_state_send_on_startup\":true,\"channel\":11,\"elapsed\":true,\"ext_pan_id\":[221,221,221,221,221,221,221,221],\"homeassistant_discovery_topic\":\"homeassistant\",\"homeassistant_legacy_triggers\":true,\"homeassistant_status_topic\":\"hass/status\",\"last_seen\":\"ISO_8601_local\",\"legacy_api\":true,\"log_directory\":\"/app/data/log/%TIMESTAMP%\",\"log_file\":\"log.txt\",\"log_level\":\"info\",\"log_output\":[\"console\",\"file\"],\"log_rotation\":true,\"log_syslog\":{},\"pan_id\":6754,\"report\":true,\"rtscts\":false,\"soft_reset_timeout\":0,\"timestamp_format\":\"YYYY-MM-DD HH:mm:ss\"},\"ban\":[],\"blocklist\":[],\"device_options\":{},\"devices\":{\"0x000b57fffe2f87f3\":{\"debounce\":0.2,\"debounce_ignore\":[\"action\",\"brightness\"],\"friendly_name\":\"Piano Dimmer\",\"legacy\":false},\"0x000b57fffe346c9b\":{\"friendly_name\":\"Piano Lamp\",\"retain\":false},\"0x000d6f000e1b7b68\":{\"friendly_name\":\"osram switch\"},\"0x00124b0019430c3b\":{\"friendly_name\":\"Router 1\"},\"0x00124b001f669750\":{\"friendly_name\":\"Main Room Light Door\"},\"0x00124b002228e310\":{\"friendly_name\":\"sonoff switch\"},\"0x00124b00224194de\":{\"friendly_name\":\"Plug 1\"},\"0x00124b002242b0af\":{\"friendly_name\":\"Temp Laundry\"},\"0x00124b00226aab73\":{\"friendly_name\":\"Plug laundry heater\"},\"0x00158d0003f0565c\":{\"friendly_name\":\"Temp Office\",\"retain\":false},\"0x00158d0003f18415\":{\"friendly_name\":\"Temp Kitchen\",\"retain\":false},\"0x00158d0003fae39e\":{\"friendly_name\":\"Motion Kitchen Window\"},\"0x00158d0003faf1dd\":{\"friendly_name\":\"Motion Landing Upstairs\"},\"0x00158d00040592f6\":{\"friendly_name\":\"Motion Landing Downstairs\"},\"0x00158d000423d5d8\":{\"friendly_name\":\"Temp Living\",\"retain\":false},\"0x00158d000446239a\":{\"friendly_name\":\"Motion Entrance\",\"retain\":false},\"0x00158d00044cac01\":{\"friendly_name\":\"Motion Living\",\"retain\":false},\"0x001e5e090229f2fb\":{\"friendly_name\":\"Hive Signal Booster\"},\"0x086bd7fffedae4dd\":{\"friendly_name\":\"Landing Upstairs\"},\"0x90fd9ffffe1ef5cd\":{\"friendly_name\":\"kids 1\"},\"0xd0cf5efffe2ebfaa\":{\"friendly_name\":\"kids 2\"},\"0xd0cf5efffebeaa96\":{\"friendly_name\":\"Piano 1\"},\"0xd0cf5efffec8139e\":{\"friendly_name\":\"Piano 2\"},\"0xd0cf5efffef80e92\":{\"friendly_name\":\"tradfri remote control\"}},\"experimental\":{\"output\":\"json\"},\"external_converters\":[],\"frontend\":{\"host\":\"192.168.1.20\",\"port\":8080},\"groups\":{\"1\":{\"dev [... cut ...] 25,\"transportrev\":2},\"type\":\"zStack3x0\"},\"log_level\":\"error\",\"network\":{\"channel\":11,\"extended_pan_id\":\"0xdddddddddddddddd\",\"pan_id\":6754},\"permit_join\":true,\"version\":\"1.17.0\"}'"
}
QoS: 0 - Retain: false
Message 227 received on zigbee2mqtt/bridge/logging at 14:10:
{
    "level": "info",
    "message": "Zigbee: allowing new devices to join."
}
QoS: 0 - Retain: false
Message 226 received on zigbee2mqtt/bridge/request/permit_join at 14:10:
False
QoS: 0 - Retain: false
```

sebastianovide, 2021-01-10:
some logs
```
debug 2021-01-10 14:27:59: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data 'True'
info  2021-01-10 14:27:59: Zigbee: allowing new devices to join.
info  2021-01-10 14:27:59: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":60,"availability_whitelist":["0x001e5e090229f2fb"],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":true,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"ISO_8601_local","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":true,"rtscts":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x000b57fffe2f87f3":{"debounce":0.2,"debounce_ignore":["action","brightness"],"friendly_name":"Piano Dimmer","legacy":false},"0x000b57fffe346c9b":{"friendly_name":"Piano Lamp","retain":false},"0x000d6f000e1b7b68":{"friendly_name":"osram switch"},"0x00124b0019430c3b":{"friendly_name":"Router 1"},"0x00124b001f669750":{"friendly_name":"Main Room Light Door"},"0x00124b002228e310":{"friendly_name":"sonoff switch"},"0x00124b00224194de":{"friendly_name":"Plug 1"},"0x00124b002242b0af":{"friendly_name":"Temp Laundry"},"0x00124b00226aab73":{"friendly_name":"Plug laundry heater"},"0x00158d0003f0565c":{"friendly_name":"Temp Office","retain":false},"0x00158d0003f18415":{"friendly_name":"Temp Kitchen","retain":false},"0x00158d0003fae39e":{"friendly_name":"Motion Kitchen Window"},"0x00158d0003faf1dd":{"friendly_name":"Motion Landing Upstairs"},"0x00158d00040592f6":{"friendly_name":"Motion Landing Downstairs"},"0x00158d000423d5d8":{"friendly_name":"Temp Living","retain":false},"0x00158d000446239a":{"friendly_name":"Motion Entrance","retain":false},"0x00158d00044cac01":{"friendly_name":"Motion Living","retain":false},"0x001e5e090229f2fb":{"friendly_name":"Hive Signal Booster"},"0x086bd7fffedae4dd":{"friendly_name":"Landing Upstairs"},"0x90fd9ffffe1ef5cd":{"friendly_name":"kids 1"},"0xd0cf5efffe2ebfaa":{"friendly_name":"kids 2"},"0xd0cf5efffebeaa96":{"friendly_name":"Piano 1"},"0xd0cf5efffec8139e":{"friendly_name":"Piano 2"},"0xd0cf5efffef80e92":{"friendly_name":"tradfri remote control"}},"experimental":{"output":"json"},"external_converters":[],"frontend":{"host":"192.168.1.20","port":8080},"groups":{"1":{"devices":["0x90fd9ffffe1ef5cd/1","0xd0cf5efffe2ebfaa/1"],"friendly_name":"kids ceiling"},"901":{"devices":["0xd0cf5efffebeaa96/1","0xd0cf5efffec8139e/1"],"friendly_name":"piano ceiling"}},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://192.168.1.20"},"passlist":[],"permit_join":false,"queue":{"delay":100},"serial":{"disable_led":false,"port":"/dev/ttyUSB0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":true,"version":"1.17.0"}'
info  2021-01-10 14:27:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":"True"},"status":"ok"}'
debug 2021-01-10 14:28:07: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data 'False'
info  2021-01-10 14:28:07: Zigbee: allowing new devices to join.
info  2021-01-10 14:28:07: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":60,"availability_whitelist":["0x001e5e090229f2fb"],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":true,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"ISO_8601_local","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":true,"rtscts":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x000b57fffe2f87f3":{"debounce":0.2,"debounce_ignore":["action","brightness"],"friendly_name":"Piano Dimmer","legacy":false},"0x000b57fffe346c9b":{"friendly_name":"Piano Lamp","retain":false},"0x000d6f000e1b7b68":{"friendly_name":"osram switch"},"0x00124b0019430c3b":{"friendly_name":"Router 1"},"0x00124b001f669750":{"friendly_name":"Main Room Light Door"},"0x00124b002228e310":{"friendly_name":"sonoff switch"},"0x00124b00224194de":{"friendly_nam [... cut ...]
debug 2021-01-10 14:28:09: Received Zigbee message from 'Router 1', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"30485/0x000B57FFFE346C9B","inactiveText":"CHILD_FFD_RXI","minimumOffTime":0,"presentValue":1,"relinquishDefault":2}' from endpoint 8 with groupID 0
info  2021-01-10 14:28:09: MQTT publish: topic 'zigbee2mqtt/Router 1', payload '{"description":"30485/0x000B57FFFE346C9B","elapsed":8,"last_seen":"2021-01-10T14:28:09+00:00","led":false,"linkquality":144,"rssi":1,"type":"CHILD_FFD_RXI"}'
```

Koenkk, 2021-01-10:
The correct payload is `true` not `True`. Anyway this is also allowed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-18 by github-actions[bot]