Koenkk/zigbee2mqtt #5646: SNZB-03 Sonoff Motion Sensor Always occupancy=true with light on

srtxema, 2021-01-09:
i have another problem with this sensor.
after 1 minute without presence, it turn off but immediately turn on again. It occur for 15-20 times always that it detects presence

i send a log. Only first occupancy=true was with movement. after then bedroom was closed without people.

09/01/2021 11:52:17 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:52:17(-0.62 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:51:17(-59.94 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:51:16(-0.68 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:50:16(-60.01 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:50:15(-0.72 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:49:15(-60.12 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:49:15(-0.7 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:48:15(-59.96 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:48:14(-0.65 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:47:14(-60.13 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:47:13(-0.59 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:46:13(-60.02 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:46:13(-0.69 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:45:13(-60.01 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:45:12(-0.72 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:44:12(-60.03 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}
09/01/2021 11:44:11(-0.6 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":false,"tamper":false,"voltage":3000}
09/01/2021 11:43:11(-60.02 seconds) 
{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":false,"voltage":3000}

srtxema, 2021-01-09:
left is the motion sensor with problems. right is a ok motion sensor
[image]

itavero, 2021-01-10:
In the past I've experienced similar problems with other PIR sensors.
In that case there was something in the room that reflected IR (it was a large foil balloon that triggered our alarm at work during the night if I remember correctly).

As zigbee2mqtt generally just passes along any data seen from the sensor, I would suggest to verify that it's not the sensor or something in is environment that is causing this behavior.
At work, if we don't want a PIR sensor to trigger, we put it in a closed box or drawer.

srtxema, 2021-01-10:
It is in a bathroom with mirror without windows with light off totally. is possible it reflected IR?

itavero, 2021-01-10:
It is possible. I don't know how humid it is in the bathroom, but even condensation on the sensor can sometimes cause the sensor to trigger (as a drop of water I'd basically a lens and might bend some infrared light or reflection).

I must say if there's no window, it sounds less likely, unless there's some source of IR we are forgetting (is it a heated mirror for instance).

As suggested, the best way to eliminate the environment is putting it in a closed cardboard box or a closed drawer.

srtxema, 2021-01-10:
Thanks! I will look for a box

jlaitinen81, 2021-02-11:
I'm having the exactly same problem with one of my Sonoffs. I eliminated the reflection possibility, as there is no problem in the exactly same location with 2 other similar sensors.

closed 2021-02-19 by github-actions[bot]

kulcasosnowiec, 2022-02-17:
I'm having same problem. Always occupancy: true

alamoudimoh, 2022-07-08:
Same here and only with sonoff great sensors 😅