Koenkk/zigbee2mqtt #5654: Can't read state (`get`) for Immax Radiator Valve

jukben, 2021-01-09:
## What happened

I can't operate with [Immax 07703L](https://www.zigbee2mqtt.io/devices/07703L.html). Set and reads works as expected (as far I know) however once I try `get`:

```
mosquitto_pub -d -t zigbee2mqtt/valve-workroom/get -m '{"local_temperature": ""}'
```

I will see something like this

```
Jan 09 16:36:05 homebridge npm[467]: Zigbee2MQTT:info  2021-01-09 16:36:05: MQTT publish: topic 'zigbee2mqtt/valve-workroom', payload '{"away_mode":"OFF","battery_low":"OFF","child_lock":"UNLOCKED","current_heating_setpoint":"21.0","device_offline":"OFF","external_sensor_error":"OFF","high_temperature":"OFF","internal_sensor_error":"OFF","linkquality":48,"local_temperature":"22.0","low_temperature":"OFF","preset":"none","running_state":"idle","system_mode":"heat"}'
Jan 09 16:41:05 homebridge npm[467]: Zigbee2MQTT:info  2021-01-09 16:41:05: MQTT publish: topic 'zigbee2mqtt/valve-workroom', payload '{"away_mode":"OFF","battery_low":"OFF","child_lock":"UNLOCKED","current_heating_setpoint":"21.0","device_offline":"OFF","external_sensor_error":"OFF","high_temperature":"OFF","internal_sensor_error":"OFF","linkquality":51,"local_temperature":"22.0","low_temperature":"OFF","preset":"none","running_state":"idle","system_mode":"heat"}'
Jan 09 16:46:57 homebridge npm[467]: Zigbee2MQTT:error 2021-01-09 16:46:57: No converter available for 'local_temperature' ()
```

Thanks for this great project! I'm playing around with this project being integrate here https://github.com/itavero/homebridge-z2m/issues/37 so I'm trying to understand how everything works and is connected together. However I have a feeling that I might found a bug. 

---

## What did you expect to happen

I expect to see message with `local_temperature` field

## How to reproduce it (minimal and precise)

`get` topic is not working, every message I tried ended up with `no converted available`

## Debug info
Zigbee2MQTT version: 1.17
Adapter hardware: [CC2652RB](https://slae.sh/projects/cc2652/)
Adapter firmware version: [3.0](https://github.com/Koenkk/Z-Stack-firmware/pull/226)

itavero, 2021-01-10:
Please check the `access` field in the exposes information. I'm not sure if the particular field you are mentioning us "getable".
I don't know the bit definitions by head (yet), so you'd have to verify what the value (`5`, I believe?) means exactly.

jukben, 2021-01-10:
Hey @itavero! Thanks for reply, however I'm not sure I follow. Where can I get the affirmation? I was following the device "page" – https://www.zigbee2mqtt.io/devices/07703L.html#climate. 

```
local_temperature: Current temperature measured on the device (in °C). To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"local_temperature": ""}.
```

Based on this I thought it should work as I wrote above

Koenkk, 2021-01-10:
Reading this attribute is not supported by this device. The access value is `1`, to allow reading (`/get`) it should be `5` or `7`. Note that in z2m 1.17.0 this value is stil 5 but this has been corrected already in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-10 by Koenkk

jukben, 2021-01-10:
Ah I see, thanks a lot @Koenkk, sorry for resurrecting the thread – can you please point me where can I read those bits you are mentioning? Should I fix the readme for the device (basically removing the `/get` parts?) ? 👍

Koenkk, 2021-01-10:
It's not trivial to read them from the code, but it's set here https://github.com/Koenkk/zigbee-herdsman-converters/blob/ad2970502d1641a3912a3d6206872d8eb6ac4dd2/lib/exposes.js#L288 / https://github.com/Koenkk/zigbee-herdsman-converters/blob/ad2970502d1641a3912a3d6206872d8eb6ac4dd2/devices.js#L10112 .

Docs will be automatically updated with new release (1 February)