Koenkk/zigbee2mqtt #5655: Can't pair new devices until Coordinator is power cycled.

Morphy99, 2021-01-09:
## What happened
After some time my coordinator will fail to accept new devices.
## What did you expect to happen
The device will join when enabled.
## How to reproduce it (minimal and precise)
Enable joining and try to join a device, nothing happens in the logs.
Running Z2M for some time, or even whilst pairing new devices it randomly stops accepting new devices.
Restarting Z2M does not work.
Rebooting the system does not work.
Powering off the system (and coordinator) completely then powering on resolves the issue.
This has been an issue for a long time, even while using default firmware 20190608 this occurred.
Initially I thought it was something I was doing (interference or environmental) however it seems I'm not the only one:
https://community.home-assistant.io/t/zigbee2mqtt-cant-pair-new-devices/255174/16

## Debug info
Zigbee2MQTT version: 1.17.0 commit: 07cdc9d
Adapter hardware: CC2530
Adapter firmware version: 20190619
System is Raspberry Pi 4 running Z2M in docker.

Koenkk, 2021-01-10:
Looks similar to https://github.com/Koenkk/zigbee2mqtt/issues/3177 , might already be fixed with a new firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_Home_1.2/bin/default

Morphy99, 2021-01-10:
Hmm, would the issue described in #3177 show anything in the logs other than joining of new devices enabled message? Would the joining start and then fail or would nothing be visible at all? If it's the latter it may well be the same issue.

Is there any risk of using the dev firmware that I may lose the pairings? Obvs I can backup database but I don't want to try if I have to re-pair everything.

Koenkk, 2021-01-10:
> Would the joining start and then fail or would nothing be visible at all?
Nothing visible at all. There is no big risk in reflashing, pairings are peristed.

skumka, 2021-01-12:
@Koenkk Hi, Problem exist also for Usb cc2538+cc2592 with antenna. It does not seem to be a firmware problem.

Koenkk, 2021-01-12:
@skumka cc2538+cc2592 could also be affected since firmwares are similar.

Morphy99, 2021-01-30:
So after some weeks of testing, I'm finding it very difficult to reproduce but it's still happening with the latest firmware. Initially I thought it had worked but I think this was only due to the limited joins I'd attempted. I know this had occured whilst attempting to remove and join an Ikea Dimmer I've been testing for another issue. It has also occured when I've not done any joining but had joining disabled. Enabling it and then attempting to join several devices resulted in nothing until I power cycled the system.

Morphy99, 2021-03-02:
Still an issue

Morphy99, 2021-03-10:
Why did this close if I commented that it's still an issue??

Koenkk, 2021-03-10:
@Morphy99 did you already check https://github.com/Koenkk/zigbee2mqtt/issues/5655#issuecomment-757481660 ?

Morphy99, 2021-03-11:
> @Morphy99 did you already check [#5655 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/5655#issuecomment-757481660) ?

If you mean am I running the latest coordinator firmware, then yes have been running 20201128 for a while and it still occurs.

Morphy99, 2021-04-09:
Sorry @Koenkk been really busy but I think I did a sniff of this a while ago. Let me see if I can find it. Where do I upload this?

Koenkk, 2021-04-09:
You can drag it here.

Morphy99, 2021-04-25:
[dimmer-join_enabled-no_answer.zip](https://github.com/Koenkk/zigbee2mqtt/files/6371523/dimmer-join_enabled-no_answer.zip)
Is this ok?

Koenkk, 2021-04-25:
I see the device tries to join via a Philips Hue device (0x0017880103f1f476) but for some reason I doesn't transport the key. No clue why but doesn't look like we can solve this from z2m. I recommend to pair the device close to the coordinator instead.

Morphy99, 2021-04-25:
Thanks for taking a look! I was next to the coordinator when I did this but the Hue bulb is in the same room. I will try and see if I can recreate it and then switch off the hue bulb to see if it tries the coordinator.

Morphy99, 2021-05-01:
Here's two sniffs, the dimmer is the same as before but without the closest Hue bulb powered. I have one more Hue powered but further away so I hope it's not that one. 
The other sniff is a Xiaomi contact sensor that had dropped off the network. It's in a low coverage area (outside) but it seems to be ok for months then randomly loses the network. I'm not sure if it's related to this issue that it can't join again? I had to reset it and power down the coordinator to get it re-join. I was meant to try with just powering down the coordinator but the first time I did this it didn't seem to work. Maybe I didn't leave the power out long enough for the circuit charge to dissipate. 
[sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/6409529/sniff.zip)

Koenkk, 2021-05-01:
- Joining: it now tries to join via `0x000b57fffe260ef4`, but this device doesn't respond. What kind of device is this?
- Xiaomi devices are known to drop off when they have a low signal quality: https://www.zigbee2mqtt.io/devices/WXKG01LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network

Morphy99, 2021-05-04:
> * What kind of device is this?

This is an IKEA bulb [LED1623G12](https://www.zigbee2mqtt.io/devices/LED1623G12.html#ikea-led1623g12) The problem seems to be with the coordinator or system as hard powering it off and on results in the end device being able to pair again. This is across multiple CC2530's and multiple end devices.

> * Xiaomi devices are known to drop off when they have a low signal quality:

Thanks. I read [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1074#issuecomment-482937992) that deconz put in a [fix](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/31ef3962a4507e6310c23e4c497278f1fedb050b) for a similar issue, Was this implemented by Z2M? I know you were [keeping an eye on it](https://github.com/Koenkk/zigbee2mqtt/issues/1485#issuecomment-488568337) but I looked [here ](https://openbase.com/js/zigbee2mqtt/versions)and couldn't see anything searching for "aqara"?

Koenkk, 2021-05-04:
- Joining issue: just noticed you will still using an old firmware, the latest version contains a fix for device not being able to join after some uptime: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- This special handling is also in z2m, looked at the code but I don't see how this can prevent a drop-off, it looks like it is parsing code only (https://github.com/dresden-elektronik/deconz-rest-plugin/blob/31ef3962a4507e6310c23e4c497278f1fedb050b/de_web_plugin.cpp#L8762)

Morphy99, 2021-05-04:
> * just noticed you will still using an old firmware,

Where do you see that? Maybe the first sniff I uploaded was from a while ago and I hadn't updated at that point. Definitely running it now and have been for a while and the problem still happens:
[image]

Koenkk, 2021-05-04:
That was based on the OP (https://github.com/Koenkk/zigbee2mqtt/issues/5655#issue-782636071). However I currently don't know how we can fix this, I have no clue why the TRADFRI led doesn't transmit the network key.

Morphy99, 2021-05-04:
It's not just the TRADFRI bulb, the hue did the same and I'd wager any other routing device. That's a bummer we can't get to the bottom of it! Is there anything else I can try? Maybe removing all routers? Is there any debugging that can be done with the coordinator? I have spare devices to set another network up...

Koenkk, 2021-05-04:
To investigate this further you need to create a sniff in both the working and non-working situation: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2021-06-12 by github-actions[bot]