Koenkk/zigbee2mqtt #5664: Want to change device's Configure Reporting Interval

zijings, 2021-01-10:
## Describe the solution you'd like
I want to change device's Configure Reporting Interval while do not change and flash the firmware. Zigbee3.0 seems to support this, but I do not sure whether zigbee2mqtt can do this, or how to implement?

sledzik1984, 2021-01-10:
Yeah! That would be great feature. I have some WSDCGQ11LM sensors. That sensors reporting interval is quite long and charts made from data that came from that sensor are ugly. 

I have managed how to change reporting interval of Sonoff SNZB-02 but doing it was pretty hard and time consuming.

Koenkk, 2021-01-10:
This is supported since Zigbee2mqtt 1.17.0, see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceconfigure_reporting . Can also be done via the frontend, click on your device and go to the reporting tab.

closed 2021-01-10 by Koenkk

crowbarz, 2021-01-18:
I wanted to modify the reporting interval/min change settings as well to dial down updates from chatty devices (typically temperature, illumination and current), and had been patching zigbee-herdsman-converters (and rebasing after every z2m upgrade!) to change the model defaults until I discovered this feature.

Are the manually configured reporting changes persistent across z2m restarts, device reconfigures and/or z2m upgrades? I'm pretty sure that to date I have only seen the Zigbee configure commands being sent immediately after the initial interview at join time, but have not analysed the code to verify this.

EnsuingRequiem, 2022-05-09:
> what are the units of measure in the reporting view?

I checked in the PDF referenced and it indicates the units are in seconds

> Allows to send a Zigbee configure reporting command to a device. Refer to the Configure Reporting Command in the ZigBee Cluster Library for more information.)

spry-salt, 2024-02-28:
Do you find your sensors are still burning through their batteries at a fast rate?

spry-salt, 2025-06-29:
Did you figure out how to slow down the reporting rate from these devices using z2m? I've got some Smartthings Multipurpose Sensors that are reporting every minute and I need to slow them down because they're burning through their batteries.

spry-salt, 2025-06-30:
> Did you try this from a few posts earlier: [#5664 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/5664#issuecomment-757493678)

Thanks, I did but it either refused to change them (any of the attributes to do with the accelerometer) or it had no effect on the reporting frequency (temperature, battery). Were you successful?