Koenkk/zigbee2mqtt #5665: Quit (restart) after X numbers of consecutive soft resets

itavero, 2021-01-10:
Over the last few months, I've been noticing that zigbee2mqtt would become unresponsive every week or so (sometimes twice a week and a few weeks it worked without problems).
Usually the log indicate that something in the communication with the adapter is going wrong.
Restarting the container usually fixes the problem. Sometimes I have the unplug the adapter and plug it back in.

Because of this, I figured I'd configure the `soft_reset_timeout` to 20 seconds (it used to be disabled).

This morning my spouse notified me that the lights weren't working and I had a look in the logs.
It seems that the soft reset was triggered... a lot.

```
Zigbee2MQTT:warn  2021-01-10 07:00:18: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:00:20: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:00:40: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:00:43: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:01:03: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:01:05: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:01:25: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:01:27: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:01:47: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:01:49: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:02:09: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:02:11: Soft resetted ZNP due to timeout,
-- even more of these log entries --
Zigbee2MQTT:warn  2021-01-10 07:11:02: Soft reset timeout triggered
Zigbee2MQTT:warn  2021-01-10 07:11:04: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:11:24: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:11:26: Soft resetted ZNP due to timeout,
Zigbee2MQTT:warn  2021-01-10 07:11:46: Soft reset timeout triggered,
Zigbee2MQTT:warn  2021-01-10 07:11:48: Soft resetted ZNP due to timeout,
```

It seems to trigger every 20 seconds (as configured), but it apparently does not help.
When I noticed this, I restarted the container and immediately everything worked again (without physically disconnecting the adapter).

It might be a bug in the soft reset functionality that should be solved.
Not sure how to validate if the soft reset is working correctly.

## Describe the solution you'd like
If it is not a bug in the soft reset, I propose to add an option to configure a maximum number of consecutive soft resets (or maximum soft resets per amount of time, that would also work).
If this threshold is reached, zigbee2mqtt should stop.
That way the Docker container will end and, if configured to do so, restart automatically.

## Additional context
Running v1.17.0 since it was released, before that I was on `latest-dev` (which I updated frequently).

I'm using a TI LAUNCHXL-CC26X2R1 as adapter, which is connected to an old desktop (my home server) via a USB cable of about 1.5m.
```json
{
   "meta": {
      "maintrel": 1,
      "majorrel": 2,
      "minorrel": 7,
      "product": 1,
      "revision": 20201026,
      "transportrev": 2
   },
   "type": "zStack3x0"
}
```

Koenkk, 2021-01-10:
`soft_reset_timeout` is a deprecated feature, I don't expect it to work well.

Instead we should investigate the reason of the CC26X2R1 crash. Can you provide the herdsman debug log from starting z2m till fail? 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

itavero, 2021-01-10:
I'll enable debug logging tonight and I'll try to get some logs when it fails again.

itavero, 2021-01-10:
Also went through the logs I already had to see what kind of errors are in there. Not sure how helpful this is without the zigbee-herdsman logs, but I figured I might as well add them here.

```
error 2021-01-08 15:49:15: Publish 'set' 'state' to 'light_officedesk_left' failed: 'Error: Command 0x0017880106117b15/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'

error 2021-01-08 15:49:23: Publish 'set' 'state' to 'light_office' failed: 'Error: Command 0x0017880100d9b08f/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'
```

```
error 2021-01-10 06:45:56: Publish 'set' 'state' to 'light_kitchen' failed: 'Error: Command 0x0017880100d9b1cb/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":52136,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":43,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,29,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'

error 2021-01-10 06:46:00: Publish 'set' 'state' to 'light_kitchen' failed: 'Error: Command 0x0017880100d9b1cb/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":52136,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":44,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,30,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
```

itavero, 2021-01-11:
It happened again this morning. The log file however is rather large (over 60M). Anything particular I should look for in the zigbee-herdsman portion, or shall I just dump the entire file somewhere?

Koenkk, 2021-01-11:
- The first error happens at line 4316 (`failed with status '(0x02: INVALID_PARAM)'`)
- The first soft reset happens some time above that at line 4137. I suggest you disable the soft reset timeout first as I expect it to do more harm than good.
- If that doesn't help I suggest you disable the availability feature first and see if your coordinator still crashes.

itavero, 2021-01-11:
Thanks for digging in to the logs.

I'll disable the soft reset. I had problems before, so it won't be the root cause though.

Regarding availability, I didn't have it on before. Turned it on yesterday to test integration of this data into homebridge-z2m. I'll also disable it again.

I'll post another log when it becomes unresponsive again.

itavero, 2021-01-15:
This morning it became unresponsive again. Unfortunately I forgot to turn off permit join yesterday evening (previously I used Zigbee2MqttAssistant to do this, which automatically turns it back off after a certain amount of time, but nowadays I'm using the new front-end).
I hope this doesn't "taint" the logs too much.

This time I actually had to physically 
unplug the adapter and plug it back in again to get zigbee2mqtt to work.

The log can be found here: [z2m_log_202101150705_excerpt.txt](https://github.com/Koenkk/zigbee2mqtt/files/5820319/z2m_log_202101150705_excerpt.txt)

Koenkk, 2021-01-16:
- Can you also try with this fw: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin ?
- Can you provide the logs from starting z2m till the error? This logs differs from other logs where I saw the coordinator crashing. Usually you see commands failing and after some time the whole coordinator crashes. Here I only see the complete crash, everything is ok until `2021-01-15T04:52:39.215Z`. Would also be interesting to see the `dmesg` output at that time. 
- Can you also try connecting it through a shorter USB extension cable? (the one that came bundled)

itavero, 2021-01-16:
Thanks again for taking the time to look into it.
I'll see if I can find a moment this weekend to flash the firmware you mentioned and try to get the dmesg logs.

A shorter cable I can do for sure.

itavero, 2021-01-16:
Dmesg log from my docker host from that particular day: [dmesg_20210115_host.txt](https://github.com/Koenkk/zigbee2mqtt/files/5824323/dmesg_20210115_host.txt)

I've uploaded the larger log of the Zigbee2MQTT container to OneDrive, as it was too large for GitHub. You can find it [via this link](https://1drv.ms/u/s!AqXCSwUD0YAGhvkQV6hid4zg8gt8Og?e=WtCTYq).

Koenkk, 2021-01-16:
```
[Fri Jan 15 05:59:51 2021] usb 3-9.3: USB disconnect, device number 5
....
[Fri Jan 15 05:59:59 2021] usb 3-9.3: new high-speed USB device number 6 using xhci_hcd
[Fri Jan 15 06:00:00 2021] usb 3-9.3: New USB device found, idVendor=0451, idProduct=bef3, bcdDevice= 1.00
[Fri Jan 15 06:00:00 2021] usb 3-9.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Jan 15 06:00:00 2021] usb 3-9.3: Product: XDS110 (03.00.00.12) Embed with CMSIS-DAP
[Fri Jan 15 06:00:00 2021] usb 3-9.3: Manufacturer: Texas Instruments
[Fri Jan 15 06:00:00 2021] usb 3-9.3: SerialNumber: L1100GP2
[Fri Jan 15 06:00:00 2021] cdc_acm 3-9.3:1.0: ttyACM0: USB ACM device
[Fri Jan 15 06:00:00 2021] cdc_acm 3-9.3:1.3: ttyACM1: USB ACM device
```

Looks that it disconnects for some reason. Could be due to the long usb cable.

itavero, 2021-01-16:
Given the timestamp, I'm pretty sure it's me unplugging the adapter and plugging it back in to get things working again.
But I'll try the other firmware and a shorter cable starting this weekend. Hopefully it helps 🤞

itavero, 2021-01-26:
No issues so far since I updated the adapter firmware and started using a shorter USB cable. (Hopefully I didn't jynx it 🤡 )

Closing this issue. If it starts occurring I gain, I might reopen it.

closed 2021-01-26 by itavero